espressif/esp-hosted #417: esp32 and esp32c6 sdio register addresses are different

dxzzxb, 2024-06-28:
For example, ESP_SLAVE_SLCHOST_BASE is 0x3FF55000 in the esp32 manual, but in my esp32c6 manual it is 0x60018000, but in esp_hosted_fg\host\linux\host_driver\esp32\sdio\esp_sdio_decl.h I only see #define ESP_SLAVE_SLCHOST_BASE 0x3FF55 000, this is on esp32c6 Can it work? I want to know why.

[image]

SohKamYung-Espressif, 2024-06-28:
@dxzzxb The ESP SDIO slave driver only cares about the lower 10 bits when the host wants to access the SDIO registers.

If you check `esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio_api.c`, we apply `ESP_ADDRESS_MASK` (0x3FF) to the register value in `esp_read_reg()` and `esp_write_reg()`.

We may do some clean-up in the header file to make this clearer.

dxzzxb, 2024-06-28:
> @dxzzxb The ESP SDIO slave driver only cares about the lower 10 bits when the host wants to access the SDIO registers.
>
> If you check `esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio_api.c`, [...]

ok ,I got it! thanks for your reply

closed 2024-06-28 by dxzzxb