espressif/esp-hosted #419: SPI operation without handshake and data ready pins

thelieb, 2024-07-01:
Hello,
Could the esp-hosted-ng solution work without handshake and data ready gpios connected?
How heavily does the linux driver (and firmware) rely on these? 
What would be the consequences of trying to install an ESP32-C3 module without it? 
Should parts of the software be significantly rewritten? Or would performance be severely degraded?

Thank you.

mantriyogesh, 2024-07-01:
Some design understanding:

[SPI_design_note.pdf](https://github.com/user-attachments/files/16051729/SPI_design_note.pdf)
Handshake is mandatory, as it lets host know that ESP has populated next SPI desciptor to lower ESP-IDF spi driver or not. (mandatory)
To make SPI bus idle over the time (instead of running continuous dummy data, we use data ready as interrupt).

Data ready is for async trigger when esp has some data even when SPI bus is idle (no clock).
In either of below cases, Data Ready pin could be made optional with some changes:
1. If you do not intend to have async ESP tx to host **OR**
2. You are ready to sacrifice SPI bus always on (no SI clock idling)

We are also going to implement QSPI (half duplex) which would need one less GPIO, but should take good time (no immediate plans).

Apart from these two, reset pin is also mandatory for SPI.
You can also check solution with ESP32-C6 with SDIO.

We are also thinking of adding UART (bth Wi-Fi & Bluetooth), with would be low data rate solution but fewer pins. Again, no immediate plans.

thelieb, 2024-07-01:
@mantriyogesh I appreciate your detailed reply. Thank you.

closed 2024-07-01 by thelieb