espressif/esp-hosted #421: How to set a custom mac as a base mac

xmlhp, 2024-07-02:
We have already burned a custom mac into Block3 via espefuse. How can I set Block3 as the base mac?

mantriyogesh, 2024-07-02:
using [esp_efuse_read_block()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/efuse.html#_CPPv420esp_efuse_read_block17esp_efuse_block_tPv6size_t6size_t)?

For example,
```c
#include <stdio.h>
#include "esp_system.h"
#include "esp_log.h"
#include "esp_efuse.h"
#include "esp_efuse_table.h"

static const char *TAG = "MAC_READER";

void app_main(void)
{
    uint8_t mac[6]; // Array to hold the MAC address

    // Read the MAC address from Block 3
    esp_err_t ret = esp_efuse_read_block(EFUSE_BLK3, mac, 0, 48);

    if (ret == ESP_OK) {
        ESP_LOGI(TAG, "MAC address read from eFuse Block 3:");
        ESP_LOGI(TAG, "%02x:%02x:%02x:%02x:%02x:%02x",
                 mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    } else {
        ESP_LOGE(TAG, "Failed to read MAC address from eFuse Block 3. Error: %d", ret);
    }
}
```

xmlhp, 2024-07-02:
hosted-ng，Which file and where should I modify?

xmlhp, 2024-07-02:
I was told to do the following, but I don't know which file to modify and where to modify it in hosted-ng

https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/system/misc_system_api.html#mac

After calling esp_efuse_mac_get_custom() or esp_read_mac() to get a custom eFuse MAC address, set this MAC address as the base MAC address. There are two ways:

Using the old API: call esp_base_mac_addr_set().

Using the new API: call esp_iface_mac_addr_set(), and use the ESP_MAC_BASE parameter.

mantriyogesh, 2024-07-02:
These are different things.
Little confused, what you wish to do exactly.

esp_iface_mac_addr_set() should be temporary, till next boot.
If you wish to do it permanent, you might need to use NVS or eFuse.

But anyway, it would be better answered in IDF forum: https://github.com/espressif/esp-idf/issues than esp-hosted.

xmlhp, 2024-07-02:
I hope it is permanent, we have burned a customized mac into efuse. I don't understand where the decision is made to use the customer's mac or the ESP's own MAC.

xmlhp, 2024-07-04:
In the network_adapter/main/app_main.c file, I use esp_efuse_mac_get_custom to read the mac before esp_read_mac in the app_main function, and then use esp_base_mac_addr_set to set the read mac to the base MAC. Is this OK?

`void app_main()
{
    esp_err_t ret;
    uint8_t prio_q_idx = 0;
    uint8_t mac[MAC_ADDR_LEN] = {0};
    uint8_t capa = 0;

    debug_log_firmware_version();

    capa = get_capabilities();

    /*Initialize NVS*/
    ret = nvs_flash_init();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    ret = initialise_wifi();
    ESP_ERROR_CHECK( ret );

    init_sem = xSemaphoreCreateBinary();
    if (init_sem == NULL) {
        ESP_LOGE(TAG, "Failed to create init semaphore\n");
        return;
    }
  //added by felix  get mac from efuse blk3,then set it be base mac
    **esp_efuse_mac_get_custom(mac);
   esp_base_mac_addr_set(mac);**
#ifdef CONFIG_BT_ENABLED
    initialise_bluetooth();

    ret = esp_read_mac(mac, ESP_MAC_BT);
    if (ret) {
        ESP_LOGE(TAG,"Failed to read BT Mac addr\n");
    } else {
        ESP_LOGI(TAG, "ESP Bluetooth MAC addr: %2x-%2x-%2x-%2x-%2x-%2x\n",
                mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    }
#endif

    if_context = interface_insert_driver(event_handler);
#if CONFIG_ESP_SPI_HOST_INTERFACE
    datapath = 1;
`

mantriyogesh, 2024-07-04:
With the new API, with specific esp_mac_type_t you have in mind, specific mac could be set

```
esp_err_t esp_iface_mac_addr_set(const uint8_t *mac, esp_mac_type_t type)
```

But anyway logs good to me

kapilkedawat, 2024-08-06:
Hi @xmlhp, the logic looks correct, let us know if you need further help or we can close this.