espressif/esp-hosted #422: Crash when the linux reboot

caipiblack, 2024-07-02:
Hello,

I am using esp_hosted_ng from this commit: 29208b6c156be06dfd90dadc6dd43acd4d98d748 (TAG: ng-v1.0.2)

Our modifications on the driver are minimalist:

1. Change esp_board.bus_num
2. Change resetpin, HANDSHAKE_PIN, SPI_DATA_READY_PIN

The driver works pretty well, we use it with wpa_supplicant without issues since many months.

But since the begining we see a bad issue that we would like to treat now:

- Every time we call the `reboot` command in our system we have a crash on esp_hosted_ng driver
- It doesn't appears if we call `poweroff` command

Here is the logs:

```
[  OK  ] Finished System Reboot.
[  OK  ] Reached target System Reboot.
[   51.987262] systemd-shutdown[1]: Using hardware watchdog 'Renesas RZ/G2L WDT Watchdog', version 0, device /dev/watchdog0
[   51.998263] systemd-shutdown[1]: Failed to set timeout to 10min: Invalid argument
[   52.051851] systemd-shutdown[1]: Syncing filesystems and block devices.
[   52.060096] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   52.081066] systemd-journald[175]: Received SIGTERM from PID 1 (systemd-shutdow).
[   52.101966] audit: type=1335 audit(1719926228.763:7): pid=175 uid=0 auid=4294967295 tty=(none) ses=4294967295 comm="systemd-journal" exe="/lib/systemd/systemd-journald" nl-mcgrp=1 op=disconnect res=1
[   52.132627] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   52.151996] systemd-shutdown[1]: Unmounting file systems.
[   52.159411] systemd-shutdown[1]: All filesystems unmounted.
[   52.165062] systemd-shutdown[1]: Deactivating swaps.
[   52.170236] systemd-shutdown[1]: All swaps deactivated.
[   52.175574] systemd-shutdown[1]: Detaching loop devices.
[   52.184954] systemd-shutdown[1]: All loop devices detached.
[   52.190813] systemd-shutdown[1]: Stopping MD devices.
[   52.196301] systemd-shutdown[1]: All MD devices stopped.
[   52.201654] systemd-shutdown[1]: Detaching DM devices.
[   52.207066] systemd-shutdown[1]: All DM devices detached.
[   52.212500] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[   52.226598] systemd-shutdown[1]: Syncing filesystems and block devices.
[   52.234823] systemd-shutdown[1]: Rebooting.
[   52.269569] reboot: Restarting system
[   52.273560] ------------[ cut here ]------------
[   52.278166] No atomic I2C transfer handler for 'i2c-0'
[   52.283337] WARNING: CPU: 0 PID: 1 at drivers/i2c/i2c-core.h:40 i2c_smbus_xfer+0x108/0x120
[   52.291569] Modules linked in: moal(O) mlan(O) [last unloaded: esp_hosted_ng]
[   52.298696] CPU: 0 PID: 1 Comm: systemd-shutdow Tainted: G           O      5.10.175-cip29-yocto-standard #1
[   52.308482] Hardware name: Development board based on r9a07g043u11 (DT)
[   52.315070] pstate: 60400085 (nZCv daIf +PAN -UAO -TCO BTYPE=--)
[   52.321053] pc : i2c_smbus_xfer+0x108/0x120
[   52.325218] lr : i2c_smbus_xfer+0x108/0x120
[   52.329381] sp : ffff800011073bb0
[   52.332680] x29: ffff800011073bb0 x28: ffff000009450000 
[   52.337973] x27: 0000000000000000 x26: 0000000000000000 
[   52.343265] x25: 0000000000000013 x24: 0000000000000000 
[   52.348557] x23: 0000000000000000 x22: 0000000000000058 
[   52.353850] x21: ffff800011073c16 x20: 0000000000000002 
[   52.359143] x19: ffff00000a3220c8 x18: 0000000000000030 
[   52.364435] x17: 0000000000000000 x16: 0000000000000000 
[   52.369727] x15: ffffffffffffffff x14: 0720072007200720 
[   52.375019] x13: ffff800010f11a18 x12: 00000000000004ce 
[   52.380311] x11: 000000000000019a x10: ffff800010f69a18 
[   52.385604] x9 : 00000000fffff000 x8 : ffff800010f11a18 
[   52.390896] x7 : ffff800010f69a18 x6 : 0000000000000000 
[   52.396189] x5 : 000000000000bff4 x4 : 0000000000000000 
[   52.401481] x3 : 00000000ffffffff x2 : 0000000000000000 
[   52.406773] x1 : 0000000000000000 x0 : ffff000009450000 
[   52.412066] Call trace:
[   52.414503]  i2c_smbus_xfer+0x108/0x120
[   52.418324]  i2c_smbus_write_byte_data+0x40/0x70
[   52.422926]  da9062_wdt_restart+0x30/0x90
[   52.426921]  watchdog_restart_notifier+0x1c/0x3c
[   52.431523]  atomic_notifier_call_chain+0x60/0x90
[   52.436209]  do_kernel_restart+0x24/0x30
[   52.440118]  machine_restart+0x50/0x54
[   52.443851]  __do_sys_reboot+0x200/0x260
[   52.447758]  __arm64_sys_reboot+0x24/0x30
[   52.451751]  el0_svc_common.constprop.0+0x78/0x1c4
[   52.456523]  do_el0_svc+0x24/0x9c
[   52.459826]  el0_svc+0x14/0x20
[   52.462867]  el0_sync_handler+0xb0/0xb4
[   52.466687]  el0_sync+0x180/0x1c0
[   52.469986] ---[ end trace 8abe1e55710591ae ]---
[   52.474707] ------------[ cut here ]------------
[   52.479315] WARNING: CPU: 0 PID: 1 at kernel/rcu/tree_plugin.h:297 rcu_note_context_switch+0x44/0x340
[   52.488496] Modules linked in: moal(O) mlan(O) [last unloaded: esp_hosted_ng]
[   52.495614] CPU: 0 PID: 1 Comm: systemd-shutdow Tainted: G        W  O      5.10.175-cip29-yocto-standard #1
[   52.505400] Hardware name: Development board based on r9a07g043u11 (DT)
[   52.511988] pstate: 20400085 (nzCv daIf +PAN -UAO -TCO BTYPE=--)
[   52.517970] pc : rcu_note_context_switch+0x44/0x340
[   52.522828] lr : __schedule+0xb4/0x6e0
[   52.526559] sp : ffff800011073840
[   52.529859] x29: ffff800011073840 x28: 0000000000000002 
[   52.535151] x27: 0000000000000000 x26: ffff00000a3220c8 
[   52.540444] x25: ffff000009450000 x24: ffff800010fe8000 
[   52.545736] x23: 0000000000000000 x22: ffff000009450000 
[   52.551028] x21: ffff000009450000 x20: ffff800010de29c0 
[   52.556322] x19: ffff00003fe109c0 x18: 0000000000000030 
[   52.561614] x17: 0000000000000000 x16: 0000000000000000 
[   52.566906] x15: ffffffffffffffff x14: 0000000000000215 
[   52.572198] x13: 0000000000000000 x12: 0000000000000000 
[   52.577491] x11: 0000000000000000 x10: 0000000000000940 
[   52.582783] x9 : ffff8000110737c0 x8 : 0000000000000001 
[   52.588075] x7 : ffff00003fe0aba8 x6 : 0000000042101000 
[   52.593368] x5 : 0000010042101000 x4 : 0000000000000000 
[   52.598660] x3 : 0000000000000000 x2 : ffff800010ef8b50 
[   52.603952] x1 : ffff800010a8cec0 x0 : 0000000000000001 
[   52.609244] Call trace:
[   52.611682]  rcu_note_context_switch+0x44/0x340
[   52.616194]  __schedule+0xb4/0x6e0
[   52.619581]  schedule+0x70/0x104
[   52.622796]  schedule_timeout+0x80/0xf0
[   52.626616]  wait_for_completion_timeout+0x80/0x10c
[   52.631475]  riic_xfer+0xe4/0x160
[   52.634779]  __i2c_transfer+0x160/0x4ec
[   52.638598]  i2c_smbus_xfer_emulated+0xe0/0x600
[   52.643109]  __i2c_smbus_xfer+0x108/0x210
[   52.647101]  i2c_smbus_xfer+0x7c/0x120
[   52.650834]  i2c_smbus_write_byte_data+0x40/0x70
[   52.655432]  da9062_wdt_restart+0x30/0x90
[   52.659425]  watchdog_restart_notifier+0x1c/0x3c
[   52.664025]  atomic_notifier_call_chain+0x60/0x90
[   52.668710]  do_kernel_restart+0x24/0x30
[   52.672616]  machine_restart+0x50/0x54
[   52.676350]  __do_sys_reboot+0x200/0x260
[   52.680256]  __arm64_sys_reboot+0x24/0x30
[   52.684249]  el0_svc_common.constprop.0+0x78/0x1c4
[   52.689020]  do_el0_svc+0x24/0x9c
[   52.692320]  el0_svc+0x14/0x20
[   52.695361]  el0_sync_handler+0xb0/0xb4
[   52.699180]  el0_sync+0x180/0x1c0
[   52.702479] ---[ end trace 8abe1e55710591af ]---
[   52.707212] da9062 0-0058: Failed to shutdown (err = -6)
[   53.212582] Reboot failed -- System halted
```

Did someone know what appens and how to fix it ?

mantriyogesh, 2024-07-02:
Does it also crash when you manually unload the driver?

Thank you for the detailed log.if in case you have collected the ESP log as well, would be good to attach.

@kapilkedawat

caipiblack, 2024-07-02:
I try to do it, unfortunately the problem stil appears:

```
root@hubmz-rzg2ul:~# rmmod esp_hosted_ng
root@hubmz-rzg2ul:~# echo $?
0
root@hubmz-rzg2ul:~# rmmod esp_hosted_ng
rmmod: ERROR: Module esp_hosted_ng is not currently loaded
root@hubmz-rzg2ul:~# reboot
root@hubmz-rzg2ul:~#
.......
[  OK  ] Finished System Reboot.
[  OK  ] Reached target System Reboot.
[   59.015968] systemd-shutdown[1]: Using hardware watchdog 'Renesas RZ/G2L WDT Watchdog', version 0, device /dev/watchdog0
[   59.026964] systemd-shutdown[1]: Failed to set timeout to 10min: Invalid argument
[   59.073910] systemd-shutdown[1]: Syncing filesystems and block devices.
[   59.082181] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   59.103134] systemd-journald[174]: Received SIGTERM from PID 1 (systemd-shutdow).
[   59.124421] audit: type=1335 audit(1719927643.662:7): pid=174 uid=0 auid=4294967295 tty=(none) ses=4294967295 comm="systemd-journal" exe="/lib/systemd/systemd-journald" nl-mcgrp=1 op=disconnect res=1
[   59.153161] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   59.172465] systemd-shutdown[1]: Unmounting file systems.
[   59.179843] systemd-shutdown[1]: All filesystems unmounted.
[   59.185535] systemd-shutdown[1]: Deactivating swaps.
[   59.190677] systemd-shutdown[1]: All swaps deactivated.
[   59.195994] systemd-shutdown[1]: Detaching loop devices.
[   59.205506] systemd-shutdown[1]: All loop devices detached.
[   59.211273] systemd-shutdown[1]: Stopping MD devices.
[   59.216839] systemd-shutdown[1]: All MD devices stopped.
[   59.222195] systemd-shutdown[1]: Detaching DM devices.
[   59.227652] systemd-shutdown[1]: All DM devices detached.
[   59.233078] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[   59.247218] systemd-shutdown[1]: Syncing filesystems and block devices.
[   59.255360] systemd-shutdown[1]: Rebooting.
[   59.289657] reboot: Restarting system
[   59.293644] ------------[ cut here ]------------
[   59.298248] No atomic I2C transfer handler for 'i2c-0'
[   59.303419] WARNING: CPU: 0 PID: 1 at drivers/i2c/i2c-core.h:40 i2c_smbus_xfer+0x108/0x120
[   59.311651] Modules linked in: moal(O) mlan(O) [last unloaded: esp_hosted_ng]
[   59.318776] CPU: 0 PID: 1 Comm: systemd-shutdow Tainted: G           O      5.10.175-cip29-yocto-standard #1
[   59.328562] Hardware name: Development board based on r9a07g043u11 (DT)
[   59.335151] pstate: 60400085 (nZCv daIf +PAN -UAO -TCO BTYPE=--)
[   59.341133] pc : i2c_smbus_xfer+0x108/0x120
[   59.345298] lr : i2c_smbus_xfer+0x108/0x120
[   59.349462] sp : ffff800011073bb0
[   59.352761] x29: ffff800011073bb0 x28: ffff000009450000 
[   59.358054] x27: 0000000000000000 x26: 0000000000000000 
[   59.363346] x25: 0000000000000013 x24: 0000000000000000 
[   59.368639] x23: 0000000000000000 x22: 0000000000000058 
[   59.373931] x21: ffff800011073c16 x20: 0000000000000002 
[   59.379224] x19: ffff00000a3220c8 x18: 0000000000000030 
[   59.384516] x17: 0000000000000000 x16: 0000000000000000 
[   59.389808] x15: ffffffffffffffff x14: 0720072007200720 
[   59.395100] x13: ffff800010f11a18 x12: 00000000000004ce 
[   59.400393] x11: 000000000000019a x10: ffff800010f69a18 
[   59.405685] x9 : 00000000fffff000 x8 : ffff800010f11a18 
[   59.410977] x7 : ffff800010f69a18 x6 : 0000000000000000 
[   59.416270] x5 : 000000000000bff4 x4 : 0000000000000000 
[   59.421562] x3 : 00000000ffffffff x2 : 0000000000000000 
[   59.426854] x1 : 0000000000000000 x0 : ffff000009450000 
[   59.432147] Call trace:
[   59.434585]  i2c_smbus_xfer+0x108/0x120
[   59.438406]  i2c_smbus_write_byte_data+0x40/0x70
[   59.443008]  da9062_wdt_restart+0x30/0x90
[   59.447003]  watchdog_restart_notifier+0x1c/0x3c
[   59.451605]  atomic_notifier_call_chain+0x60/0x90
[   59.456291]  do_kernel_restart+0x24/0x30
[   59.460200]  machine_restart+0x50/0x54
[   59.463934]  __do_sys_reboot+0x200/0x260
[   59.467840]  __arm64_sys_reboot+0x24/0x30
[   59.471834]  el0_svc_common.constprop.0+0x78/0x1c4
[... 32 lines cut ...]
[   59.639906]  schedule+0x70/0x104
[   59.643121]  schedule_timeout+0x80/0xf0
[   59.646942]  wait_for_completion_timeout+0x80/0x10c
[   59.651800]  riic_xfer+0xe4/0x160
[   59.655103]  __i2c_transfer+0x160/0x4ec
[   59.658923]  i2c_smbus_xfer_emulated+0xe0/0x600
[   59.663434]  __i2c_smbus_xfer+0x108/0x210
[   59.667426]  i2c_smbus_xfer+0x7c/0x120
[   59.671159]  i2c_smbus_write_byte_data+0x40/0x70
[   59.675757]  da9062_wdt_restart+0x30/0x90
[   59.679751]  watchdog_restart_notifier+0x1c/0x3c
[   59.684349]  atomic_notifier_call_chain+0x60/0x90
[   59.689033]  do_kernel_restart+0x24/0x30
[   59.692939]  machine_restart+0x50/0x54
[   59.696673]  __do_sys_reboot+0x200/0x260
[   59.700579]  __arm64_sys_reboot+0x24/0x30
[   59.704572]  el0_svc_common.constprop.0+0x78/0x1c4
[   59.709343]  do_el0_svc+0x24/0x9c
[   59.712644]  el0_svc+0x14/0x20
[   59.715685]  el0_sync_handler+0xb0/0xb4
[   59.719504]  el0_sync+0x180/0x1c0
[   59.722803] ---[ end trace 5d80fe005b9ee258 ]---
[   59.727541] da9062 0-0058: Failed to shutdown (err = -6)
[   60.232907] Reboot failed -- System halted
```

mantriyogesh, 2024-07-02:
Although, the symbols in the crash doesn't originate from ESP-Hosted.

I think worth to isolate the origin. For example, check your i2c driver.

i2c_smbus_xfer+0x108/0x120 specifically

mantriyogesh, 2024-07-02:
Sure, no worries.
If you have anything for Hosted, you know where to find us now!