espressif/esp-hosted #423: In ng mode, there is a situation where esp32c3 cannot apply for memory

linchanghe123, 2024-07-05:
Hi mantriyogesh,

In ng mode, I repeatedly tried to connect to a non-existent AP, but the esp32c3 program encountered a situation where it could not apply for memory. And then all the connections and scanning actions stopped responding.

I am using version 1.0.3. Is there any related bug in this version? Has it been fixed so far?

[image]

mantriyogesh, 2024-07-05:
This looks like a bug.

Can you please help in stating exact git commit used at slave and host side?
Also, a patch what you had changed either side on above git commit would be really helpful to reproduce this issue.

@kapilkedawat

linchanghe123, 2024-07-05:
The version and node information we use are as follows：

a. ESP idf path and nodes:
Software package name: esp-idf-5.1.3. zip
Download path: https://github.com/espressif/esp-idf/tree/v5.1.1
Download node: https://github.com/espressif/esp-idf/commit/e088c3766ba440e72268b458a68f27b6e7d63986
Download command: git clone - b v5.1.1-- recursive https://github.com/espressif/esp-idf.git
b. ESP hosted path and nodes:
Software package name: esp hosted master.zip
Download path: https://github.com/espressif/esp-hosted
Download node: https://github.com/espressif/esp-hosted/commit/490d2c3ed03139650d82b09f079d6fc4b1c48e49
Download command: git clone - b master https://github.com/espressif/esp-hosted.git

We have hardly made any modifications to the source code.

kapilkedawat, 2024-07-05:
fixed by https://github.com/espressif/esp-hosted/commit/76085845fa007342e5f2c3811b03abc9f5ddbbde#diff-10e9cedc9122408623d5822d8a1e18cbab943915746c35fc6822b185270adeb4R1252

closed 2024-07-06 by linchanghe123