espressif/esp-hosted #426: Unable to configure WiFi

sergey-suloev, 2024-07-07:
hello,

with the latest NG version I can't get WiFi connection working .

here is my log:

[   10.666271] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   10.673256] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   10.679840] esp32_spi: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SPI
[   10.687509] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   10.693940] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   10.700401] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   10.710602] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   10.719548] esp32_spi: print_reset_reason: POWERON_RESET
[   10.726128] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   10.736793] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   10.848879] esp32_spi: esp_cfg80211_get_tx_power: 
[   11.083498] esp32_spi: init_bt: ESP Bluetooth init
[   11.088990] esp32_spi: print_capabilities: Capabilities: 0xf8. Features supported are:
[   11.100390] esp32_spi: print_capabilities: 	 * WLAN on SPI
[   11.106204] esp32_spi: print_capabilities: 	 * BT/BLE
[   11.111431] esp32_spi: print_capabilities: 	   - HCI over SPI
[   11.117321] esp32_spi: print_capabilities: 	   - BT/BLE dual mode
[   11.352102] esp32_spi: esp_cfg80211_get_tx_power: 
[   11.378482] esp32_spi: esp_cfg80211_get_tx_power: 
[   11.407622] esp32_spi: esp_cfg80211_get_tx_power: 
[   11.455947] esp32_spi: esp_cfg80211_get_tx_power: 
[   11.506951] esp32_spi: esp_cfg80211_get_tx_power: 
[   11.679009] esp32_spi: esp_cfg80211_scan: 
[   11.788889] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.794301] Bluetooth: BNEP filters: protocol multicast
[   11.802396] Bluetooth: BNEP socket layer initialized
[   11.822691] Bluetooth: MGMT ver 1.22
[   11.842187] NET: Registered PF_ALG protocol family
[   11.871857] esp32_spi: esp_inetaddr_event: NETDEV_UP interface lo ip changed to  127.000.000.001
[   11.881190] esp32_spi: cmd_set_ip_address: Invalid argument
[   13.416126] esp32_spi: esp_cfg80211_authenticate: 
[   13.421469] esp32_spi: cmd_auth_request: Authentication request: b0:4e:26:f9:d8:a8 11 0 0 0
[   13.719742] esp32_spi: esp_cfg80211_associate: 
[   13.728233] esp32_spi: cmd_assoc_request: Association request: b0:4e:26:f9:d8:a8 11 40
[   13.742629] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   13.798119] esp32_spi: process_assoc_event: Connection status: 0
[   13.804397] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for RU, current=\x00\x00
[   13.813197] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   13.846735] esp32_spi: esp_inetaddr_event: NETDEV_UP interface eth0 ip changed to  192.168.100.230
[   13.856550] esp32_spi: cmd_set_ip_address: Invalid argument
[   13.863500] esp32_spi: esp_cfg80211_get_tx_power: 
[   13.884056] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   13.899929] esp32_spi: esp_cfg80211_get_tx_power: 
[   13.899972] dwmac-sun8i 1c30000.ethernet eth0: Too many address, switching to promiscuous
[   14.890112] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   15.890383] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   16.903248] esp32_spi: process_disconnect_event: Disconnect event for ssid HANNI-ARCHERC7 [reason:15]
[   16.912888] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=RU
[   16.957268] esp32_spi: esp_cfg80211_get_tx_power: 
[   17.000970] esp32_spi: esp_cfg80211_get_tx_power: 
[   17.021340] esp32_spi: esp_cfg80211_scan: 
[   18.754563] esp32_spi: esp_cfg80211_authenticate: 
[   18.759450] esp32_spi: cmd_auth_request: Authentication request: b0:4e:26:f9:d8:a8 11 0 0 0
[   19.025561] esp32_spi: esp_cfg80211_associate: 
[   19.030292] esp32_spi: cmd_assoc_request: Association request: b0:4e:26:f9:d8:a8 11 40
[   19.091641] esp32_spi: process_assoc_event: Connection status: 0
[   19.101392] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for RU, current=00
[   19.113792] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   19.143214] esp32_spi: esp_cfg80211_get_tx_power: 
[   19.170549] esp32_spi: esp_cfg80211_get_tx_power: 
[   19.192370] systemd-journald[116]: Time jumped backwards, rotating.
[   20.091879] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   21.085298] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   22.086186] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   23.095263] esp32_spi: process_disconnect_event: Disconnect event for ssid HANNI-ARCHERC7 [reason:15]
[   23.104704] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=RU
[   23.136243] esp32_spi: esp_cfg80211_get_tx_power: 
[   23.163843] esp32_spi: esp_cfg80211_get_tx_power: 
[   23.209537] esp32_spi: esp_cfg80211_scan: 
[   24.940335] esp32_spi: esp_cfg80211_authenticate: 
[   24.946871] esp32_spi: cmd_auth_request: Authentication request: b0:4e:26:f9:d8:a8 11 0 0 0
[   25.287590] esp32_spi: esp_cfg80211_associate: 
[   25.293424] esp32_spi: cmd_assoc_request: Association request: b0:4e:26:f9:d8:a8 11 40
[   25.348176] esp32_spi: process_assoc_event: Connection status: 0
[   25.354398] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for RU, current=00
[   25.363252] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   25.390270] esp32_spi: esp_cfg80211_get_tx_power: 
[   25.419740] esp32_spi: esp_cfg80211_get_tx_power: 
[   26.342278] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   27.342254] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   28.342264] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   29.351308] esp32_spi: process_disconnect_event: Disconnect event for ssid HANNI-ARCHERC7 [reason:15]
[   29.360908] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=RU
[   29.411166] esp32_spi: esp_cfg80211_get_tx_power: 
[   29.439736] esp32_spi: esp_cfg80211_get_tx_power: 
[   29.468122] esp32_spi: esp_cfg80211_scan: 
[   31.191767] esp32_spi: esp_cfg80211_authenticate: 
[   31.197593] esp32_spi: cmd_auth_request: Authentication request: b0:4e:26:f9:d8:a8 11 0 0 0
[   31.533665] esp32_spi: esp_cfg80211_associate: 
[   31.538295] esp32_spi: cmd_assoc_request: Association request: b0:4e:26:f9:d8:a8 11 40
[   31.599199] esp32_spi: process_assoc_event: Connection status: 0
[   31.605488] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for RU, current=00
[   31.623508] esp32_spi: process_disconnect_event: Disconnect event for ssid HANNI-ARCHERC7 [reason:7]
[   31.642593] esp32_spi: esp_cfg80211_get_tx_power: 
[   31.674678] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=RU
[   31.707188] esp32_spi: esp_cfg80211_get_tx_power: 
[   31.715178] dc1sw: disabling
[   31.718252] cpvdd: disabling
[   31.725518] vcc-1v2-hsic: disabling
[   31.750707] esp32_spi: esp_cfg80211_get_tx_power: 
[   31.783807] esp32_spi: esp_cfg80211_get_tx_power: 
[   31.807297] esp32_spi: esp_cfg80211_scan: 
[   41.531546] esp32_spi: esp_cfg80211_scan: 

root@nanopi-a64-bookworm:~# networkctl 
IDX LINK  TYPE     OPERATIONAL SETUP      
  1 lo    loopback carrier     unmanaged
  2 eth0  ether    routable    configured 
  3 wlan0 wlan     dormant     configuring

root@nanopi-a64-bookworm:~# nano /etc/wpa_supplicant/wpa_supplicant-wlan0.conf

ctrl_interface=/var/run/wpa-supplicant
ap_scan=1
network={
        ssid="HANNI-ARCHERC7"
        scan_ssid=1
        key_mgmt=WPA-PSK
        psk="abc"
}

root@nanopi-a64-bookworm:~# nano /etc/systemd/network/00-wireless.network

[Match]
Name=wlan0

[Network]
DHCP=yes
## to use static IP uncomment these instead of DHCP
# Address=192.168.1.160
Gateway=192.168.100.1
DNS=192.168.100.1
DNS=
Domains=
NTP=
NTP=

What am I doing wrong ?

mantriyogesh, 2024-07-08:
@kapilkedawat ++

kapilkedawat, 2024-07-08:
Hi @sergey-suloev, your config file is incorrect, please provide correct AP password(passpharse cannot be less than 8 characters long).

sergey-suloev, 2024-07-08:
> Hi @sergey-suloev, your config file is incorrect, please provide correct AP password(passpharse cannot be less than 8 characters long).

funny answer) of course I am using a correct password in reality. this is not the issue

kapilkedawat, 2024-07-08:
AP is sending a disconnect with reason code 15(Handshake timeout), which happens when there is passphrase mismatch. I also see EAPOL packets coming 3 times(probably EAPOL1). Please share supplicant logs.

sergey-suloev, 2024-07-08:
[wpa_supplicant.log](https://github.com/user-attachments/files/16124316/wpa_supplicant.log)

kapilkedawat, 2024-07-08:
@sergey-suloev thanks for the logs, seems like supplicant is not getting EAPOL packets.

If you have recently updated the esp code, did you also run [setup.sh](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/setup.sh) which updates RF libraries?

sergey-suloev, 2024-07-08:
> @sergey-suloev thanks for the logs, seems like supplicant is not getting EAPOL packets.
> 
> If you have recently updated the esp code, did you also run [setup.sh](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/setup.sh) which updates RF libraries?

no.. good point, I'll try it and let know

kapilkedawat, 2024-08-06:
Hi @sergey-suloev is this resolved or you are still facing this issue?