espressif/esp-hosted #427: Where can I find enumeration of the reasons for the disconnection?

ShengjieCh, 2024-07-09:
Where can I find enumeration of the reasons for the disconnection?

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#415-struct-event_station_disconn_t
[image]

mantriyogesh, 2024-07-09:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv417wifi_err_reason_t

mantriyogesh, 2024-07-09:
https://github.com/espressif/esp-idf/blob/3b8741b172dc951e18509698dee938304bcf1523/components/esp_wifi/include/esp_wifi_types.h#L78-L140

closed 2024-07-09 by mantriyogesh

ShengjieCh, 2024-07-09:
Sorry, I didn't find the enumeration of 0. What is the definition of this?
[image]

mantriyogesh, 2024-07-09:
0 shouldn't have come as such.

mantriyogesh, 2024-07-09:
The reason for disconnection is not populated currently for sta disconnected from ESP softap.
only mac address is sent to identify which station got disconnected.

We would need to add the reason in this event.
[image]

ShengjieCh, 2024-07-11:
Please let me know approximately when this feature will be supported.