espressif/esp-hosted #428: Can't use bluetoothctl  for advertise on  funtion，Bluetooth broadcast packet setting failed.

244219195, 2024-07-12:
Software operating environment：
1. linux version ：Linux version 5.4.238 (gnar@glserver) (gcc version 9.3.0 (OpenWrt GCC 9.3.0 r16847-f8282da11e)) #0 SMP Mon Apr 17 13:15:36 2023
the linux was work for openwrt .
2.BlueZ version : bluez-5.64.tar.xz
/ # bluetoothctl -v
bluetoothctl: 5.64
/ # bluetoothd -v
5.64
3.ESP32 chip : ESP32C3 
4.esp-hosted: ESP32c3 was spi connect to linux ,so I was use esp_hosted_fg ,wifi and bt funtion over spi.

this error message appears after setting it up
[image]

when i use advertise on  funtion ,phone show that Broadcast was empty
[image]

here are some hci log .

244219195, 2024-07-12:
[
[Follow_the_official_steps_of_ESP32_CFA_has_logs.zip](https://github.com/user-attachments/files/16188231/Follow_the_official_steps_of_ESP32_CFA_has_logs.zip)
](url)

244219195, 2024-07-12:
some hci log was update.please check it .

mantriyogesh, 2024-07-12:
Can you please add ESP side and host side logs:

Esp: 
idf.py monitor OR
minicom logs
from **start**

Host:
- dmesg log from bootup
- sudo btmon from loading of kernel driver
- commands associated
- possibly bluetoothd (BlueZ) debug logs

mantriyogesh, 2024-07-12:
Also send us the picture or photo how do you connect esp32-c3 to Linux.

244219195, 2024-07-12:
[image]
maybe can't see more things ,but i was open esp32 SPI log .so  i think SPI channel was ok.
[image]
[image]

mantriyogesh, 2024-07-12:
Can you please send us full logs stated in https://github.com/espressif/esp-hosted/issues/428#issuecomment-2224392347

Where is ESP32-C3 in your picture?
Is it PCB embedded?

Also please send us **full** and **textual** logs.

244219195, 2024-07-12:
I will go and retrieve the corresponding log. ESP32 C3 was on pcba

244219195, 2024-07-12:
[logs_for_spi_esp32c3.zip](https://github.com/user-attachments/files/16189325/logs_for_spi_esp32c3.zip)
@mantriyogesh 
Here are log file.
how to get possibly bluetoothd (BlueZ) debug logs?hci log not enough?

244219195, 2024-07-12:
[image]

mantriyogesh, 2024-07-12:
Please spare us some time, to get back on this.

mantriyogesh, 2024-07-12:
Your current Linux kernel version is:

Linux version 5.4.238 (aarch64)

Please check https://stackoverflow.com/questions/61935284/invalid-params-issue-when-doing-hciconfig-hci0-reset

This seems to be fixed in newer kernel, so please consider down/upgrading your kernel.
This doesn't look like ESP-Hosted bug as such, anyway.

You kernel code references:
https://elixir.bootlin.com/linux/v5.4.238/source/net/bluetooth/hci_request.c#L1617
[image]

[image]

fix 5.7.7 :
https://elixir.bootlin.com/linux/v5.7.7/source/net/bluetooth/hci_request.c#L942
[image]

244219195, 2024-07-12:
Do you have another method?down/upgrading my kernel maybe difficulty.down/upgrading blue-z version can resolve this issue?

244219195, 2024-07-12:
I will try to change my kernel code for https://elixir.bootlin.com/linux/v5.7.7/source/net/bluetooth/hci_request.c#L942
maybe it can useful.

mantriyogesh, 2024-07-12:
I think it would be better, to change and re-flash your kernel as per expected change, in case you do not wish to upgrade.
Let us know how it goes.

244219195, 2024-07-12:
Change my kernel code for https://elixir.bootlin.com/linux/v5.7.7/source/net/bluetooth/hci_request.c#L942
it can't useful.

mantriyogesh, 2024-07-12:
Although the correct way to fix this issue is to upgrade the kernel, but which seems not possible for your side.

**Disclaimer:**
1. Suggestions below are just as help and It doesn't relate to ESP-Hosted or Espressif software in any manner. Further, we would not be responsible for any damages or system instability, if any occurs.
3.  I myself did not use the 'live patching' explained below in my entire life!!

After a lot of playing with gpt, and customisations, I ended up in `live patching feature` of kernel. You yourself can dig more info, how is it done. Obviously it is **non tested** code, I will leave it to you. :

<details>
<summary>Live Patch Kernel Module for `get_adv_instance_scan_rsp_len`</summary>

## Overview

This kernel module demonstrates live patching of the `get_adv_instance_scan_rsp_len` function in the Linux kernel, using the implementation from Linux v5.7.7.

## Implementation Details
```
// SPDX-License-Identifier: GPL-2.0-or-later
#include <linux/module.h>
#include <linux/livepatch.h>
#include <net/bluetooth/hci_core.h>

// The new implementation from Linux v5.7.7
static u8 new_get_adv_instance_scan_rsp_len(struct hci_dev *hdev, u8 instance)
{
    struct adv_info *adv_instance;

    /* Instance 0x00 always set local name */
    if (instance == 0x00)
        return 1;

    adv_instance = hci_find_adv_instance(hdev, instance);
    if (!adv_instance)
        return 0;

    /* TODO: Take into account the "appearance" and "local-name" flags here.
     * These are currently being ignored as they are not supported.
     */
    return adv_instance->scan_rsp_len;
}

// Struct that describes the livepatch
static struct klp_func funcs[] = {
    {
        .old_name = "get_adv_instance_scan_rsp_len",
        .new_func = new_get_adv_instance_scan_rsp_len,
    },
    { }
};

// Struct that describes the livepatch object
static struct klp_object objs[] = {
    {
        .funcs = funcs,
    },
    { }
};

// The livepatch module descriptor
static struct klp_patch patch = {
    .mod = THIS_MODULE,
    .objs = objs,
};

static int __init livepatch_get_adv_instance_scan_rsp_len_init(void)
{
    return klp_enable_patch(&patch);
}

static void __exit livepatch_get_adv_instance_scan_rsp_len_exit(void)
{
    klp_disable_patch(&patch);
}

module_init(livepatch_get_adv_instance_scan_rsp_len_init);
module_exit(livepatch_get_adv_instance_scan_rsp_len_exit);
MODULE_LICENSE("GPL");
MODULE_INFO(livepatch, "Y");
```
## Instructions

1. **Setup Development Environment**:
   - Install Linux kernel headers and build tools for your kernel version.

3. **Create Kernel Module Source File**:
   - Copy the above code into a file named `livepatch_get_adv_instance_scan_rsp_len.c`.

5. **Create Makefile**:
   - Create a Makefile in the same directory with the following content:

     > obj-m := livepatch_get_adv_instance_scan_rsp_len.o
     > 
     > KDIR := /lib/modules/$(shell uname -r)/build
     > 
     > all:
     >     make -C $(KDIR) M=$(PWD) modules
     > 
     > clean:
     >     make -C $(KDIR) M=$(PWD) clean

6. **Build the Kernel Module**:
   - Open a terminal, navigate to the directory containing `livepatch_get_adv_instance_scan_rsp_len.c` and `Makefile`, and run `make`.

7. **Load the Kernel Module**:
   - Load the compiled module into the kernel using `sudo insmod livepatch_get_adv_instance_scan_rsp_len.ko`.

8. **Testing**:
   - Test the live patched function in a controlled environment to ensure correct behavior.

9. **Unloading the Module**:
   - To remove the module, use `sudo rmmod livepatch_get_adv_instance_scan_rsp_len`.

## Notes

- Ensure you have the appropriate permissions and understand the implications of live patching on your system's stability.

</details>

244219195, 2024-07-12:
I was use the patch ,it doesn't work .

244219195, 2024-07-12:
Can you please provide the process for successfully using BLE broadcasting, which is closer to the Linux system version I am using and the blue Z version. Use the latest SPI solution from ESP32C3. I am willing to try replacing the Bluez version. I tried to modify the kernel code, but it didn't work.So i want to change blue Z version

mantriyogesh, 2024-07-12:
As explained, if kernel code is a problem. I am curious,how you will solve kernel code issue with different BlueZ version?

244219195, 2024-07-12:
I directly replaced the version of BlueZ without modifying its code

mantriyogesh, 2024-07-12:
Please refer to https://stackoverflow.com/questions/61935284/invalid-params-issue-when-doing-hciconfig-hci0-reset
again.

Check the function definition.
If the function returns value of 0 than 1, which is handled internal by kernel itself (not exposed to outer module), how changing the input or output, through BlueZ would help, I am not sure..

244219195, 2024-07-13:
for your answer，https://stackoverflow.com/questions/61935284/invalid-params-issue-when-doing-hciconfig-hci0-reset
This seems to be fixed in newer kernel, so please consider down/upgrading your kernel.Therefore, I attempted to replace the Bluetooth related code.however ,it doesn't work.
Can you provide another suggestion？
For my Esp log,The SPI log shows that the parameter 'name' has been obtained, but why was it not broadcasted during Bluetooth broadcast.？
The Linux system runs on ARM and is connected to ESP32-C3 through SPI. The version of Blue Z-5.64 needs to support the operation of ESP32-C3. Do you need to apply some blue-Z or bluetooth patches ？Because I have used Bluetooth chips from other projects and have had similar operations before.

mantriyogesh, 2024-07-13:
If you have any success log with non ESP device with exact same environment , with same Linux with same machine, please collect exact logs and send to us.

mantriyogesh, 2024-07-13:
For our Linux systems we do not observe this issue

Also, as we are talking about ESP, do you observe this issue with pure IDF example with similar scenario?

If it fails, we will ask bluetooth team to look into it.

Esp-Hosted is simple relay. It doesn't do any processing of packets. It sends packets from one end to other. Apart from this communication, any issues arising out of kernel, bluetooth stack, etc are out of scope for us.

If you face issues in controller as stated above with esp-idf example, either you can raise the issue with esp-idf team or we pass the logs to them.

mantriyogesh, 2024-07-13:
Esp hosted code is available to you. If the tx msg is end at ESP and vice versa, ESP-Hosted scope is done. Do you see any packets corrupt in between the communication from/to ESP on the spi/sdio?

244219195, 2024-07-14:
Thank you for your reply,SPI packet not seen lost. I will continue to try debugging Linux/ESP32 code.If there is a result, I will reply.
Based on this driver, can l implement Bluetooth broadcasting, reading and writing of Bluetooth characteristic values via SPl communication, thereby achieving Bluetoothtransparent transmission functionality?
if there's no quick solution available on the Linux side, then l might consider trying this approach.

mantriyogesh, 2024-07-14:
Actually there is one more way to even verify this issue, if it is hosted specific or not.

Right now you use HCI over spi.
There is also spi+uart way, where control path and wifi goes over wifi and bluetooth uses uart.

HCI over UART doesn't even enter hosted kernel driver and directly ends in bluetooth controller of ESP chipset.

If you can configure spi+uart solution.
You might have to additionally configure extra pins for uart, change the ESP to use bluetooth controller over uart and reflash - check documentation)

You can collect HCI over spi vs HCI vs uart.

Obviously bt controller issues cannot be ruled out, but then at least scope is reduced.
But I seriously doubt if it is controller issues as such.

I hope this comment helps.

244219195, 2024-07-19:
hello,mantriyogesh
Can you provide me with an HCI log that includes the complete Bluetooth broadcast process? I use it to compare my existing HCI logs and see if there are any errors, which makes it easier for me to fix kernel related issues

244219195, 2024-07-19:
It is best to use adjacent versions of the kernel, as well as the BlueZ version. 
Thank you.

mantriyogesh, 2024-07-19:
We cannot use exact version what you used, so the dumps would be really not helpful.
If your SoC can hold multiple kernels, evaluate upgrading your kernel and test on one of the fixed kernel.

Easiest would be to rebuild and load current kernel with fix. if you want to revert, the procedure is same and everything is still in your control.

the kernel versions we have are much ahead than 5.4.x.

244219195, 2024-07-20:
I don't need the same kernel version. On my end, I need to confirm whether the HCI command sent to ESP32 is correct for analysis, and identify which stage of the command is problematic.
[image]
When I tried to modify the Bluetooth name using hcitool cmd, the Bluetooth broadcast was successful. But the commands issued by Bluez are not consistent. This is clearly not an issue with the kernel version

244219195, 2024-07-20:
In theory, when I issue the HCI command to Bluetooth, the Bluetooth should respond accordingly.
So I need to compare your normal HCI process to analyze the problem.

mantriyogesh, 2024-07-20:
Okay, I will send you all the success logs at our side.
What all the things minimally you wish to do? I will do that on my existing setup and get you all the logs.

244219195, 2024-07-22:
1. HCI log output by btmon on the host side.
2.On esp32 chip ,CONFIG_ESP_BT_DEBUG=y and then show  the esp32 log.
Here are some host step
host side:   use bluetoothctl 
1.power on
2.menu advertise
3.name esp32_test
4.back
5.advertise on

6.Enable GATT
7. Add a new service
8. add a new uuid :0x1801
9. add a new characteristic Value
10.  characteristic.Properties:READ  ,WRITE
Here are the new GATT sample
[image]

244219195, 2024-07-22:
I have made progress on my end. The Bluetooth driver repair is in linux version 5.8, not version 5.7.so I was change the kernel code .And then ,it was working .
Thank for you help .

mantriyogesh, 2024-07-22:
Thank you for the update. This info might help someone later.

If you are unblocked, can we please this issue?

244219195, 2024-07-23:
OK

closed 2024-07-23 by mantriyogesh