espressif/esp-hosted #430: Where is the release version? The latest ESP32-C6 bin file cannot be compiled by myself and I do not know how to use ESP IDF.

hangeV, 2024-07-12:
Where is the release version? The latest ESP32-C6 bin file cannot be compiled by myself and I do not know how to use ESP IDF.

mantriyogesh, 2024-07-12:
Please follow source based compilation steps documented:

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#222-source-compilation

hangeV, 2024-07-13:
> Please follow source based compilation steps documented:
> 
> https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#222-source-compilation

为什么会卡在这里呢？
[image]

mantriyogesh, 2024-07-13:
For windows, https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/setup_windows11.md

hangeV, 2024-07-13:
> For windows, https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/setup_windows11.md

我是按这个来的，可是卡在了我截图的内容上，你能编译好传上来吗？

mantriyogesh, 2024-07-13:
Let us know the issues you faced. We can resolve the issues you faced.

mantriyogesh, 2024-07-15:
Any updates @hangeV ?