espressif/esp-hosted #432: Any soloution for SDMMC for esp-hosted-fg

alireza3137, 2024-07-15:
Hi 
Is there any soloution for port ESP-hosted-fg to SDMMC protocol?

mantriyogesh, 2024-07-15:
Sorry, did not understand correctly. 
Can you please state your requirement including host details? 

Did you mean sdio (communication protocol in sdmmc , without storage)? Sdio is already supported.

alireza3137, 2024-07-15:
I need port stm32h7. But in this MCU there is SDMMC instead of SDIO. How can I import esp_hosted_fg for SDIO interface for STM32H7

mantriyogesh, 2024-07-15:
We had to rewrite the selective sdio layer from sdmmc hal layer. But this layer is for F4.
You can at least get some idea how is it done, please check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/stm32/driver/transport/sdio/sdio_ll.c

Also, we have made ESP as host which reliably works with one ESP as host and other as slave, with sdio communication.
Branch: feature/esp_as_mcu_host
Port layer:
https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/host/hosted_os_adapter.h

Port implementation for ESP as host: https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/host/port/sdio_wrapper.c

We use some IDF component, which could be simply copied for other hosts.
This branch provides bluetooth and wifi capability.

But you need some work for porting.

In past we had also created stmf4 based sdio, in master: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/Getting_started.md

alireza3137, 2024-07-15:
Thanks for your reply. 
I will check that. 
- Have you ever port LWIP and esp-hosted-fg with SDIO interface and evaluate the performance? 
Is there any difference with SPI interface?

mantriyogesh, 2024-07-16:
Yes, it is reliably working fine with ESP as host, with sdio and spi both.

Sdio being 4-bit, is a lot faster, and reached 45mbps+ in host tx and 35mbps in host rx, with iperf2 UDP.

We also introduced sdio streaming mode which even crosses 60mbps in host rx case but yet to concluded.

For spi, the porting should be more straight forward. It is full duplex spi, with mosi, miso. Depending upon chipset spi clock max, the speed vary. For 30Mhz, you should reach 18mbps-20mbps in either direction.

All these figures are for feature/esp_as_mcu_host.
Where lwip and bluetooth is integrated and super stable.

We also are introducing QSPI, with 2-bit and 4-bit mode, which is half duplex, but with 4 bit mode the throughout is much better than spi. The exact figures, I will have to check and come back.

mantriyogesh, 2024-07-16:
If you have two ESP chipsets, it would be much better to evaluate the solution without any changes, with the buses you are interested in. Then you can take a call for STM32H4.

alireza3137, 2024-07-16:
How about stm as MCU? 
Have you port LWIP with SDIO interface? 
Evaluate performance?

mantriyogesh, 2024-07-16:
Oh sorry I misunderstood.

We wish to provide, but not available yet. As for today, Porting layer needs to be done yet on your own.

Some time later, we are thinking to provide this.

mantriyogesh, 2024-07-17:
https://github.com/espressif/esp-hosted/issues/367 was similar query, they got it working, with polling. I think sdmmc interrupt was the only bit missing/remaining at stm32h7 (unsure if exist at stm32 or not)

alireza3137, 2024-07-20:
Could you send me gis commit?

mantriyogesh, 2024-07-20:
@claydonkey can you please help, if this is feasible?