espressif/esp-hosted #433: wlan0 not brought up when running "bash rpi_init.sh spi"

markmusil, 2024-07-18:


mantriyogesh, 2024-07-18:
Have tested on latest master?

markmusil, 2024-07-18:
Yes, I pulled today. I am at commit number [da21e36](https://github.com/espressif/esp-hosted/commit/da21e362d5bced2e8ed9bc4eadb9cffb1d6c1be1)

markmusil, 2024-07-18:
Which version of raspberry pi OS should I be using? Do you have a version number?

mantriyogesh, 2024-07-18:
Can you please attach full textual logs of:

- host : Dmesg from system bootup 
- ESP log

mantriyogesh, 2024-07-18:
> Which version of raspberry pi OS should I be using? Do you have a version number?

Just use master at both ESP and host side .

markmusil, 2024-07-18:
**Taken from the linux machine plugged into esp32-c6 via usb**

[complete-esp-log-jul18.txt](https://github.com/user-attachments/files/16286818/complete-esp-log-jul18.txt)
[host_rpi4_dmesg_from_boot.log](https://github.com/user-attachments/files/16286863/host_rpi4_dmesg_from_boot.log)

mantriyogesh, 2024-07-18:
Need to get RPI log from start of the boot.

markmusil, 2024-07-18:
[rpi4-boot.log](https://github.com/user-attachments/files/16286953/rpi4-boot.log)

markmusil, 2024-07-18:
Entire file is 81MB. [Here is a onedrive link instead. ](https://armadapower-my.sharepoint.com/:u:/p/mark_musil/EUzHBwDapDpEqEN70YNZZEwBWwys5cDsz484EBsNdgMaEA?e=g2Ta6c)

mantriyogesh, 2024-07-18:
Can you try FG master ESP and FG master kernel driver first?

mantriyogesh, 2024-07-19:
I am not sure, but something is really wrong even **before** using esp32 kernel module.
Can you please attach :

- `/boot/config.txt` or `/boot/firmware/config.txt`

Also, is this Raspberry Pi fresh installed? or any changes in kernel done?
Is this Rpi flashed using default image or something else, for example?

markmusil, 2024-07-19:
_Can you try FG master ESP and FG master kernel driver first?_

I will work on this. 

_Can you please attach  /boot/config.txt_

See attached

[config.txt](https://github.com/user-attachments/files/16312314/config.txt)

_ is this Raspberry Pi fresh installed?_

Yes I am using the latest raspberry pi 64 bit OS for Raspberry pi 4B with nothing else installed on the raspberry pi. 

_or any changes in kernel done?_

No. I made sure to do what is [shown on the guide here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md#11-host-software-setup) which is to say that I installed the raspberrypi-kernel-headers. 

_Is this Rpi flashed using default image or something else, for example?_

Default Raspberry pi 4 OS

markmusil, 2024-07-19:
When I try to run `bash rpi_init.sh wifi=spi bt=spi` from `esp-hosted/esp-hosted-fg/linux/host_control/` I get the following error: 

`fatal error: protobuf-c/protobuf-c.h` no such file or directory

I am going to build protobuf myself from scratch but I don't think this is what is supposed to happen.

mantriyogesh, 2024-07-19:
No no, you should not build it from scratch.

Just update the submodules recursively

mantriyogesh, 2024-07-19:
[image]

markmusil, 2024-08-08:
What do you mean by 

"_Can you try FG master ESP and FG master kernel driver first?_"

Are you suggesting that I switch to using FG? I want to use NG so that I can directly control the wifi interface using WPA_supplicant.

markmusil, 2024-08-08:
So, even esp-hosted-fg isn't working. I ran the following command

_./rpi_init.sh wifi=spi bt=spi_ 

And got an error in line 165 of rpi_init.sh "5090 segmentation fault     sudo rmod esp32_spi &> /dev/null"

Is there a possibility of doing a video call to address this directly? Or would you prefer to just continue with this ticket?

mantriyogesh, 2024-08-09:
Hello @markmusil ,

I regret the problem you faced.

Can you please send the full textual logs when you face this issue?
- Host: dmesg from system bootup
- Slave: `idf.py monitor` or uart/minicom logs ?

markmusil, 2024-08-09:
[idfPYmonitor.txt](https://github.com/user-attachments/files/16563606/idfPYmonitor.txt)

[Aug9_dmesg_raspi.txt](https://github.com/user-attachments/files/16563713/Aug9_dmesg_raspi.txt)

mantriyogesh, 2024-08-12:
1. Logs related
The `Aug9_dmesg_raspi.txt` log is not from start of the loading of Linux, so it is hard to understand where the issue started.

The log starts at `[562.370665]` time, I think your log is getting huge (due to crashes) and resulting in losing the log.
Easy way, create a new terminal, and run on system bootup:
```
$ sudo dmesg -c
$ sudo dmesg -w # tail and follow the dmesg
```
**before loading the esp32 kernel module**

2. Wrong firmware?
[image]
the firmware flashed seems to be SDIO, but you intend to use SPI. You can refer documentation to flash spi firmware.

3. spidev was not removed:
From your log, I see module linked in doesn't have `esp32_spi`. Also, there is entry of `spidev`
[image]

The spidev was expected to be unloaded/removed for your /dev/spidev._<bus_instance>_._<cs_instance>_ in the setup step at :
https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_ng/host/rpi_init.sh#L168-L173

Can you please check why spidev was not removed?

markmusil, 2024-08-12:
Would you recommend using a different raspberry pi?

Mark Musil | Embedded Systems Engineer

P (740) 703-2110

***@***.***

[cid:4d5847c6-8e33-47aa-a4f7-675bdcdfcba7]

  1.  Wrong firmware?
Screenshot.2024-08-12.at.10.23.05.AM.jpg (view on web)<https://github.com/user-attachments/assets/cfcbb14a-9244-4188-9965-7a490608071b>

  2.  spidev was not removed:
Screenshot.2024-08-12.at.10.20.02.AM.jpg (view on web)<https://github.com/user-attachments/assets/2636ebe6-34f4-4186-b628-44f8e7949925>

https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_ng/host/rpi_init.sh#L168-L173