espressif/esp-hosted #437: FG mode WiFi connect block

lwj850533527, 2024-07-26:
Hi mantriyogesh,

Feedback on an issue with the NG mode: In the code on the ESP32 side, the function req_connect_ap_handler will block when SSID errors, or password errors.
If there are more than one AP with the same SSID, the blocking time will reach 1 minute.
Is there any good solution?

[image]

mantriyogesh, 2024-07-26:
Hello @lwj850533527 ,

Thank you for reporting this.

Yes, this we can just keep as two or three. If any delays in re attempting connect, can as well lowered, if any.

We will take this up in upcoming task list.

lwj850533527, 2024-07-31:
@mantriyogesh 
It seems that a background task should be added.
Report connection results through events，because it is a blocked interface

mantriyogesh, 2024-07-31:
Can you please elaborate?

Slave side or host?
Blocked interface, do you mean serial at slave?

In general, the events and responses are handled through pserial task, a dedicated task with is responsible to serialise and send the msg to sdio/spi task.

Sdio/spi task queues (see prio-quueue) would be blocking nature, as they would be cleared only when transaction is done on bus.

lwj850533527, 2024-07-31:
Host side.
When I tried to connect to WiFi with the wrong password, all other interfaces returned timeout because it block more than 1 minute.

Blocked interface, I mean CTRL_REQ_CONNECT_AP at host.

mantriyogesh, 2024-07-31:
So 

> we can just keep as two or three attempts to connect at the max. If any delays present in re attempts also need to be lowered (if any)

Would potentially resolve your problem?

For now, 
https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L45
You can change this to `0`

And change this
https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L49

to
Say, 
`(20*TIMEOUT_IN_SEC)`

Would fix the problem for you

mantriyogesh, 2024-07-31:
Also, timeout need to be increased as per above time taken at

https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L733

req.cmd_timeout_sec = 60; /* don't timeout early, need to time as per slave wait*/

closed 2024-07-31 by lwj850533527