espressif/esp-hosted #439: 使用NUCLEO-F412ZG 连接 ESP32C6 不成功

renhil, 2024-07-29:
[esp32c6.txt](https://github.com/user-attachments/files/16407251/esp32c6.txt)
[stm32f412.txt](https://github.com/user-attachments/files/16407252/stm32f412.txt)
是使用同一个版本的库搞的
为啥不行呢？

mantriyogesh, 2024-07-29:
[image]

`No AP available as SSID:'MyWiFi'`

Change config to use SSID which is broadcasted by external AP/router.

renhil, 2024-07-29:
SdioDriverInit:409 CMD5 timeout!

SdioDriverInit:417 CMD5 timeout!

初始化的这两行 timeout 没有关系吗？

mantriyogesh, 2024-07-29:
Check if you have external pull-ups and wire connections are correctly connected.
The commands can timeout,
- If there is loose wire connections
- SDIO is sensitive protocols and should not use with jumper cables due to signal integrity issues.
- Smaller you jumper cables to 3-4 CM
- external pull-ups are mandatory.

Check porting guide for common issues. This porting guide is for Linux but the hardware concepts around SDIO remains the same. https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md

renhil, 2024-07-29:
好的 非常感谢~