espressif/esp-hosted #440: esp-hosted-ng: WiFi SPI + BLE HCI - Bluetooth interface doesn't go up

QuiGonRazor, 2024-07-29:
- ESP32-C3
- esp-hosted-ng firmware + kernel linux driver
- sha: 91b0933
- configuration: WiFi SPI + BT HCI (H4)

Hello all,
I've correctly setup an ESP32-C3 with NG fw.
It correctly talks, brings up wlan0 interface and it works as expected (it connects via WiFi + communicates via Internet)

I'm now interested in having the BLE interface working. I've tried with both WiFi and BLE via SPI and all worked fine.
Now I'm trying to switch to BLE HCI (H4). I followed the guide into the repo and connected all related GPIOs (CST, RTS, BLE RX, BLE TX).

It seems not working. When I launched the command `hciattach [PORT] 115200 any flow` my system tries to bring up the BLE interface, failing miserably. Using dmesg I can clearly see that hci bring up commands are failing due to timeout.

Can you help me please? What am I missing?

mantriyogesh, 2024-07-29:
Bluetooth over uart, is actually pass through for ESP-Hosted.

Have you configured your host uart pins to use bluetooth?

[image]

If you configure your Linux gpios in device tree or any other way, it should work.

Make sure you connect the uart pins correctly.

Check uart setup:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md#23-sdiospi--uart-configuration

Here instead of Raspberry Pi setup, you need to configure for your host. ESP side setup steps can be referred from documentation.

mantriyogesh, 2024-07-29:
ESP side controller directly supports standard HCI interface.

Check if you have configured ESP correctly from setup document. Also check the HCI over UART baud rate used in IDF.py menuconfig.
Is it two pin or four pin uart, accordingly flow control enabled or disabled in both ESP and host,

Pin connection are correct (rx to tx and vice versa, rts to CTS and vice versa)

mantriyogesh, 2024-07-29:
Once you setup Linux uart driver in HCI mode, it should just work. These messages are pass through for ESP-Hosted, meaning the HCI over UART messages from Linux directly end from Linux driver to ESP controller. These messages do not enter hosted kernel module.

QuiGonRazor, 2024-07-29:
I reply you per steps
- I've checked the same guide posted by you
- I have the device tree correctly set 
`
&uart5{
    uart-has-rtscts;
    pinctrl-names = "default", "sleep";
    pinctrl-0 = <&uart5_pins_mx>;
    pinctrl-1 = <&uart5_sleep_pins_mx>;
    status = "okay";
};
`

This uart I'm 100% works (used before for other meanings)
- I've changed the default baud rate to 115200
- I should have connected all as expected, tomorrow I'll give a further look on the setup

mantriyogesh, 2024-07-29:
As fas I remember ESP32 had default bad rate set to  921600, unless explicitly changed.

Also for c3, there was no default config from esp-idf, so we had added manual option that time. (Setup doc should be covering this, also you can search manually for 'UART.*BAUD' in sdkconfig generated after idf.py menuconfig. I will also check if any default option introduced and being used newly in esp-idf.

mantriyogesh, 2024-07-31:
Could you try with 921600?

```
$ hciconfig -a
$ sudo killall hciattach
$ sudo hciattach -s <baud_rate> <serial_device_name> /dev/serial0 any <baud_rate> flow
$ hciconfig -a
```
<baud_rate> : 921600
<serial_device_name> : uart serial device exposed through host device tree config, to use for `hci over uart`

QuiGonRazor, 2024-07-31:
Hello,
sorry for being late.

Yesterday I tried with a second devkit I have and it worked correctly (setting HCI baud rate to 115200), same flashed binary

Today I'll try with the first one flashing it again with the default baud rate you've suggested

mantriyogesh, 2024-07-31:
oh okay. Can you share your sdkconfig generated while building, for chipset you had experienced the problem?

QuiGonRazor, 2024-07-31:
Yeah sure, here attached 
I had to add txt extension to be able to upload it

[sdkconfig.txt](https://github.com/user-attachments/files/16437283/sdkconfig.txt)

mantriyogesh, 2024-07-31:
considering 

```
CONFIG_EXAMPLE_HCI_UART_BAUDRATE=115200
```

Edited https://github.com/espressif/esp-hosted/issues/440#issuecomment-2259588979 to correct baud rate for your case.

QuiGonRazor, 2024-07-31:
Hello, 
restoring the BAUDRATE to the default one (921600) works for both the devkits.

It seems all is working good.