espressif/esp-hosted #441: NG mode, WiFi connected, signal poll get rssi not refresh

lwj850533527, 2024-07-31:
Hi mantriyogesh,

Feedback on an issue with the NG mode.
After connected the AP, use "wpa_cli signal_poll" get rssi, but it doesn't seem to refresh
When use  "wpa_cli scan",  signal level refresh, but it doesn't refresh to signal_poll

[image]

mantriyogesh, 2024-07-31:
@kapilkedawat , PTAL.

lwj850533527, 2024-08-08:
Hi mantriyogesh
Hi kapilkedawat 
@mantriyogesh , @kapilkedawat 
Due to the inability to report signal rssi,  I tried using the scan interface to perform regular scans.
To prevent signal disconnection, I need to limit the scanning frequency band to be consistent with the connected AP.

"wpa_cli SCAN freq=2457", The instruction sent to wpa_supplicant has taken effect, but the slave still performs scanning of all frequency bands
[dmesg.txt](https://github.com/user-attachments/files/16536800/dmesg.txt)
[image]

kapilkedawat, 2024-08-26:
@lwj850533527 fixed in latest master

closed 2024-08-27 by lwj850533527