espressif/esp-hosted #442: NG mode, WiFi connected Multiple AP same ssid, connection use for a long time

lwj850533527, 2024-07-31:
Hi mantriyogesh,

Feedback on an issue with the NG mode.The scene is connecting multiple APs same ssid. Sometimes the connection use for a long time.
As shown in the figure，“06:f9:f8:1f:cd:fd”,“06:f9:f8:14:69:c7”,"06:f9:f8:14:66:4c"，these three are APs with the same SSID and password.
14:35:12, it initiated an AP connection
14:36:46 it connect success
The connection cost 1.5 minutes

From packet capture data, ESP32 sends authentication to an incorrect MAC address.For example: 14:35:14, kernel log it should sents authentication to “06:f9:f8:1f:cd:fd”, but cat packet it sents to “06:f9:f8:14:69:c7”
From esp32c3 log, 14:35:14, it seems received an authentication response, but the processing result is abnormal

it is kernel log
[image]

it is esp32c3 log
[image]
[image]
[image]
[image]

it is omnipeek air interface packet capture data
[image]

mantriyogesh, 2024-07-31:
@kapilkedawat , PTAL.

lwj850533527, 2024-08-07:
@mantriyogesh 
Is there any progress on this issue

kapilkedawat, 2024-08-07:
Hi @lwj850533527, could you please share above dmesg, sniffer capture and firmware logs ?

lwj850533527, 2024-08-07:
@kapilkedawat 
dmesg not saved， only screenshots available
[image]
[sniffer capture.zip](https://github.com/user-attachments/files/16520732/sniffer.capture.zip)
[firmware logs.txt](https://github.com/user-attachments/files/16520734/firmware.logs.txt)

lwj850533527, 2024-08-14:
@kapilkedawat
I found some related settings. Is the MAC address incorrect due to lack of settings
bool bssid_set;
whether set MAC address of target AP or not. Generally, station_config.bssid_set needs to be 0; and it needs to be 1 only when users need to check the MAC address of the AP

[image]
[image]

kapilkedawat, 2024-08-26:
@lwj850533527 please recheck in latest mater.

closed 2024-08-27 by lwj850533527