espressif/esp-hosted #443: NG mode, WiFi connected hide ssid failed

lwj850533527, 2024-07-31:
Hi mantriyogesh,

Feedback on an issue with the NG mode.
I want connect to hide ssid AP.
Set the scan_ssid = 1, wpa_supplicant can scan the ap, but can't connected.
Does it require any other operations

wpa_cli scan log:
[image]

esp32 log
[image]

kernel log
[image]

wpa_supplicant.conf
[image]

mantriyogesh, 2024-07-31:
@kapilkedawat, PTAL

lwj850533527, 2024-08-03:
Hi mantriyogesh,
I found some related settings

[image]
[image]

mantriyogesh, 2024-08-05:
Yes, let we verify/fix this.
@kapilkedawat ++

lwj850533527, 2024-08-13:
Hi mantriyogesh,
Hi kapilkedawat,
I try to set up show_hidden to true, it seems to have already searched for this AP.
But in the end, the esp_wifi_connect interface returned a failure.
Do I need to make any other settings
[image]
[image]
[image]

kapilkedawat, 2024-08-13:
Hi @lwj850533527 , please wait we will be fixing this.

kapilkedawat, 2024-08-26:
Hi @lwj850533527 fixed in https://github.com/espressif/esp-hosted/commit/8445046094e7a09afe6dd4d3494801b472a85f73

closed 2024-08-27 by lwj850533527