espressif/esp-hosted #444: 按照开箱方式STM32F469+ESP32C6报27error

renhil, 2024-08-01:
[image]
这正常吗？

mantriyogesh, 2024-08-01:
Hello @renhil ,

STM32 needs a resync from latest code, alternatively, you can think of using branch: https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host where ESP as host is demonstrated

renhil, 2024-08-01:
不能正常工作呢
串口信息只接收到：
ESP-Hosted for ESP32-C2/C3/C6/S2/S3

esp32c6的监控信息是跟指导书一致的。

mantriyogesh, 2024-08-01:
stm32 code is out of sync right now, as said earlier.
We are fixing this on master.

We are trying to get the MCU branch on https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host, which is easy for porting and any mcu can redefine porting layer for their os.

renhil, 2024-08-01:
好的 我测试完你之前推荐那两个方案 就转战你说的这个新方案！

renhil, 2024-08-05:
mantriyogesh, 你好~
我现在用你说的这个新方案：
 https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host
但是遇到了问题：
从机编译、下载都正常；
主机却无法编译：
错误为：Unknown CMake command "idf_component_register"
：：：：
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:56 (idf_component_register):
  Unknown CMake command "idf_component_register".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.24)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.
请指示......

mantriyogesh, 2024-08-05:
> should be added at the top of the file. The version specified may be lower
if you wish to support older CMake versions for this project. For more
information run "cmake --help-policy CMP0000".
This warning is for project developers. Use -Wno-dev to suppress it.

Point taken. Will make such changes.

> Unknown CMake command "idf_component_register"

Actually we demonstrate ESP as MCU host. Which MCU right now are you using?
**If you use ESP chipset**, please set up IDF from https://github.com/espressif/esp-idf?tab=readme-ov-file#setting-up-esp-idf.
In this, you clone the https://github.com/espressif/esp-idf , update submodules recursively and run `./install.sh` and `. ./export.sh` on Linux platform, you can go back & resume the esp-hosted documentation.
For non-Linux systems also ESP-IDF is supported, the steps could be followed from GitHub, for now.

Again, apologies, this is a documentation bug, which should have clearly mentioned, We will fix this as soon as possible.
Our README.md is already in rigorous changes and new documentation for this branch would be out very soon, with all these fixes.

renhil, 2024-08-09:
Actually we demonstrate ESP as MCU host. Which MCU right now are you using?
我用的是esp32s3.
set up IDF from https://github.com/espressif/esp-idf?tab=readme-ov-file#setting-up-esp-idf.
In this, you clone the https://github.com/espressif/esp-idf , update submodules recursively and run ./install.sh and . ./export.sh
我按照你这方法配置了，但是错误还是存在：
CMake Error at CMakeLists.txt:56 (idf_component_register):
  Unknown CMake command "idf_component_register".
这个错误怎么处理呢？

mantriyogesh, 2024-08-09:
Your IDF is not setup correctly.
Setup IDF first, please refer:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/index.html

renhil, 2024-08-13:
我的IDF安装配置应该是正确的：
我进行hello_world编译运行正常。
我有疑问：
hello_world工程有一个CMakeLists.txt，
里头有main文件夹，里头也有个CMakeLists.txt。
里头的CMakeLists.txt，有idf_component_register。
这是可以正常编译运行的。
esp-hosted项目中，slave文件夹也是能正常运行的。里头也是有main文件夹，
里外也都有一个CMakeLists.txt文件。
但是，host文件夹里没有main也没有CMakeLists.txt。
我在idf_component.yml文件中已经添加了：
dependencies:
  idf:
    version: ">=5.3"
  espressif/esp_hosted:
    version: "0.0.9"
    pre_release: true
  espressif/esp_wifi_remote:
    version: "0.3.0"
但是没起作用。
我是不是缺少了什么文件？
[image]

还是，需要其它步骤？有指导说明吗？

mantriyogesh, 2024-08-13:
>

There are two CMakeList.txt types:

### Project-Level `CMakeLists.txt`

- **Purpose**: Manages global project settings and includes subdirectories.
- The `slave` directory is a complete project directory for the slave ESP.

### Component-Level `CMakeLists.txt`

- **Purpose**: Manages settings specific to individual components or modules. It defines the build settings for specific components or modules of the project, including source files, include directories, and dependencies.

When integrating `esp-hosted` component, you can refer to the [ESP-Hosted GitHub repository](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host?tab=readme-ov-file#adding-required-components) for instructions. This is different from master branch, which targets ESP as slave and ESP as host. The project’s `CMakeLists.txt` will handle fetching and including the `esp-hosted` component as described in the [integration steps](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host?tab=readme-ov-file#adding-required-components).

You can use any ESP-IDF Wi-Fi example, like [iperf](https://github.com/espressif/esp-idf/tree/release/v5.3/examples/wifi/iperf) on host ESP, using the steps mentioned above.

You can optionally remove 'pre_release: true' line, as it is no more 'pre-release', but a stable component. We will update the documentation accordingly.

Also, we are going to bump the revision to 1.0.0 in upcoming weeks. For now, you can use:

```
dependencies:
  idf:
    version: ">=5.3"
  espressif/esp_hosted:
    version: ">=0.0.9"
  espressif/esp_wifi_remote:
    version: ">=0.3.0"
```

renhil, 2024-08-13:
好的 谢谢~ 我稍后测试你说的方案。
我用另外一个方案又遇到了一个问题：
我特意买了你推荐的这一整套：
[image]
硬件完全一样。
编译还是出现：
[107/108] Generating binary image from built executable
esptool.py v4.8.dev4
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated E:/esp32-stm32f412/esp-hosted/esp_hosted_fg/esp/esp_driver/network_adapter/build/bootloader/bootloader.bin
[108/108] cmd.exe /C "cd /D E:\esp32-stm32f412\esp-hosted\...sp_driver/network_adapter/build/bootloader/bootloader.bin"
Bootloader binary size 0x6880 bytes. 0x780 bytes (7%) free.
[1017/1019] Linking CXX executable network_adapter.elf
FAILED: network_adapter.elf
cmd.exe /C "cd . && D:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=E:/esp32-stm32f412/esp-hosted/esp_hosted_fg/esp/esp_driver/network_adapter/build/network_adapter.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\network_adapter.elf.rsp -o network_adapter.elf  && cd ."
D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: network_adapter.elf section `.iram0.text' will not fit in region `iram0_0_seg'
D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: IRAM0 segment data does not fit.
D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 3496 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
HINT: The applications static IRAM usage is larger than the available IRAM size.
For more information on how to reduze IRAM usage run 'idf.py docs -sp api-guides/performance/ram-usage.html#optimizing-iram-usage'
ninja failed with exit code 1, output of the command is in the E:\esp32-stm32f412\esp-hosted\esp_hosted_fg\esp\esp_driver\network_adapter\build\log\idf_py_stderr_output_17376 and E:\esp32-stm32f412\esp-hosted\esp_hosted_fg\esp\esp_driver\network_adapter\build\log\idf_py_stdout_output_17376
！！！
HINT: The applications static IRAM usage is larger than the available IRAM size.
！！！
不够内存的情况，是哪里设置不对吗？

mantriyogesh, 2024-08-13:
okay, i think you are using latest IDF.

Alternatives could be:
1. Use older IDF, say 5.0
2. **Preferred** solution: We had changed the partition size in ESP32-C6([here](https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_fg/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2#L27-L28))
Need similar  (not exactly same) changes to be done for ESP32 case.
such that default partition with 'factory' is removed and partitions, 
`nvs`, `otadata`, `phy_init`, `ota_0`  and `ota_1`
are only available. the starting address for **non-app** should be retained same as that of previous.

mantriyogesh, 2024-08-13:
if you are not able to proceed on (2), please wait till tomorrow, I will correct and push/provide you the fix.

renhil, 2024-08-14:
please wait till tomorrow, I will correct and push/provide you the fix.
好的 搞好了 请跟我说一声。
谢谢~

mantriyogesh, 2024-08-14:
Can you please attach sdkconfig you use?

mantriyogesh, 2024-08-14:
for now, to unblock you, can you please disable bluetooth and try again?
[image]

We are working on this and fix asap.

renhil, 2024-08-15:
esp32+esp32s3
[image]
握手和和数据准备好都用了中断 会报错？
我应该怎么改？

I (474) os_wrapper_esp: GPIO [26] configuring as Interrupt
I (474) spi: No element in any queue found
I (479) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1
I (479) main_task: Started on CPU0
I (493) os_wrapper_esp: GPIO [4] configuring as Interrupt
I (503) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1
E (512) gpio: gpio_install_isr_service(502): GPIO isr service already installed
I (521) main_task: Calling app_main()

mantriyogesh, 2024-08-15:
Can you share the photo of your setup?

You need to connect SPI lines, Handshake, DataReady and Resetpin in between slave and host.
In a new directtory, outside esp-hosted, Flash the slave from https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host?tab=readme-ov-file#preparing-slave-for-esp-hosted. once you flash the slave, flash the host and hardware connections are okay, the ESP-Hosted would start.

renhil, 2024-08-16:
是的 我测试成功了！
E (512) gpio: gpio_install_isr_service(502): GPIO isr service already installed
这个报错好像没有关系。
非常感谢~~~

mantriyogesh, 2024-08-16:
It's actually, two times isr registration is done for (although, for two different GPIOs), ignore this error as if now.
May we know the iperf udp performance you get with this setup?

Which is slave and which is host by the way?

renhil, 2024-08-16:
我还没测过udp只测了tcp:
用的是ESP32(host) + ESP32S3(slave)
[image]
[image]

mantriyogesh, 2024-08-16:
Okay. I think you had connected with wires. If you connect with PCB, you can increase your clock as well.

Also there is QSPI solution available, which would bump the throughout much higher.

renhil, 2024-08-16:
STM32F412 + ESP32 方案：
log:
sdioclk=50000000Hz
req_freq=10000000Hz
out_freq=10000000Hz
div=3
SdioDriverInit:409 CMD5 timeout!
SdioDriverInit:417 CMD5 timeout!
Relative Card Address: 0x0001
Card selected! RESP1_00001e00
STM32ReadReg:735 CMD52 timeout!
STM32ReadReg:735 CMD52 timeout!
STM32ReadReg:735 CMD52 timeout!
Use 4bit bus width
STM32WriteReg:718 CMD52 timeout!
我配的st clock：
[image]
我设的esp32 sdio clock：
[image]
st gpio 有上拉：
[image]
通讯不上？请问哪里出问题了？

renhil, 2024-08-16:
[image]

renhil, 2024-08-16:
原来是 GND 没连
本以为共用一个USB Hub 就不用连地了
实际上
[image]
这 GND 不能少！

renhil, 2024-08-16:
sdioclk=50000000Hz
req_freq=10000000Hz
out_freq=10000000Hz
div=3
SdioDriverInit:409 CMD5 timeout!
SdioDriverInit:417 CMD5 timeout!
Relative Card Address: 0x0001
Card selected! RESP1_00001e00
Use 4bit bus width
Slave Initialization completed
ESP-Hosted for ESP32
event packet type
Received INIT event from ESP peripheral
EVENT: 2
EVENT: 0
priv capabilty 
capabilities: 0x1
Features supported are:
	 * WLAN
EVENT: 3

priv test raw tp

EVENT: 4
Unsupported tag in event
ESP board type is : 0 
Received INIT event
Base transport is set-up
+-----------------------------------+-------------------------------------------+
|           Parameters              |             Values                        |
+-----------------------------------+-------------------------------------------+
|      INPUT__OPERATING_MODE        |             STATION+SOFTAP                |
|      INPUT_GET_AP_SCAN_LIST       |             yes                           |
+-----------------------------------+-------------------------------------------+
|            SOFTAP                 |                                           |
+-----------------------------------+-------------------------------------------+
|        INPUT_SOFTAP__SSID         |             ESPWifi                       |
|       INPUT_SOFTAP_PASSWORD       |             ESPWifi@123                   |
|      INPUT_SOFTAP_BANDWIDTH       |             HT40                          |
|       INPUT_SOFTAP_CHANNEL        |             1                             |
|      INPUT_SOFTAP_ENCRYPTION      |             WPA2_PSK                      |
|      INPUT_SOFTAP_MAX_CONN        |             4                             |
|     INPUT_SOFTAP_SSID_HIDDEN      |             no                            |
|       INPUT_SOFTAP_SRC_IP         |             192.168.2.1                   |
|      INPUT_SOFTAP_ARP_DEST_IP     |             192.168.2.22                  |
+-----------------------------------+-------------------------------------------+
|            STATION                |                                           |
+-----------------------------------+-------------------------------------------+
|        INPUT_STATION__SSID        |             MyWifi                        |
|       INPUT_STATION_BSSID         |                                           |
|   INPUT_STATION_IS_WPA3_SUPPORTED |             no                            |
|      INPUT_STATION_PASSWORD       |             MyWifi@123                    |
|       INPUT_STATION_SRC_IP        |             192.168.1.233                 |
|     INPUT_STATION_ARP_DEST_IP     |             192.168.1.203                 |
+-----------------------------------+-------------------------------------------+
1969-12-31 23:59:59 >  App EVENT: ESP INIT
timed out while reading!!!
sdio_driver_read_bytes 64 CMD53 error
Number of available APs is 15
0) ssid "Xiaomi_E511" bssid "d4:35:38:75:1f:46" rssi "-34" channel "11" auth mode "3" 
1) ssid "Hi-Redmi-VPN" bssid "a4:a9:30:50:93:f6" rssi "-37" channel "11" auth mode "3" 
2) ssid "" bssid "aa:a9:30:50:93:f6" rssi "-37" channel "11" auth mode "0" 
3) ssid "Xiaomi_FA84" bssid "4c:c6:4c:53:c6:a6" rssi "-39" channel "10" auth mode "4" 
4) ssid "Hi-vpn" bssid "b8:3a:08:00:8b:ca" rssi "-43" channel "3" auth mode "3" 
5) ssid "ZTE_5GCPE_96FB" bssid "b0:0a:d5:b9:96:fb" rssi "-52" channel "5" auth mode "3" 
6) ssid "Hi-HUAWEI-F31FP2" bssid "02:e4:1d:a6:c0:c2" rssi "-58" channel "11" auth mode "3" 
7) ssid "" bssid "9c:09:71:e6:88:90" rssi "-63" channel "1" auth mode "3" 
8) ssid "TP-LINK_CC7F" bssid "80:8f:1d:9a:cc:7f" rssi "-63" channel "6" auth mode "0" 
9) ssid "Hi-HUAWEI-鍑岄渼-EE14OA" bssid "e2:da:91:ac:b0:d0" rssi "-67" channel "6" auth mode "3" 
10) ssid "" bssid "e2:da:91:ac:b0:d5" rssi "-68" channel "6" auth mode "3" 
11) ssid "涓€灞遍珮" bssid "68:dd:b7:31:22:e4" rssi "-78" channel "11" auth mode "4" 
12) ssid "ChinaNet-jZkK" bssid "d8:a8:c8:e5:25:78" rssi "-79" channel "8" auth mode "4" 
13) ssid "" bssid "c2:f9:e4:73:82:c2" rssi "-84" channel "6" auth mode "3" 
14) ssid "ChinaNet-G7V4" bssid "6c:d2:ba:83:77:c8" rssi "-85" channel "8" auth mode "4" 
Station mode: ssid: MyWifi passwd MyWifi@123 
timed ou
ctrl lib error[0] in sem of timeout[30]
Response not received
Failed to connect with AP 
Station mode: ssid: MyWifi passwd MyWifi@123 

timed out while reading!!!
sdio_driver_read_bytes 64 CMD53 error
wifi mode is set
mac address is 48:e7:29:a3:a6:60
Failure resp/event: possibly precondition not met
ctrl lib error[0] in sem of timeout[30]
Response not received
Failed to connect with AP 
Station mode: ssid: MyWifi passwd MyWifi@123 
timed out while reading!!!
sdio_driver_read_bytes 64 CMD53 error
wifi mode is set
mac address is 48:e7:29:a3:a6:60
Failure resp/event: possibly precondition not met
ctrl lib error[0] in sem of timeout[30]
Response not received
Failed to connect with AP 
Station mode: ssid: MyWifi passwd MyWifi@123 
timed out while reading!!!
sdio_driver_read_bytes 64 CMD53 error
wifi mode is set
mac address is 48:e7:29:a3:a6:60
Failure resp/event: possibly precondition not met
ctrl lib error[0] in sem of timeout[30]
Response not received
Failed to connect with AP 
Maximum retry done to connect with AP
timed out while reading!!!
sdio_driver_read_bytes 64 CMD53 error

离成功又近了一步！！！

renhil, 2024-08-16:
遇到问题：
STA 模式 打印信息显示已经连接到了我的本地路由器
但是实际上我的路由器并没有显示有esp连接上来：
[image]
[image]
[image]

AP模式也有问题：
idf monitor 里有显示我电脑连上和离开
我也设置了静态IP：192.168.2.22
但ping不通：192.168.2.1

请问哪里出问题了呢？

mantriyogesh, 2024-08-16:
sdio over stm32 may have some issues.
STM32 HAL layer doesn't directly support the SDIO. we had to introduce the HAL layer tweaking to get the SDIO only out of SDMMC HAL driver. If there are any changes in latest driver, it might have gone out of sync.

We might need good time to reply back for sdio related issues on STM32.

Regarding not connected, is your program **crashed** or stuck on exception handler? Can check through debug mode, where current PC is pointing.

Also it is worth to hook disconnect events, to get host notified for any disconnection from router.

renhil, 2024-08-16:
程序没有崩溃：
能进主循环 一直运行
[image]

mantriyogesh, 2024-08-20:
Did you get ahead @renhil ?

renhil, 2024-09-11:
[image]
又遇到问题了……

mantriyogesh, 2024-09-11:
It looks like your transport is not reliable. what is the spi frequency and spi modes used at both places?

renhil, 2024-09-12:
用的是 MODE 3  速度10MHz
好像确实是通讯不稳定  现在已经可以了
谢谢~