espressif/esp-hosted #446: 使用F469开机方案 能正常连接 ping不成功 为啥

renhil, 2024-08-02:
变量保持没变：
[image]
WiFi连接成功：
[image]
IP设置：
[image]
[image]
ping失败：
[image]
为何？

mantriyogesh, 2024-08-07:
can you please send full logs at host and slave?
Textual logs only.

HP4LEVI, 2024-08-09:
你好，我也遇到了相同的问题，我使用的STM32H743与ESP32S3的组合方式。STM32与ESP32的运行日志如下所示：
ESP32部分:
I (27) boot: ESP-IDF v5.2-dirty 2nd stage bootloader
I (27) boot: compile time Jul 23 2024 11:24:19
W (27) boot: Unicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000        
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000        
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000        
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (114) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=2a330h (172848) map
I (158) esp_image: segment 1: paddr=0003a358 vaddr=3fc9a600 size=043e4h ( 17380) load
I (162) esp_image: segment 2: paddr=0003e744 vaddr=40374000 size=018d4h (  6356) load
I (165) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8b240h (569920) map
I (274) esp_image: segment 4: paddr=000cb268 vaddr=403758d4 size=14c2ch ( 85036) load
I (302) boot: Loaded app from partition at offset 0x10000
I (302) boot: Disabling RNG early entropy source...
I (314) cpu_start: Unicore app
I (314) cpu_start: Single core mode
I (323) cpu_start: Pro cpu start user code
I (323) cpu_start: cpu freq: 160000000 Hz
I (323) cpu_start: Application information:
I (326) cpu_start: Project name:     network_adapter
I (332) cpu_start: App version:      v5.2-dirty
I (337) cpu_start: Compile time:     Jul 24 2024 09:33:58
I (343) cpu_start: ELF file SHA256:  521896ae0...
I (348) cpu_start: ESP-IDF:          v5.2-dirty
I (354) cpu_start: Min chip rev:     v0.0
I (358) cpu_start: Max chip rev:     v0.99
I (363) cpu_start: Chip rev:         v0.1
I (368) heap_init: Initializing. RAM available for dynamic allocation:
I (375) heap_init: At 3FCA2E60 len 000468B0 (282 KiB): RAM
I (381) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (387) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (393) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (401) spi_flash: detected chip: generic
I (404) spi_flash: flash io: dio
W (408) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (422) sleep: Configure to isolate all GPIO pins in sleep state
I (428) sleep: Enable automatic switching of GPIO sleep configuration
I (435) coexist: coex firmware version: 77cd7f8
I (441) coexist: coexist rom version e7ae62f
I (446) main_task: Started on CPU0
I (446) main_task: Calling app_main()
I (446) NETWORK_ADAPTER: *********************************************************************
I (456) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5 

I (466) NETWORK_ADAPTER:                 Transport used :: SPI only              

I (476) NETWORK_ADAPTER: *********************************************************************
I (486) NETWORK_ADAPTER: Supported features are:
I (496) NETWORK_ADAPTER: - WLAN over SPI
I (496) ESP_BT: - BT/BLE
I (496) ESP_BT:    - HCI Over SPI
I (506) ESP_BT:    - BLE only
I (506) NETWORK_ADAPTER: capabilities: 0xe8
I (516) BLE_INIT: BT controller compile version [b877d66]
I (516) BLE_INIT: Bluetooth MAC: f4:12:fa:d7:9c:3e
I (526) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
I (566) NETWORK_ADAPTER: ESP Bluetooth MAC addr: f4:12:fa:d7:9c:3e
[... 238 lines cut ...]
|       INPUT_STATION_BSSID         |                                           |
|   INPUT_STATION_IS_WPA3_SUPPORTED |             no                            |
|      INPUT_STATION_PASSWORD       |             1234qwer                      |
|       INPUT_STATION_SRC_IP        |             192.168.0.233                 |
|     INPUT_STATION_ARP_DEST_IP     |             192.168.0.11                  |
+-----------------------------------+-------------------------------------------+
1969-12-31 23:59:59 >  App EVENT: ESP INIT
Number of available APs is 7
0) ssid "Gannz_2.4G" bssid "78:60:5b:c0:72:23" rssi "-52" channel "6" auth mode "4" 
1) ssid "TP-LINK_kmzn" bssid "7e:60:5b:c0:72:23" rssi "-53" channel "6" auth mode "4" 
2) ssid "hollysys" bssid "04:f9:f8:32:70:cc" rssi "-57" channel "1" auth mode "4" 
3) ssid "" bssid "0e:f9:f8:32:70:cc" rssi "-57" channel "1" auth mode "4" 
4) ssid "qyhs602" bssid "a4:1a:3a:34:15:fe" rssi "-72" channel "11" auth mode "4" 
5) ssid "QYHS1" bssid "8c:8e:0d:4d:2e:95" rssi "-80" channel "11" auth mode "3" 
6) ssid "QYHS2" bssid "dc:65:55:5f:da:db" rssi "-81" channel "1" auth mode "3" 
Station mode: ssid: Gannz_2.4G passwd 1234qwer 
wifi mode is set
mac address is f4:12:fa:d7:9c:3c
Connected
Connected to Gannz_2.4G 
SoftAP mode: ssid: ESP32_TEST passwd 1234qwer 
wifi mode is set
mac address is f4:12:fa:d7:9c:3d
ESP softAP started
started ESP32_TEST softAP

HP4LEVI, 2024-08-09:
日志打印几乎都能与您的示例对应上，但ping时没有任何反应，有可能是我ping的方式有问题吗？

HP4LEVI, 2024-08-09:
@mantriyogesh

HP4LEVI, 2024-08-09:
@renhil 你的可以ping通了吗？

mantriyogesh, 2024-08-09:
- if you are referring to master branch, layer2 [arping](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/Getting_started.md#23-arp-testing-) is only supported.
master branch code is not having network stack integrated, but have note how to integrate it.
> 1. [Network API explained](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/mcu_api.md#2-network-interface-layer)
> 2. [netif](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/MCU_based_architecture.md#14-network-interface-layer-netif)
> 3. [network stub](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/MCU_based_architecture.md#15-network-stack-stub)

- If you want the ready integration, we showcase esp as slave and esp as host, where you can ping , iperf etc.
but stm port is not yet ready, ad hence not integrated. Although the port layer is available which could be used for adaption.
  - `ESP as Host` branch: https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host

HP4LEVI, 2024-08-09:
好的，谢谢。我看看

HP4LEVI, 2024-08-09:
我的ESP32使用的是esp-hosted-master\esp_hosted_fg\esp\esp_driver\network_adapter编译后的代码，STM32运行的是esp-hosted-master\esp_hosted_fg\host\stm32中的代码（我是将其中用到的各种文件移植到了生成的cunbeide中编译运行的
[image]
）

HP4LEVI, 2024-08-09:
我的ESP32使用的是esp-hosted-master\esp_hosted_fg\esp\esp_driver\network_adapter编译后的代码，STM32运行的是esp-hosted-master\esp_hosted_fg\host\stm32中的代码（我是将其中用到的各种文件移植到了生成的cunbeide中编译运行的）
@mantriyogesh

HP4LEVI, 2024-08-09:
这样的搭配有问题吗？

HP4LEVI, 2024-08-09:
而且现在又出现了新的问题。。。日志显示接收长度过大
[image]

mantriyogesh, 2024-08-12:
@renhil ,
Are you still blocked?

@HP4LEVI , 
Can you please raise separate issue, as otherwise, it would be hard for tracking by @renhil .
It looks like you are not getting the correct message from slave to host.
Please cross check some considerations in porting guide. Although created for Linux, many SPI connection related things are relatable for MCU case. check [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md)
It would be worth to check if both sides use same SPI mode, use small wires and have tested all SPI GPIOs, Handshake and Data Ready & Reset pin tested.

HP4LEVI, 2024-08-12:
@mantriyogesh Yes, I will continue to check this week.

mantriyogesh, 2024-08-12:
@HP4LEVI , sure, but please open new issue, otherwise the original issue becomes unmanageable. I hope you understand.

HP4LEVI, 2024-08-12:
OK.

HP4LEVI, 2024-08-13:
好的，我立即尝试一下，谢谢。