espressif/esp-hosted #448: Thread nad ZugBee support

MacWyznawca, 2024-08-02:
When will there be Thread and ZigBee support with the ESP32-Hx?

mantriyogesh, 2024-08-02:
This is our long pending to do list. Might take some time.

dmartauz, 2024-12-28:
@mantriyogesh could you please provide update on Zigbee / Thread support? We have a solution built around P4 as main MCU and C6 as RF coprocessor running esp-hosted and we would like to make use of these protocols.

mantriyogesh, 2024-12-28:
We want to update the thread and zigbee, without doubt, but unfortunately there is no plan for this.

We need to have serious thought on this, @SohKamYung-Espressif

dmartauz, 2024-12-30:
@mantriyogesh @SohKamYung-Espressif could you please come up with a roadmap for Thread/Zigbee/Matter support? We need to know if we can commit to a project. With P4 reaching MP status in a few months I am sure there will be a demand for it.

smart4energy, 2025-02-11:
@mantriyogesh @SohKamYung-Espressif  

is there an update about these topics of a timeline. or will it be never possible to use zigbee/thread with hosted-ng ?

mantriyogesh, 2025-02-11:
Let me come back on this tomorrow

mantriyogesh, 2025-02-11:
Although, we planned to do it on Esp32-P4 (MCU) as host before.

mantriyogesh, 2025-02-17:
Hello @smart4energy ,

After good discussion, we are yet unsure how much time would it take to support the thread on hosted.

We would first target on MCU case (will develop keeping linux in mind as well)
And the n we will support on Linux (yet thinking of FG for simplicity, but should be later easy extended to NG as well)