espressif/esp-hosted #450: How to get esp_hosted_ng  esp32 side version in linux side

lipengvita, 2024-08-06:


mantriyogesh, 2024-08-06:
already available as partt of bootup event.
https://github.com/espressif/esp-hosted/commit/2e6603687a2dab44e1ef4a6fdf615b61b2b4bd3a#diff-f9355f17a7097e147bbde7999388506308832e1fa344809f343ead7757a6423aR346-R352

https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_ng/host/main.c#L285

lipengvita, 2024-08-06:
I  konw this , In linux the application program need  a interface to get the esp version

mantriyogesh, 2024-08-06:
We do this in FG. NG by design, we did not want to do this.

May I know why would you need this information at app? If you need application driven, FG would be easier and more suitable?

kapilkedawat, 2024-08-06:
HI @lipengvita, please let us know exact requirement and need.

lipengvita, 2024-08-07:
We want to know the esp32 software version so that in the future we can upgrade the software in linux side

kapilkedawat, 2024-08-26:
@lipengvita available through https://github.com/espressif/esp-hosted/commit/1bbb7630cf521e5c58f4c63676597053a4f6d229