espressif/esp-hosted #451: Cannot flash via UART0/1

QuiGonRazor, 2024-08-07:
- ESP32-C3
- esp-hosted-ng firmware + kernel linux driver
- sha: https://github.com/espressif/esp-hosted/commit/91b093365cc58214532b976dd8749f65523d1e6d
- configuration: WiFi SPI + BT HCI (H4)

Hello,
I'd like to be able to flash the ESP micro via UART directly from my target (linux)

I have a devkit with a ESP32C3 XXN4.
I'm able to flash the micro via the USB UART interface but, when in download mode, the micro cannot be flashed from UART interfaces

Into your [documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html) you say that when in download mode the micro is suitable for the esptool utility

quote: **DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2) or DOWNLOAD(USB/UART0) - ESP32-C3 is in download flashing mode (suitable for esptool)**

What's strange is that in your documentation you have an example who says this 

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),**boot:0x3** (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))

But when I go in download mode (working with USB but not directly with UARTs) it prints a differe boot value, even though you say it is ok (in your documentation)

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),**boot:0x4** (DOWNLOAD(USB/UART0/1))
waiting for download

What could be the issue here? I really need to make it flashable from inside the target

Attached a trace log from esptool
[esptool.log](https://github.com/user-attachments/files/16529619/esptool.log)

I've tried to use a logical analyzer and I don't see any data from the ESP

mantriyogesh, 2024-08-07:
This is general flashing, comes under ESP-IDF scope.

But anyway I doubt some basic things:
1. Make sure the input voltage is 3.3v at esp32-c3. Power externally, possibly with good data cable capacity and correct output voltage of adapter output
2. Try to remove reset pin and try flashing again.
3. Ensure you use correct uart baud rate (check the sdkconfig file and UART flashing BAUD RATE). Use correct expected baud rate for flashing.

QuiGonRazor, 2024-08-08:
Hello

1. The voltage is correct, I have an external power supply connected to the devkit's 3v3
2. What do you mean with "remove the reset pin"? I press the devkit's BOOT and RST buttons and it goes in DOWNLOAD mode
3. I'm using the idf.py utility who sets automatically the baudrate (460800). I've tried using 115200 as well with no luck

Do you have any feedback about my idea?

In your example the boot mode as a value of 0x3 while mine prints 0x4. Is it correct?

QuiGonRazor, 2024-08-08:
Hello,
a quick update

I managed to get the main UART working removing the resistors. Now I can do queries and flash the micro.

Despite that the HCI UART still doesn't work as flashing way.

From the host I try to query the micro and it prints this

`
esptool -p /dev/ttySTM1 -b 46080 chip_id
esptool.py v4.7.0
Serial port /dev/ttySTM1
Connecting...
Failed to get PID of a device on /dev/ttySTM1, using standard reset sequence.
...................................
A fatal error occurred: Failed to connect to Espressif device: No serial data received.

`

mantriyogesh, 2024-08-12:
HCI over UART has no dependency over ESP-Hosted code.

I am attaching a small write up for your understanding:
[ESP-Hosted-Bluetooth.pdf](https://github.com/user-attachments/files/16580513/ESP-Hosted-Bluetooth.pdf)

mantriyogesh, 2024-08-12:
This documentation is for  branch, where we showcase [ESP as host MCU](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host). For other MCUs, we have port layer. For other MPUs (Linux hosts), you can also refer [Porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md).
Nevertheless, Be it ESP as MCU or Linux as Host or STM32 as host, all would behave in the similar fashion.

mantriyogesh, 2024-08-12:
If you have any SPI or SDIO transport working, you can first verify vHCI, which would be extremely easy. once the host bluetooth stack is fine and working, you can then switch to HCI over UART.
Please cross check HCI over UART documentation, as some changes in firmware configurations would be needed using `idf.py menuconfig`. check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/UART_setup.md