espressif/esp-hosted #452: stm32F407+esp32-C3+  ESP-Hosted-FG Release 0.0.5

alficfte, 2024-08-08:


alficfte, 2024-08-08:
Hi：
Does it support WPA2 Enterprise? stm32F407+esp32-C3+(ESP-Hosted-FG Release 0.0.5, Use default firmware)
What encryption methods do AP support?

mantriyogesh, 2024-08-08:
Station and softap support different auth modes.

Station supports a lot of modes, but softap doesn't.
ESP-Hosted relied on ESP-IDF Wi-Fi driver for this.

On master [ESP-IDF Wi-Fi documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/wifi-security.html#wi-fi-enterprise), I can see, 

> [!NOTE]
> ESP32-C3 supports Wi-Fi Enterprise only in station mode.

Also, security protocol support depends upon the ESP-IDF version you use. We suggest you use latest stable ESP-IDF (5.3).

SoftAP supports WPA2, WPA3 and many other lower security protocols. Exact supported security protocols and limitation in `softap mode` could be found on https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/wifi-security.html#wi-fi-security

While starting softap from host, you need to mention which mode it needs to start with.
You can also build the firmware using https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#222-source-compilation for SPI

closed 2024-08-12 by alficfte