espressif/esp-hosted #453: Failed to register advertisement: org.bluez.Error.Failed

HXHZQF, 2024-08-09:
kernel 5.15 + c6-ng(sdio)
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/bluetooth.md#gatt-server
GATT server
Steps:

Run sudo bluetoothctl。
Run list to get MAC address of ESP.
To set device name, run menu advertise. Then name <enter_any_name>。
To come back to main menu, run back。
To start advertising, run advertise on. Perform below steps on Mobile Phone:
Turn on mobile phone's bluetooth. Open nRF connect application, ESP's MAC address will be displayed under SCANNER tab as a result of scan.
Click on connect. Client tab will be open. Click on Generic Attribute option.
Perform read/write on listed characteristics fields in Generic Attribute service. To disconnet:
Run disconnect <MAC_ADDRESS_of_gatt_client> on linux host's bluetoothctrl OR click on DISCONNECT in nRF connect application's GATT client screen.

log:
bluetoothctl
Agent registered
[CHG] Controller 9C:9E:6E:40:6E:E6 Pairable: yes
[bluetooth]# list
Controller 9C:9E:6E:40:6E:E6 xag-x5pro2 [default]
[bluetooth]# menu advertise
Menu advertise:
Available commands:
-------------------
uuids [uuid1 uuid2 ...]                           Set/Get advertise uuids
service [uuid] [data=xx xx ...]                   Set/Get advertise service data
manufacturer [id] [data=xx xx ...]                Set/Get advertise manufacturer data
data [type] [data=xx xx ...]                      Set/Get advertise data
discoverable [on/off]                             Set/Get advertise discoverable
discoverable-timeout [seconds]                    Set/Get advertise discoverable timeout
tx-power [on/off]                                 Show/Enable/Disable TX power to be advertised
name [on/off/name]                                Configure local name to be advertised
appearance [on/off/value]                         Configure custom appearance to be advertised
duration [seconds]                                Set/Get advertise duration
timeout [seconds]                                 Set/Get advertise timeout
secondary [1M/2M/Coded]                           Set/Get advertise secondary channel
interval [min] [max]                              Set/Get advertise interval range
rsi [on/off]                                      Show/Enable/Disable RSI to be advertised
clear [uuids/service/manufacturer/config-name...] Clear advertise config
back                                              Return to main menu
version                                           Display version
quit                                              Quit program
exit                                              Quit program
help                                              Display help about this program
export                                            Print environment variables
[bluetooth]# name test
[bluetooth]# back
Menu main:
Available commands:
-------------------
advertise                                         Advertise Options Submenu
monitor                                           Advertisement Monitor Options Submenu
scan                                              Scan Options Submenu
gatt                                              Generic Attribute Submenu
admin                                             Admin Policy Submenu
player                                            Media Player Submenu
endpoint                                          Media Endpoint Submenu
transport                                         Media Transport Submenu
list                                              List available controllers
show [ctrl]                                       Controller information
select <ctrl>                                     Select default controller
devices [Paired/Bonded/Trusted/Connected]         List available devices, with an optional property as the filter
system-alias <name>                               Set controller alias
reset-alias                                       Reset controller alias
power <on/off>                                    Set controller power
pairable <on/off>                                 Set controller pairable mode
discoverable <on/off>                             Set controller discoverable mode
discoverable-timeout [value]                      Set discoverable timeout
agent <on/off/capability>                         Enable/disable agent with given capability
default-agent                                     Set agent as the default one
advertise <on/off/type>                           Enable/disable advertising with given type
set-alias <alias>                                 Set device alias
scan <on/off/bredr/le>                            Scan for devices
info [dev/set]                                    Device/Set information
pair [dev]                                        Pair with device
cancel-pairing [dev]                              Cancel pairing with device
trust [dev]                                       Trust device
untrust [dev]                                     Untrust device
block [dev]                                       Block device
unblock [dev]                                     Unblock device
remove <dev>                                      Remove device
connect <dev>                                     Connect device
disconnect [dev]                                  Disconnect device
menu <name>                                       Select submenu
version                                           Display version
quit                                              Quit program
exit                                              Quit program
help                                              Display help about this program
export                                            Print environment variables
[bluetooth]# advertise on
Failed to register advertisement: org.bluez.Error.Failed
[bluetooth]#

HXHZQF, 2024-08-13:
Thank you very much for your support!

I’ve reviewed the details in #381 (comment) carefully, and it seems like the issue we’re encountering is indeed similar. However, following the same steps didn’t resolve the problem. To help you better understand the situation, I’ve provided the relevant logs in the attachment:

Host:

dmesg logs from the start
Bluetooth steps and their results
Output from btmon
Output from hcidump
Output from hciconfig -a
ESP:

sdkconfig file
minicom or idf.py monitor logs
I hope this information will help us in resolving the issue. Please let me know if you need any further details.
[bluetoothctl.log](https://github.com/user-attachments/files/16593873/bluetoothctl.log)
[btmon.log](https://github.com/user-attachments/files/16593874/btmon.log)
[c6_monitor.log](https://github.com/user-attachments/files/16593875/c6_monitor.log)
[dmesg.log](https://github.com/user-attachments/files/16593876/dmesg.log)
[hcidump.log](https://github.com/user-attachments/files/16593877/hcidump.log)
[sdkconfig.txt](https://github.com/user-attachments/files/16593878/sdkconfig.txt)

mantriyogesh, 2024-08-13:
What is BlueZ version you use?

As mentioned in (2) above, 
> starting from BlueZ version 5.65 experimental features really work

Can you try latest BlueZ built with experiementtal features ?

Dorismei, 2024-08-13:
Hello,

I am also encountering the same issue. I’m a Linux beginner, and currently, in versions above 5.55, when I run the "advertise on" command, I see "command unknown" in btmon. If you could provide some guidance or direction on how to resolve this, it would be greatly appreciated.

Thank you!
[image]

mantriyogesh, 2024-08-13:
What happens if you skip running 'name thomas123' , @Dorismei ?

HXHZQF, 2024-08-13:
> What is BlueZ version you use?
> 
> As mentioned in (2) above,
> 
> > starting from BlueZ version 5.65 experimental features really work
> 
> Can you try latest BlueZ built with experiementtal features ?

Hi,

I checked the version using bluetoothctl --version, and it shows that I am currently using BlueZ version 5.70. As noted, experimental features are expected to work starting from version 5.65.

I will proceed with testing the latest BlueZ build with experimental features enabled and update you on the results.

Thank you!

Dorismei, 2024-08-13:
> What happens if you skip running 'name thomas123' , @Dorismei ?

Hello mantriyogesh,

I tried removing the name "thomas123," but the issue still persists. Are there any other possible solutions to this problem?

Thank you!

luisCactuasys, 2024-08-19:
Hi people,

I have also been strugling with this error for the past few days, here is what I tried:
- I started with this https://github.com/espressif/esp-hosted/issues/381#issue-2290812657. I was able to follow the step indicated and advertise with ```hcitool```, which was a good step in the right direction. Because now we constrained the problem to Bluez itself and not some underlying issue.
- However, I could not do it with ```bluetoothctl``` utility. 
- This was no good because my aim was to use my costum application to fit my exact needs (if you are courious, I am using this very nice library https://github.com/weliem/bluez_inc), but this was also not working.

My solution was to simply downgrade BlueZ5 to v5.55 (I was using v5.65)
- I am using a costum Linux image built with Yocto (Kirkstone branch). Tell me if you need help!
- I suspected the version could be the issue because I used Debian Image with the exact same setup (ESP32-C3 WIFI/BLE via SPI) the everything worked just fine.

mantriyogesh, 2024-08-20:
Hello @luisCactuasys ,

I understand it might be frustrating some times.

I think getting to lower versions is good for debugging, but may not be great for long term.

If you have controlled environment, can you please upgrade kernel version locally to 5.10.x +, if your kernel version is lower than that? I don't exactly remember your your kernel version, but some users reported issues with kernel versions around 5.7.x. not exactly sure of range right now. But let me know if your kernel falls in this catagory, I will check and let you know.

Overall, bluetoothctl or btmgmt should be able to cater your all needs.  bluetoothctl is user friendly and btmgmt is for advanced use cases, AFAIK.

Deprecated tools like hciconfig etc, are easy to use, but they invoke some command sequences, which are deprecated for the bluetooth specs. So something is working fine, then great. But sometimes something not working, you might need to reconsider of blutoothctl. In general, the spec would be followed at the same time by kernel version , bluetooth stack, bluetooth controller and bluetooth commands. I would suggest you to for for decent & recent stable LTS OS (if your  eco allows) and without changes, if your issues are getting reproduced or not.

mantriyogesh, 2024-08-20:
In general, ESP-Hosted is just relaying of the bluetooth messages as is, from controller to host stack and vice versa. So ESP-Hosted would have least issues and controller and/or bluetooth stack configs or issues may be present, if you face any kind of issues.

Controller is from ESP-IDF. As per my experience, till now only single issue we had faced in controller over the years, otherwise it behaved as per the specs.
But anyway, btmon and other logs would help if it is controller issues or not. Over the time, we typically observe the bluetooth stack and it's changing nature as per kernel version caused issues, also the incorrect configs, while creating valid use case, was typical issue.

But I think it is important to rule out which entity started to deviate from spec, which could be checked from bluetooth specs. I know specs are technical. But we are also dependent on internal bluetooth teams. ESP-Hosted, being relay, only intention is to deliver msg from one point to other, and vice versa. If you cannot verify on your own, you can get dumps and logs to cross check. Alternatively, in debugging side, if you can recreate use case using ESP-IDF bluetooth examples, to see if the issue is from stack or controller.

luisCactuasys, 2024-08-20:
Hi, @mantriyogesh,

I doubt the issue lies with the ESP-hosted setup. As I mentioned earlier, when I tested with another Linux environment using the exact same hardware (Beaglebone Black and ESP32-C3), I had no problem using ```bluetoothctl``` to advertise.

Here are the details of both setups:

**Debian:**

- ESP32-C3 with esp-hosted_ng (latest version).
- Beaglebone Black running Debian ([AM335x image](https://www.beagleboard.org/distros/am335x-11-7-2023-09-02-4gb-microsd-iot)) with Kernel 5.10 and Bluez 5.55. This setup worked perfectly.
**Yocto (Kirkstone):**

- ESP32-C3 with esp-hosted_ng (latest version).
- Beaglebone Black running Yocto Poky with Kernel 5.15 and Bluez 5.65. In this setup, the advertisement did not work with bluetoothctl.
To troubleshoot, I decided to downgrade Bluez to version 5.55 on the Yocto setup:

**Yocto (Kirkstone) after Bluez downgrade:**

- ESP32-C3 with esp-hosted_ng (latest version).
- Beaglebone Black running Yocto Poky with Kernel 5.15 and Bluez 5.55. This setup worked perfectly.
Based on this, I believe the issue is related to the Bluez stack rather than the ESP-hosted setup. It’s possible that changes in the BLE specification have caused a mismatch between your solution and the Bluez stack, though I doubt it. In my case, downgrading Bluez works fine, so I’m satisfied with this solution.

If you’re interested, I can share the logs from ```btmon```, ```btmgmt```, ```dbus-monitor```, etc. However, they are quite similar to others already shared here.

Thank you!

mantriyogesh, 2024-08-20:
Thanks for full details and clarifications.
As you suggested the blueZ stack may have some difference. Further thank you for confirming that it is not ESP-Hosted issue.

However it would be good to study and point the difference with detailed logs. May be it gets solved, may be not solved, but at least would be good to capture for any similar issues if someone faces in future.

luisCactuasys, 2024-08-20:
I agree. I'm currently short on time, but I plan to explore the issue further in the near future and will keep you informed.