espressif/esp-hosted #454: ESP-HOSTED FG packet classification problem

dreamcmi, 2024-08-09:
I am studying the communication process of FG mode, and now I am wondering how the host distinguishes control packets or 802.3 packets. I noticed that there is a double tlv packet of `endpoint+data`, but I don't know how it works specifically. So my question is when the MCU acts as the host in FG mode, how does the MCU determine whether the received data packet is a network data packet or a control command packet? Thank you.

mantriyogesh, 2024-08-12:
Hello @dreamcmi ,

The network packet is never serialised. Only control packet is serialised.
Control packet uses ESP_SERIAL_IF interface in header.
So basically after serialization, the receiver knows if it needs to de-serialise or not using this interface type.

Here is very much deftailed explanation of control path: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/contrl_path.md

APIs: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md

mantriyogesh, 2024-08-12:
Please feel free to ping if something is not clear or not covered in this documentation.

dreamcmi, 2024-08-12:
Thanks for your reply. I understand a little bit now. Let me briefly describe it.
1. In the master-slave direct interaction data packets of esp-hosted FG, there is a fixed header `esp_payload_header`. Every time a data packet is received, the correctness of this header must be determined first.
2. When the header is correct, then determine the `if_type` type. If it is `ESP_SERIAL_IF`, it means that this is a control data packet. If it is `ESP_STA_IF` or `ESP_AP_IF`, it means that this is an 802.3 frame from sta or ap mode.
3. After determining the packet type, if it is a control packet, enter the deserialization function for parsing. If it is a network packet (802.3), enter the network protocol stack for processing.
4. When the host (such as stm32) wants to send a packet to the slave (such as esp32c3), if it wants to send a control packet, it will be serialized first, and then a header of the` ESP_SERIAL_IF` type will be added. If it wants to send an 802.3 network packet, it will be added with a header of `ESP_STA_IF` or `ESP_AP_IF`, and then sent it via SPI or SDIO.
The above is my understanding, and the following are the questions.
1. Is there anything wrong with my explanation above?
2. I think `ESP_HCI_IF` should indicate a BLE data packet? What type of packet does `ESP_PRIV_IF` represent?
[image]

mantriyogesh, 2024-08-12:
Hello @dreamcmi ,

1.
Your understanding is absolutely correct.
ESP_HCI_IF is bluetooth packet.

ESP_PRIV_IF  is packet that is private packet in between slave and host. However, if you wish to send something small, you can use TLV format in ESP_PRIV_IF.

As ESP_PRIV_IF is generally for some capabilities exchange, we use single byte denoting specific capability or configuration. If you need more than a byte, you might need to serialise (endian-ness)

ESP_PROV_IF is discouraged for large control packets, for the same reason.

2. In your block diagram, right side, there is typo, `want to send a network packet`

3. If you are interested in MCU based host, you can refer to https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host
Where you can connect two ESP chipsets and evaluate the solution. And later can adapt to your host using port layer added.

dreamcmi, 2024-08-12:
Thank you very much for your explanation, it helped me a lot.

mantriyogesh, 2024-08-12:
Please find updated comment: https://github.com/espressif/esp-hosted/issues/454#issuecomment-2283900891