espressif/esp-hosted #456: arping test出错

HP4LEVI, 2024-08-13:
程序正常运行，在测试arpin时不能成功，MCU以及ESP32均无反应。ESP32为ESP32-S3,MCU为STM32H7B0.是我的测试方法不对吗？我尝试了推荐的方式。

HP4LEVI, 2024-08-13:
我现在的程序分别能运行到这个状态
[image]

HP4LEVI, 2024-08-13:
我的程序现在分别能运行到这个状态
[image]
[image]

[image]

mantriyogesh, 2024-08-13:
Please update following logs, in **textual logs**, **from start tto end**:
1. ESP log
2. host log
3. arping commands

1. In general, it is worth to check if your own IP address is in the same network address that of destination IP address.
For example, stm32 and your laptop is connected to same AP, your laptop holds say '192.168.2.101' , make sure your stm32 settings have own IP address as 192.168.2.**X**.  Also ensure **X** is not already used address by someone else, else you would face IP conflict at AP. For example, check pinging 192.168.2.233 on your network. if no reply, use this in settings and re-flash.

2. Cross check if your arp table entries are updated correctly.
- In case of Linux, 
  - check if your MAC address has IP bound, using command:
```bash
arp -n
```
  - If mapping not already present, add one using command:
```bash
sudo arp -s <IP_ADDRESS> <MAC_ADDRESS>
```
<IP_ADDRESS> : your stm32 IP address, 192.168.2.233 in above case
<MAC_ADDRESS> : ESP32 mac address for that wifi interface. For example, for connecting to router/AP, you use 'sta' mode. so use mac address you get in 'sta' mode.

HP4LEVI, 2024-08-13:
这个是STM32的日志：
`
+-----------------------------------+-------------------------------------------+
|           Parameters              |             Values                        |
+-----------------------------------+-------------------------------------------+
|      INPUT__OPERATING_MODE        |             STATION                       |
|      INPUT_GET_AP_SCAN_LIST       |             yes                           |
+-----------------------------------+-------------------------------------------+
|            SOFTAP                 |                                           |
+-----------------------------------+-------------------------------------------+
|        INPUT_SOFTAP__SSID         |             ESP123                        |
|       INPUT_SOFTAP_PASSWORD       |             1234qwer                      |
|      INPUT_SOFTAP_BANDWIDTH       |             40                            |
|       INPUT_SOFTAP_CHANNEL        |             1                             |
|      INPUT_SOFTAP_ENCRYPTION      |             WPA2_PSK                      |
|      INPUT_SOFTAP_MAX_CONN        |             4                             |
|     INPUT_SOFTAP_SSID_HIDDEN      |             no                            |
|       INPUT_SOFTAP_SRC_IP         |             192.168.0.100                 |
|      INPUT_SOFTAP_ARP_DEST_IP     |             192.168.0.122                 |
+-----------------------------------+-------------------------------------------+
|            STATION                |                                           |
+-----------------------------------+-------------------------------------------+
|        INPUT_STATION__SSID        |             Gannz_2.4G                    |
|       INPUT_STATION_BSSID         |                                           |
|   INPUT_STATION_IS_WPA3_SUPPORTED |             no                            |
|      INPUT_STATION_PASSWORD       |             1234qwer                      |
|       INPUT_STATION_SRC_IP        |             192.168.0.112                 |
|     INPUT_STATION_ARP_DEST_IP     |             192.168.0.115                 |
+-----------------------------------+-------------------------------------------+
>>> ctrl_rx_thread start
>>> ctrl_rx_func = 0x0800B9BD
>>> ctrl_msg_Q = 0x200014D0
>>> 3.1 Block on read of protobuf encoded msg
>>> 3.1 buf_len = 0x00000008, buf = 0x200007D0
>>> 3.2 Decode protobuf
>>> 3.2 buf_len = 0x00000008, buf = 0x200007D0
>>> 3.4 Send for further processing as event or response
1. Check if valid proto msg
1. proto_msg = 0x200007E0
2. proto_msg->msg_type == CTRL_MSG_TYPE__Event
ctrl_app_parse_event()
EVENT: ESP INIT
2024-01-01 00:00:00 >  App EVENT: ESP INIT
>>> 3.1 Block on read of protobuf encoded msg
>>> 3.1 buf_len = 0x000001C6, buf = 0x200007D0
>>> 3.2 Decode protobuf
>>> 3.2 buf_len = 0x000001C6, buf = 0x200007D0
>>> 3.4 Send for further processing as event or response
1. Check if valid proto msg
1. proto_msg = 0x200009A0
3. proto_msg->msg_type == CTRL_MSG_TYPE__Resp
ctrl_app_parse_resp()...start
ctrl_msg->msg_id = 205(0xCD)
ctrl_msg->msg_id = CTRL_RESP_GET_AP_SCAN_LIST
ctrl_msg->msGparaM = 0x200009D0
ctrl_msg->msGparaM->resp = 0(0x0)
###ctrl_app_parse_resp()...ESP_SUCCESS
>>> 3.1 Block on read of protobuf encoded msg
Station mode: ssid: Gannz_2.4G passwd 1234qwer 
>>> 3.1 buf_len = 0x0000000A, buf = 0x200007D0
>>> 3.2 Decode protobuf
>>> 3.2 buf_len = 0x0000000A, buf = 0x200007D0
>>> 3.4 Send for further processing as event or response
1. Check if valid proto msg
1. proto_msg = 0x200007E8
3. proto_msg->msg_type == CTRL_MSG_TYPE__Resp
ctrl_app_parse_resp()...start
ctrl_msg->msg_id = 204(0xCC)
ctrl_msg->msg_id = CTRL_RESP_SET_WIFI_MODE
ctrl_msg->msGparaM = 0x20000818
[... 201 lines cut ...]
I (3941) slave_ctrl: SSID               YJ
RSSI                    -89
Channel                 6
BSSID                   38:68:be:d9:ad:a1
Auth mode               3

I (4013) slave_ctrl: Set wifi mode 1 
I (4064) slave_ctrl: Get station mac address
I (4065) slave_ctrl: mac [f4:12:fa:d7:9c:3c]
I (4118) slave_ctrl: station mode set
I (4119) slave_ctrl: Get station mac address
I (4119) slave_ctrl: mac [f4:12:fa:d7:9c:3c]
I (5816) wifi:new:<6,2>, old:<1,1>, ap:<255,255>, sta:<6,2>, prof:1
I (6325) wifi:state: init -> auth (b0)
I (6331) wifi:state: auth -> assoc (0)
I (6337) wifi:state: assoc -> run (10)
I (6366) wifi:connected with Gannz_2.4G, aid = 13, channel 6, 40D, bssid = 78:60:5b:c0:72:23
I (6367) wifi:security: WPA2-PSK, phy: bgn, rssi: -56
I (6369) wifi:pm start, type: 1

I (6372) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (6380) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (6388) slave_ctrl: connected to ap SSID:'Gannz_2.4G', password:'1234qwer'  
I (6456) wifi:AP's beacon interval = 102400 us, DTIM period = 1
`

mantriyogesh, 2024-08-13:
Great, if you wish you can raise PR (pull req) for ESP32 partition changes

HP4LEVI, 2024-08-13:
因为我使用的是windows11电脑，所以只能通过ubuntu进行arping命令，我刚才试过了将STM32的地址设置成（不能通过 ping 命令成功的地址），然后再放到虚拟机中进行mac地址映射，最后在进行 sudo arping命令，依旧是没有日志的反应

HP4LEVI, 2024-08-13:
> 太好了，如果您愿意，您可以为 ESP32 分区更改提高 PR（拉取请求）

这个是什么意思呢？

mantriyogesh, 2024-08-13:
You can try arping now from documentation, as you are running on master branch.

HP4LEVI, 2024-08-13:
[image]

HP4LEVI, 2024-08-13:
事实上我已经执行过arping命令了，使用的这个地址

HP4LEVI, 2024-08-13:
[image]

mantriyogesh, 2024-08-13:
> 因为我使用的是windows11电脑，所以只能通过ubuntu进行arping命令，我刚才试过了将STM32的地址设置成（不能通过 ping 命令成功的地址），然后再放到虚拟机中进行mac地址映射，最后在进行 sudo arping命令，依旧是没有日志的反应

Ping cannot work. Right now there is just network stub in master. You can either hook using documentation or refer to other branch , where lwip is integrated, but need to adapt port layer for stm.

mantriyogesh, 2024-08-13:
So arping is working?

HP4LEVI, 2024-08-13:
但是我看您的arping可以工作呀，而且还有相应的arping_test日志

HP4LEVI, 2024-08-13:
> 所以arping工作吗？

是的，使用sudo arping 这两个设置的地址都能成功192.168.0.112，192.168.0.115

mantriyogesh, 2024-08-13:
Try to separate network address for STA and AP in general.
They should be in different subnet.

You can change https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_fg/host/stm32/driver/transport/spi/spi_drv.c#L305-L338 to : 

```c
static void check_and_execute_spi_transaction(void)
{
	uint8_t * txbuff = NULL;
	uint8_t is_valid_tx_buf = 0;
	GPIO_PinState gpio_handshake = GPIO_PIN_RESET;
	GPIO_PinState gpio_rx_data_ready = GPIO_PIN_RESET;

	xSemaphoreTake(mutex_spi_trans, portMAX_DELAY); //change

	/* handshake line SET -> slave ready for next transaction */
	gpio_handshake = HAL_GPIO_ReadPin(GPIO_HANDSHAKE_GPIO_Port,
			GPIO_HANDSHAKE_Pin);

	/* data ready line SET -> slave wants to send something */
	gpio_rx_data_ready = HAL_GPIO_ReadPin(GPIO_DATA_READY_GPIO_Port,
			GPIO_DATA_READY_Pin);

	if (gpio_handshake == GPIO_PIN_SET) {

		/* Get next tx buffer to be sent */
		txbuff = get_tx_buffer(&is_valid_tx_buf);

		if ( (gpio_rx_data_ready == GPIO_PIN_SET) ||
		     (is_valid_tx_buf) ) {

			/* Execute transaction only if EITHER holds true-
			 * a. A valid tx buffer to be transmitted towards slave
			 * b. Slave wants to send something (Rx for host)
			 */
			//xSemaphoreTake(mutex_spi_trans, portMAX_DELAY);
			spi_trans_func[hardware_type](txbuff);
			//xSemaphoreGive(mutex_spi_trans);
		}
	}
	xSemaphoreGive(mutex_spi_trans); // change
}
```

HP4LEVI, 2024-08-13:
好的，我试一下。

HP4LEVI, 2024-08-13:
请问图片中的日志是通过这种ip地址映射实现的吗？
[image]

mantriyogesh, 2024-08-13:
Sorry, I did not understand. Are you able to see these on stm32 log?

First log:
arpping triggered from laptop to stm32 , would be printed as 

`ARP_REQ_RECD: **`
STM32 would respond back the ARP_RSP.

Second log:
ARP ping response received at host, for request triggered earlier. This log trace may change as per the aring implementation. Essentially, you should get the response for ARP ping, irrespective of the trace printed.

HP4LEVI, 2024-08-13:
是的  这两条日志都不曾出现

HP4LEVI, 2024-08-13:
这就是我一直感到困扰的问题，所以我一直在尝试ping所设置的ip地址，我担心是我的操作问题，因为ESP32跟STM32的日志看起来都很正常，反而是我的ping操作不是很正规

mantriyogesh, 2024-08-13:
did you get this: https://github.com/espressif/esp-hosted/issues/456#issuecomment-2285566154?

HP4LEVI, 2024-08-13:
是的  这个确实是能ping通，但是我发现他的MAC地址对不上

mantriyogesh, 2024-08-13:
This is very basic network stub.
your ping can only be working when you have network stack integrated.

Diff between arping vs ping

- ARPing (Address Resolution Protocol Ping)
  - This is indeed a Layer 2 protocol in the OSI model. It is used to discover the MAC address of a host on the local network by sending an ARP request. ARPing operates at the Data Link Layer (Layer 2).
- Ping (ICMP Ping)
  - This operates at Layer 3 of the OSI model. Ping uses the Internet Control Message Protocol (ICMP), which is part of the Network Layer (Layer 3). It sends ICMP Echo Request messages and waits for Echo Reply messages to check the reachability of a network device.

ARPing is a **Layer 2** protocol.
Ping is a **Layer 3** protocol.

**Layer 3** cannot work without proper network stack integrated.

### How to integrate network stack
https://github.com/espressif/esp-hosted/issues/446#issuecomment-2277285846
There are two options discussed in the comment.

HP4LEVI, 2024-08-13:
好的，我立即尝试一下，谢谢。

mantriyogesh, 2024-08-13:
The ESP as host by the way uses as much as ESP-IDF components at host side.
Those components may be needed a copy for now, but if everything taken at port layer, should work.

I think it would be better to evaluate using esp as host and esp another esp as slave using the [feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host)

HP4LEVI, 2024-08-21:
你好 @mantriyogesh ，这周我在尝试esp_as_mcu_hosted,我的从机代码能够很好的编译通过并且下载成功。但是，主机的代码我参考了提示：选择了example中的scan文件，使用vscode打开后，添加了espressif/esp_wifi_remote: "^0.3.0"跟espressif/esp_hosted: "^0.0.9"组件，编译时却失败了，并且配置文件也不能成功打开。

HP4LEVI, 2024-08-21:
看起来是配置文件的问题

HP4LEVI, 2024-08-21:
能详细说明一下吗，谢谢

mantriyogesh, 2024-08-21:
ESP-Hosted includes another sub project (i.e. git sub-module), protobuf.

[image]
Unless all submodules are updated, build would fail.

We have captured in master branch, but feature/mcu_as_host_mcu branch missed in documentation, which we would update in short.

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Linux_based_readme.md#12-host-setup

[image]

mantriyogesh, 2024-08-21:
So simply, run:
```bash
git submodule update --init --recursive
```

In `esp-hosted` directory and all dependent the source files would be fetched.

mantriyogesh, 2024-08-21:
For more information on submodules can be found at: 
https://phoenixnap.com/kb/git-submodule : search for `Joining a Project Using Submodules`

HP4LEVI, 2024-08-23:
抱歉，忘记回复了。我正在继续测试中，谢谢你的回复。