espressif/esp-hosted #458: NG mode, WiFi connected with WPA2WPA3 mixed mode fail

lwj850533527, 2024-08-15:
Hi mantriyogesh
Hi kapilkedawat
@mantriyogesh , @kapilkedawat
The router is set to WPA2WPA3 mixed mode, connect ap fail. Reason 202
The firmware log has been sent out authentication
But sniffer capture did not capture any information

Does ESP32 support WPA2WPA3 mixed mode?
[image]

[20240812-140121 - dmesg -  wpa2wpa3 fail.txt](https://github.com/user-attachments/files/16623883/20240812-140121.-.dmesg.-.wpa2wpa3.fail.txt)
[20240812-140121 - firmware logs -  wpa2wpa3 fail.txt](https://github.com/user-attachments/files/16623884/20240812-140121.-.firmware.logs.-.wpa2wpa3.fail.txt)
[20240812-140121 - sniffer capture -  wpa2wpa3 fail.zip](https://github.com/user-attachments/files/16623889/20240812-140121.-.sniffer.capture.-.wpa2wpa3.fail.zip)

lwj850533527, 2024-08-19:
Hi mantriyogesh
Hi kapilkedawat
@mantriyogesh , @kapilkedawat

This commits doesn't seem to solve the problem
https://github.com/espressif/esp-hosted/commit/2e75e42bb8f379fe61c250846effc88da3e029d5

kapilkedawat, 2024-08-19:
Hi @lwj850533527, could you please share new logs and capture. We are able to connect to WPA2+WPA3 AP.

@Shreyas0-7 could you please check?

Shreyas0-7, 2024-08-19:
Hey @lwj850533527 thanks for reporting the issue. We were able to reproduce the issue and are in the process of resolving it.

kapilkedawat, 2024-08-22:
Hi @lwj850533527, please try with https://github.com/espressif/esp-hosted/commit/7c6d5731f473bd02544c831f4c853f82c5365f36

lwj850533527, 2024-08-22:
Hi kapilkedawat
@kapilkedawat
Can you show separately the wpa_supplicant.conf about WPA2+WPA3 and WPA3?

Shreyas0-7, 2024-08-22:
Hey @lwj850533527 
I used this configuration to connect using WPA3 for WPA3 AP as well as WPA2+WPA3 AP

network={
    ssid="YourSSID"
    psk="YourPassphrase"
    scan_ssid=1
    key_mgmt=SAE
    ieee80211w=2
}

And used this configuration to connect using WPA2 for WPA2+WPA3 AP

network={
ssid="YourSSID"
psk="YourPassphrase"
    scan_ssid=1
    ieee80211w=2
}

kapilkedawat, 2024-08-26:
Fixed in https://github.com/espressif/esp-hosted/commit/7c6d5731f473bd02544c831f4c853f82c5365f36 

@lwj850533527 can you please confirm..

lwj850533527, 2024-08-27:
Hi @kapilkedawat,
The compilation error occurred after incorporating this change.
Should I upgrade  IDF version, using 5.1.1 now 

A fatal error occurred: Segment loaded at 0x4209b3e4 lands in same 64KB flash mapping as segment loaded at 0x4209b11a. Can't generate binary. Suggest changing linker script or ELF to merge sections.
[image]

kapilkedawat, 2024-08-27:
Hi @lwj850533527, could you please clean the build folder and retry? Also did you run `setup.sh` to setup esp-idf?

lwj850533527, 2024-08-27:
Hi @kapilkedawat.
I upgrade IDF version to v5.1.3, solved the compilation error
But I encountered a new problem
All connections are reported as reason15
The password I confirmed is correct

[image]

lwj850533527, 2024-08-27:
No, I haven't run setup.sh before
Download  [esp-hosted](https://github.com/espressif/esp-hosted) code through web latest master
Combined with idf v5.1.1 compilation error， combined with idf v5.1.3 compilation completed

lwj850533527, 2024-08-27:
Reason 15 maybe it's because I didn't use the complete [esp-hosted](https://github.com/espressif/esp-hosted) code.
Esp-hosted code latest master I didn't use it directly，just change commit https://github.com/espressif/esp-hosted/commit/7c6d5731f473bd02544c831f4c853f82c5365f36，and copy lib to the old code

Shreyas0-7, 2024-08-27:
Hey @lwj850533527 
you have to compile the module again and do insmod.

kapilkedawat, 2024-08-27:
@lwj850533527 Please clone latest code for hosted ng and use that. Please use https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md to compile new firmware.

lwj850533527, 2024-08-28:
Hi @kapilkedawat.
I clone latest code for hosted ng.
But esp32 unable to communicate

```
[    9.323200] esp32_spi: loading out-of-tree module taints kernel.             
[    9.542096] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [30]                                           
[   12.641884] esp32_spi: process_esp_bootup_event: Received ESP bootup event   
[   12.647475] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3         
[   12.653806] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI                                                                        
[   12.661531] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2         
[   12.667948] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0         
[   12.674251] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1         
[   12.680653] esp32_spi: process_fw_data: Length not matching to firmware data size                                                                            
[   12.688089] esp32_spi: process_event_esp_bootup: failed to process tag=1 in bootup event   
```

Shreyas0-7, 2024-08-28:
Hey @lwj850533527 it seems you are using different commits for flashing esp fw and hosted_ng module.
Can you please check once that you are using same commit for esp firmware as well hosted driver module

lwj850533527, 2024-08-29:
Hi @kapilkedawat.
HI @Shreyas0-7.
I upgrade hosted driver module to same commit for esp firmware.
It can communicate normally

If I use wpa-psk connect to wpa2+wpa3 AP, it can communicate well.
If I use SAE connect to wpa2+wpa3 AP, it still doesn't seem to be able to connect AP

[image]
[image]
[image]

Shreyas0-7, 2024-08-29:
Hey @lwj850533527 can you please share wpa_supplicant logs, you can run this command to get supplicant logs 
` wpa_supplicant -i interface -c configuration_file.conf -dd -K`
where interface can be wlan0/wlan1
configuration file is conf you are using for connection (/etc/wpa_supplicant.conf)

lwj850533527, 2024-08-29:
```
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: Try to associate due to network getting re-enabled
Aug 29 15:24:15 wpa_supplicant[934]: Fast associate: Old scan results
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: Setting scan request: 0.000000 sec
Aug 29 15:24:15 wpa_supplicant[934]: Scan SSID - hexdump(len=9): 65 73 73 65 6e 74 69 65 6c
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: Starting AP scan for wildcard SSID
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: Add radio work 'scan'@0x1c45f0
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: First radio work item in the queue - schedule start immediately
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: Starting radio work 'scan'@0x1c45f0 after 0.000087 second wait
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: nl80211: scan request
Aug 29 15:24:15 wpa_supplicant[934]: nl80211: Scan SSID essentiel
Aug 29 15:24:15 wpa_supplicant[934]: nl80211: Scan SSID
Aug 29 15:24:15 wpa_supplicant[934]: nl80211: Scan extra IEs - hexdump(len=13): 7f 0b 00 00 00 00 00 00 40 00 00 00 20
Aug 29 15:24:15 wpa_supplicant[934]: Scan requested (ret=0) - scan timeout 30 seconds
Aug 29 15:24:15 wpa_supplicant[934]: nl80211: Event message available
Aug 29 15:24:15 wpa_supplicant[934]: nl80211: Drv Event 33 (NL80211_CMD_TRIGGER_SCAN) received for wlan0
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: nl80211: Scan trigger
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: Event SCAN_STARTED (47) received
Aug 29 15:24:15 wpa_supplicant[934]: wlan0: Own scan request started a scan in 0.000330 seconds
Aug 29 15:24:16 wpa_supplicant[934]: nl80211: Event message available
Aug 29 15:24:16 wpa_supplicant[934]: nl80211: Drv Event 34 (NL80211_CMD_NEW_SCAN_RESULTS) received for wlan0
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: nl80211: New scan results available
Aug 29 15:24:16 wpa_supplicant[934]: nl80211: Scan probed for SSID 'essentiel'
Aug 29 15:24:16 wpa_supplicant[934]: nl80211: Scan probed for SSID ''
Aug 29 15:24:16 wpa_supplicant[934]: nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Event SCAN_RESULTS (3) received
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Scan completed in 1.690059 seconds
Aug 29 15:24:16 wpa_supplicant[934]: nl80211: Received scan results (1 BSSes)
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: BSS: Start scan result update 9
Aug 29 15:24:16 wpa_supplicant[934]: BSS: last_scan_res_used=1/32
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: New scan results available (own=1 ext=0)
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Radio work 'scan'@0x1c45f0 done in 1.728057 seconds
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: radio_work_free('scan'@0x1c45f0): num_active_works --> 0
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="essentiel"
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Selecting BSS from priority group 0
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: 0: d4:3a:2e:d2:4a:d8 ssid='essentiel' wpa_ie_len=0 rsn_ie_len=30 caps=0x1c31 level=-27 freq=2427
Aug 29 15:24:16 wpa_supplicant[934]: wlan0:    selected based on RSN IE
Aug 29 15:24:16 wpa_supplicant[934]: wlan0:    selected BSS d4:3a:2e:d2:4a:d8 ssid='essentiel'
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Considering connect request: reassociate: 0  selected: d4:3a:2e:d2:4a:d8  bssid: 00:00:00:00:00:00  pending: 00:00:00:00:00:00  wpa_state: SCANNING  ssid=0x1ab588  current_ssid=(nil)
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Request association with d4:3a:2e:d2:4a:d8
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Re-association to the same ESS
Aug 29 15:24:16 wpa_supplicant[934]: TDLS: TDLS is allowed in the target BSS
Aug 29 15:24:16 wpa_supplicant[934]: TDLS: TDLS channel switch allowed in the target BSS
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: No ongoing scan/p2p-scan found to abort
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Add radio work 'sme-connect'@0x1c45f0
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: First radio work item in the queue - schedule start immediately
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Starting radio work 'sme-connect'@0x1c45f0 after 0.000117 second wait
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: WPA: clearing own WPA/RSN IE
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: RSN: clearing own RSNXE
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Automatic auth_alg selection: 0x1
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: Using SAE auth_alg
Aug 29 15:24:16 wpa_supplicant[934]: RSN: PMKSA cache search - network_ctx=(nil) try_opportunistic=0 akmp=0x0
Aug 29 15:24:16 wpa_supplicant[934]: RSN: Search for BSSID d4:3a:2e:d2:4a:d8
Aug 29 15:24:16 wpa_supplicant[934]: RSN: No PMKSA cache entry found
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: RSN: using IEEE 802.11i/D9.0
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: WPA: Selected cipher suites: group 16 pairwise 16 key_mgmt 1026 proto 2
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: WPA: Selected mgmt group cipher 32
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: WPA: clearing AP WPA IE
Aug 29 15:24:16 wpa_supplicant[934]: WPA: set AP RSN IE - hexdump(len=32): 30 1e 01 00 00 0f ac 04 01 00 00 0f ac 04 02 00 00 0f ac 02 00 0f ac 08 80 00 00 00 00 0f ac 06
Aug 29 15:24:16 wpa_supplicant[934]: WPA: set AP RSNXE - hexdump(len=3): f4 01 20
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: WPA: AP group 0x10 network profile group 0x18; available group 0x10
Aug 29 15:24:16 wpa_supplicant[934]: wlan0: WPA: using GTK CCMP
[... cut ...]
Aug 29 15:24:29 wpa_supplicant[934]: wlan0: BSS: Start scan result update 10
Aug 29 15:24:29 wpa_supplicant[934]: BSS: last_scan_res_used=1/32
Aug 29 15:24:29 wpa_supplicant[934]: wlan0: New scan results available (own=1 ext=0)
Aug 29 15:24:29 wpa_supplicant[934]: wlan0: Radio work 'scan'@0x1aae88 done in 1.723788 seconds
Aug 29 15:24:29 wpa_supplicant[934]: wlan0: radio_work_free('scan'@0x1aae88): num_active_works --> 0
Aug 29 15:24:29 wpa_supplicant[934]: wlan0: Postpone network selection by 48 seconds since all networks are disabled

```

Shreyas0-7, 2024-08-29:
Hey @lwj850533527 
we are failing to connect because AP is sending us AUTH with H2E enabled, can you please try enabling sae_pwe in supplicant conf?
`Aug 29 15:24:17 wpa_supplicant[934]: SAE: Unexpected use of status code for H2E in SAE commit when H2E was not expected`
you can add 
`sae_pwe=1`
in your supplicant.conf

lwj850533527, 2024-08-30:
Many thanks for your help , and now it can communicate wpa3
`sae_pwe=1`
It can  solve problems well

lwj850533527, 2024-08-31:
> I upgrade IDF version to v5.1.3, solved the compilation error
> But I encountered a new problem
> All connections are reported as reason15
> The password I confirmed is correct [...]

Hi @kapilkedawat. I mentioned the issue of reason 15 before
I think the issue because of the commit https://github.com/espressif/esp-hosted/commit/76085845fa007342e5f2c3811b03abc9f5ddbbde

I tried to adapt to the old version by using the version number.
This change can solve the problem with Reason 15.
I would like to inquire is this modification correct , and if there are any other adaptations needed
[image]

kapilkedawat, 2024-09-02:
Hi @lwj850533527, the change is correct, however I would again suggest you to use the latest codebase :) . Do let us know if you are facing any kind of issues while using latest code.

lwj850533527, 2024-09-02:
Hi @kapilkedawat. 
Many thanks for your suggestion. I will use the latest codebase in the products have not yet produced.
But at the same time, I also need to fit to the products that have already been produced without upgrade function

kapilkedawat, 2024-09-02:
Okay.. in that case please use the firmware and host at the same version. We can suggest host patches to any specific issue that you are facing on that version.

lwj850533527, 2024-09-02:
All the required lines, Is there a detailed explanation? I need to confirm

lwj850533527, 2024-09-02:
Hi @kapilkedawat.
HI @Shreyas0-7.
When I connect to another AP with wpa3, it prompts that the AP cannot be found

```
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: Starting AP scan for wildcard SSID
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: Determining shared radio frequencies (max len 1)
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: Shared frequencies (len=0): completed iteration
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: Add radio work 'scan'@0x1c19e8
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: First radio work item in the queue - schedule start immediately
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: Starting radio work 'scan'@0x1c19e8 after 0.000100 second wait
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: nl80211: scan request
Sep  2 16:38:41 wpa_supplicant[482]: Scan requested (ret=0) - scan timeout 30 seconds
Sep  2 16:38:41 wpa_supplicant[482]: nl80211: Drv Event 33 (NL80211_CMD_TRIGGER_SCAN) received for wlan0
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: nl80211: Scan trigger
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: Event SCAN_STARTED (47) received
Sep  2 16:38:41 wpa_supplicant[482]: wlan0: Own scan request started a scan in 0.000283 seconds
Sep  2 16:38:43 wpa_supplicant[482]: nl80211: Drv Event 34 (NL80211_CMD_NEW_SCAN_RESULTS) received for wlan0
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: nl80211: New scan results available
Sep  2 16:38:43 wpa_supplicant[482]: nl80211: Scan probed for SSID ''
Sep  2 16:38:43 wpa_supplicant[482]: nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: Event SCAN_RESULTS (3) received
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: Scan completed in 1.689242 seconds
Sep  2 16:38:43 wpa_supplicant[482]: nl80211: Received scan results (25 BSSes)
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: BSS: Start scan result update 2
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: BSS: Add new id 22 BSSID b0:df:c1:7d:94:51 SSID 'zhuoer' freq 2422
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: BSS: Add new id 23 BSSID 22:a3:52:23:c1:88 SSID 'HUAWEI2' freq 2462
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: BSS: Add new id 24 BSSID 34:f7:16:a7:c8:1b SSID 'TP-LINK_C81B' freq 2412
Sep  2 16:38:43 wpa_supplicant[482]: BSS: last_scan_res_used=25/32
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: New scan results available (own=1 ext=0)
Sep  2 16:38:43 wpa_supplicant[482]: WPS: AP 22:a3:52:23:c1:88 type 0 added
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: Radio work 'scan'@0x1c19e8 done in 1.725925 seconds
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: radio_work_free('scan'@0x1c19e8): num_active_works --> 0
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: Selecting BSS from priority group 0
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: 0: b4:37:d8:03:33:1a ssid='D-Link-3318' wpa_ie_len=22 rsn_ie_len=26 caps=0x1c31 level=-24 freq=2427
Sep  2 16:38:43 wpa_supplicant[482]: wlan0:    skip - SSID mismatch
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: 1: 06:f9:f8:1f:cd:fd ssid='M-Device' wpa_ie_len=0 rsn_ie_len=20 caps=0x1c11 level=-38 freq=2462
Sep  2 16:38:43 wpa_supplicant[482]: wlan0:    skip - SSID mismatch
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: 2: 06:f9:f8:2f:cd:fd ssid='Moorewatt-Dev' wpa_ie_len=0 rsn_ie_len=20 caps=0x1c11 level=-39 freq=2462
Sep  2 16:38:43 wpa_supplicant[482]: wlan0:    skip - SSID mismatch
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: 3: 34:98:b5:a0:7d:01 ssid='NETGEAR05' wpa_ie_len=0 rsn_ie_len=20 caps=0x531 level=-44 freq=2457
Sep  2 16:38:43 wpa_supplicant[482]: wlan0:    selected based on RSN IE
Sep  2 16:38:43 wpa_supplicant[482]: wlan0:    skip - SAE H2E required, but not supported by the AP
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: No suitable network found
Sep  2 16:38:43 wpa_supplicant[482]: wlan0: Setting scan request: 5.000000 sec
```

kapilkedawat, 2024-09-02:
> All the required lines, Is there a detailed explanation? I need to confirm

https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html#id4

Shreyas0-7, 2024-09-02:
Hey @lwj850533527 
can you please update the value of sae_pwe
`sae_pwe=2`
in supplicant.conf
here 2 means it can connect with AP supporting H2E as well as Hunt and Peck

lwj850533527, 2024-09-02:
Many thanks for your help.
`sae_pwe=2`
It can solve problems well

adityat-1812, 2024-09-18:
Hello @lwj850533527 
Can you share your device tree changes?
Thanks & Regards,
Aditya T

closed 2024-09-24 by lwj850533527