espressif/esp-hosted #459: SDIO 通讯一定数量包后会卡主

jankens, 2024-08-16:
我们在使用 sdio 连接esp32 模块的时候遇到一个传送数据卡主的问题。具体情况如下，请帮忙澄清。非常感谢！！
以下是host cpu sdio driver的代码：
[image]
[image]
上图打印的结果是：
[image]
当 *len达到4096以后，就会变成0，导致 buf_available < buf_needed

mantriyogesh, 2024-08-16:
Can you please let us know some details:
1. host used - any details? Is it MCU or Linux?
2. if linux, what's kernel version
3. Complete **textual** logs at ESP **from start-up**
4. Complete **textual** logs at host **from start-up**
5. git commit used at ESP
6. git commit used at Host

jankens, 2024-08-21:
1.host used - any details? Is it MCU or Linux?
      Linux

2. if linux, what's kernel version
      Linux version 4.4.94+ (wang@jankens-home-ubuntu) (gcc version 7.2.0 (Ingenic Linux-Release5.1.0- 
           Default(xburst2(fp64)+glibc2.29) 2021.12-22 10:52:10) ) #328 SMP PREEMPT Sat May 18 22:30:33 HKT 2024

3. Complete textual logs at ESP from start-up
     	ets Jul 29 2019 12:21:46

	rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
	configsip: 271414342, SPIWP:0xee
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:DIO, clock div:2
	load:0x3fff0030,len:7132
	ho 0 tail 12 room 4
	load:0x40078000,len:15616
	load:0x40080400,len:4
	load:0x40080404,len:3876
	entry 0x4008064c
	I (32) boot: ESP-IDF v5.1.3-4-g9a1f2764fe-dirty 2nd stage bootloader
	I (32) boot: compile time Apr 29 2024 10:21:58
	I (34) boot: Multicore bootloader
	I (38) boot: chip revision: v3.1
	I (42) boot.esp32: SPI Speed      : 40MHz
	I (46) boot.esp32: SPI Mode       : DIO
	I (51) boot.esp32: SPI Flash Size : 2MB
	I (55) boot: Enabling RNG early entropy source...
	I (61) boot: Partition Table:
	I (64) boot: ## Label            Usage          Type ST Offset   Length
	I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
	I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
	I (87) boot:  2 factory          factory app      00 00 00010000 001c2000
	I (94) boot: End of partition table
	I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4887ch (297084) map
	I (214) esp_image: segment 1: paddr=000588a4 vaddr=3ff80000 size=00004h (     4) load
	I (214) esp_image: segment 2: paddr=000588b0 vaddr=3ffbdb60 size=05944h ( 22852) load
	I (229) esp_image: segment 3: paddr=0005e1fc vaddr=40080000 size=01e1ch (  7708) load
	I (232) esp_image: segment 4: paddr=00060020 vaddr=400d0020 size=120a40h (1182272) map
	I (664) esp_image: segment 5: paddr=00180a68 vaddr=40081e1c size=19564h (103780) load
	I (721) boot: Loaded app from partition at offset 0x10000
	I (721) boot: Disabling RNG early entropy source...
	I (733) cpu_start: Unicore app
	I (733) cpu_start: Pro cpu up.
	I (733) cpu_start: Single core mode	
	W (897) clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator
	I (905) cpu_start: Pro cpu start user code
	I (905) cpu_start: cpu freq: 160000000 Hz
	I (905) cpu_start: Application information:
	I (910) cpu_start: Project name:     cino_esp32
	I (915) cpu_start: App version:      v5.1.3-10-gfc265a3774
	I (921) cpu_start: Compile time:     Aug  7 2024 11:39:36
	I (927) cpu_start: ELF file SHA256:  9ae92c267fcfc706...
	I (933) cpu_start: ESP-IDF:          v5.1.3-10-gfc265a3774
	I (939) cpu_start: Min chip rev:     v0.0
	I (944) cpu_start: Max chip rev:     v3.99
	I (949) cpu_start: Chip rev:         v3.1
	I (954) heap_init: Initializing. RAM available for dynamic allocation:
	I (961) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
	I (967) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
	I (973) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
	I (979) heap_init: At 3FFD0B38 len 0000F4C8 (61 KiB): DRAM
	I (985) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
	I (992) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
	I (998) heap_init: At 4009B380 len 00004C80 (19 KiB): IRAM
	I (1004) heap_init: At 3Fmmc3: Unknown controller version (5). You may experience problems.
	I (1012) spi_flash: detected chip: generic
	I (1015) spi_flash: flize(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
	I (1044) coexist: coex firmware version: 77cd7f8
	I (1045) app_start: Starting scheduler on CPU0
[... 252 lines cut ...]
	reset_user_frame_when_stream_on=1
	esp_serial_rx <<<<<: len=68
	esp_serial_rx <<<<<: len=68
	esp_serial_tx >>>>>: frag_len=136
	========esp_slave_get_tx_buffer_num 001c0000, 00000013
	esp_serial_rx <<<<<: len=52
	# ESP32: cradle send stuck !!
	esp32_write_buffer len = 69 pos=0
	ESP32: update pkg_idx_rsp 2
	ESP32: update pkg_idx_rsp 3
	ESP32: update pkg_idx_rsp 4
	W (19265) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(400 800)
	W (19290) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 800, status 0x0
	I (19290) L2CAP_TAG: ESP_BT_GAP_MODE_CHG_EVT mode:2 bda:[00:25:14:05:15:35]
        ......
        
5. git commit used at ESP
      	commit e7771c75bd1dbbfb7b3c5381be7e063b197c9734 (tag: v5.1.3)
	Author: Marius Vikhammer <marius.vikhammer@espressif.com>
	Date:   Mon Feb 26 11:09:16 2024 +0800

		versions: Update version to 5.1.3

6. git commit used at Host
       We are using Ingenic 君正 X2100 CPU， and the host linux is the official linux of X2100.

mantriyogesh, 2024-08-21:
Which ESP hosted commit you have used for esp-hosted slave?

I am unsure if if is esp-hosted slave, or some other binary.

jankens, 2024-08-22:
The  esp_hosted code  is merged into Ingenic x2100 linux by Ingenic company (supported by ESP company as I know), I don't know  which commit they are using. but I can send you the source code of the sdio driver. the driver can communication with esp32 module using SDIO at the beginning, but after a fix number of packets transmit from host to esp32. it is stuck as descript in the question. host can receive packets from esp32 module when sending stuck.
[esp32_linux.zip](https://github.com/user-attachments/files/16707753/esp32_linux.zip)

mantriyogesh, 2024-08-22:
Cannot exactly tell which commit.
But In any case, if the Linux was tested using ESP32, it should work if hardware is already on PCB.

Is it possible to show the connections in camera photo?

Also, the current firmware in ESP looks not to be flash from ESP-Hosted.
It looks some general ESP-IDF example been flashed.

I think, you can just back this code and copy the latest master code here (I really suggest handling this through some kind of version system, as otherwise it would become hard to manage for you).
and build and flash latest master slave code into ESP32.

reboot whole machine and get the logs at botth sides.

jankens, 2024-08-22:
I think I will not flash to the lastest version of esp-hosted, cause it works well at the beginning packets. 
Can you help me to answer some question?
     1. what is the meaning of the register ESP_SLAVE_TOKEN_RDATA which is read from host driver every sent packet? 
     2. why is this register alway be increased its number, 
     3. can this register be clear / what action can clear this register on host site or esp32 site?

jankens, 2024-08-27:
@mantriyogesh
can you help me?

mantriyogesh, 2024-08-27:
Can you please reflash the firmware at ESP from master? Also please use master at host for kernel module build.

Once you have same version of both, they should be compatible.

Get the logs with this arrangement, to debug effectively