espressif/esp-hosted #460: esp32-c5 sdio slave

skylin008, 2024-08-17:
Thank you for shared impressive project. I want to used wifi 2.4G/5G , the only esp32c5 compatible.So, I used the esp32-p4 chip for mcu host , and esp32c5 via sdio connected to esp32-p4 sdio port. How to esp32c5 sdio for sdio slave. Thanks!

mantriyogesh, 2024-08-17:
Esp32-C5 mass production chipset doesn't have SDIO slave.

Although QSPI would be supported on C5.

skylin008, 2024-08-17:
Thanks! Does QSPI can be support SPI slave?

mantriyogesh, 2024-08-17:
Yes c5 is supported as qspi slave.
But yet formal support in esp-hosted slave is not integrated. it is dependent upon full support (minimally, wifi+spi slave together) in esp-idf.
Last time we tested individually wifi and spi slave, it worked on different internal branches, but we would check again and come back,  if full support is available on ESP-IDF GitHub or not.

Until ESP-IDF support is not public, ESP-Hosted would be blocked upon.

mantriyogesh, 2024-08-19:
Hello @skylin008 ,

We already have QSPI driver in place , but only for MCU host.
QSPI frame structure in use could be referred at https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/spi_half_duplex.md#3-spi-hd-protocol-used-by-hosted

Although they are configurable, the host and slave need to refer to same frame structure. If host has some different structure (at hardware), slave code could be modified accordingly.

QSPI (half duplex SPI) is currently only supported for MCUs. We would add ESP32-C5 as slave shortly in this.
Similarly,  For Linux also, the QSPI framing would be important factor, which we would try to generalise while developing Linux SQPI driver.

full duplex spi driver, with single data line (MISO and MOSI) support for C5 is ongoing, we would have some clarity in next week.

skylin008, 2024-08-20:
Thank you @mantriyogesh 。

skylin008, 2024-11-05:
Thanks @mantriyogesh . Please Close this issue.

skylin008, 2024-11-05:
@mantriyogesh Is there C5 slave function about hardware pin define, I will compatible with new hardware design.    Thanks!

mantriyogesh, 2024-11-05:
Right now I am not sure about this. Once I have chip in my hand, I can clarify. But we would try to retain the numbers and config as much as we can.

In general if the pin numbers would remain same from hardware point off view, unless it is extremely complicated or rate scenario

closed 2024-11-05 by mantriyogesh