espressif/esp-hosted #462: ESP AS HOST sop needed

Speakpig, 2024-08-20:
I am new to ESP, I want to test the branch esp_as_mcu_host instance with two ESP-C3 development boards in my hands.  Having read the introduction of the file for the specific operation steps , I am still very confused how to start. My computer has installed ESP-IDF, and can normally use CMD related commands, the following are my detailed questions:
1. Do I need to create the host ESP-IDF project or use the host folder. If I do, how to create?
And when shall I use the following instruction?
"idf.py add-dependency "espressif/esp_wifi_remote"
"idf.py add-dependency "espressif/esp_hosted"
2. Similarly, how do I create a slave ESP-IDF project? How to use slave folder?
And when to use the following instruction?
"idf.py create-project-from-example espressif/esp_hosted:slave"

mantriyogesh, 2024-08-20:
@SohKamYung-Espressif ++

SohKamYung-Espressif, 2024-08-20:
Hello @Speakpig 

I will take a look at steps needed to get you going. We usually target the esp_as_mcu_host branch for people who are using two different ESP chips as the host and slave. Using ESP32-C3s as both the host and slave wasn't considered, so some testing needs to be done to see if this is possible.

> I am new to ESP

As a start, you can familiarise yourself with ESP-IDF first. You can look through the ESP-IDF `examples/wifi/` folder for Wi-Fi examples, build them for one C3 first to see how Wi-Fi works on ESP.

Speakpig, 2024-08-20:
Thanks for ur tips, I basically understand the WI-FI examples, I just never really built a prj based on this branch, so I acctually want more to know how to practice, I will be waiting for ur further advice, much thanks.

SohKamYung-Espressif, 2024-08-20:
The steps to create a host under ESP-Hosted are covered in Section 3 of the readme [ https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host?tab=readme-ov-file#3-preparing-the-host-to-use-esp-hosted ].

For example, suppose you want to convert the ESP-IDF `examples/wifi/scan` to use ESP-Hosted. Inside that directory, you run:

```bash
idf.py add-dependency "espressif/esp_wifi_remote"
idf.py add-dependency "espressif/esp_hosted"
```

If you do `ls` (to list the directory), you will now find a new `managed_components` directory. Inside that directory, you will find two subdirectories, `espressif__esp_wifi_remote` and `espressif__esp_hosted`.

You will also find a `idf_component.yml` file in the `main` subdirectory which will show that the project now depends on those two components. You do not need to edit or change the file.

That is all that is needed to add ESP-Hosted to your project as a host. However, there is one further step, `3.2. Disabling Native Wi-Fi Support`, which does not work for you because you want to use a ESP32-C3 as both a host and a slave device under ESP-Hosted. I will need some time to work on this so you can use a ESP32-C3 as the host.

> Similarly, how do I create a slave ESP-IDF project? How to use slave folder?
> And when to use the following instruction?
> "idf.py create-project-from-example espressif/esp_hosted:slave"

For this step, go to an empty directory and run

```bash
"idf.py create-project-from-example espressif/esp_hosted:slave"
```

This will create the ESP-Hosted slave project. Configure the target device (as ESP32-C3) and build it like you use any other ESP-IDF project. This is the code that will be flashed to the ESP32-C3 that will act as the slave under ESP-Hosted.

Speakpig, 2024-08-20:
Thanks a lot, this is so detailed for me, but I still have one big doubt:
the "host" and "slave" folders in this branch are not envolved in the opration u mentioned, how is this working?

Speakpig, 2024-08-20:
> the "host" and "slave" folders in this branch are not envolved in the opration u mentioned, how is this working?  

Should I do some replacement inside that directory formed by instuctions using the `host` or `slave `folder？

SohKamYung-Espressif, 2024-08-20:
> the "host" and "slave" folders in this branch are not envolved in the opration u mentioned, how is this working?

I'm not sure what you mean. Would the diagram from the README, help to make this clearer:

https://github.com/espressif/esp-hosted/raw/feature/esp_as_mcu_host/docs/images/hosted_diagram-ditaa.svg

For 'host', ESP-Hosted is the ESP Host Driver. It is meant to be used as an component that is added to your host code that requires Wi-Fi access.

For 'slave' under ESP-Hosted, there should be a `slave` folder created in the empty directory. This is the complete code that runs on the slave.

You should not need to replace or add any `host` folders. When building ESP-IDF will detect the presence of the components in `managed_components` and use them if required.

SohKamYung-Espressif, 2024-08-20:
> the "host" and "slave" folders in this branch are not envolved in the opration u mentioned, how is this working?

I think I understand your question now. You are asking how the `esp_as_mcu_host` github branch is related to the steps to add the ESP-Hosted component to the host code and getting the slave project?

It is a bit of GitHub and ESP-IDF Component Manager 'magic'. When triggered, the IDF Component Manager packages the ESP-Hosted code from github and makes a release. This release is fetch from the IDF Component Manager when you run `idf.py add-dependency` and `idf.py create-project-from-example`.

Speakpig, 2024-08-20:
[image]  
I understand what you're saying,  I'm sorry not being clear enough, actually, this is what I'm asking about:
how should these two folders be made use of, I believe they are not mentioned in ur clarification which must be useful in some way?

SohKamYung-Espressif, 2024-08-20:
> how should these two folders be made use of, I believe they are not mentioned in ur clarification which must be useful in some way?

The code on github is meant more for ESP-Hosted developers. You should not have to do `git clone` to use ESP-Hosted. IDF Component Manager takes care of fetching the code and putting it in the correct place for host and slave.

Speakpig, 2024-08-20:
Thx, I just missed the new comment befor asking, I deleted last msg, I now have got it! Problem solved, that's so kind of you!
But this leads me to a new one:

>  When you do idf.py create-project-from-example, it creates a copy of the code in the slave directory you see in the esp_as_mcu_host branch.

Is this copy from esp_as_mcu_host branch or master?

SohKamYung-Espressif, 2024-08-20:
> Is this copy from esp_as_mcu_host branch or master?

From the esp_as_mcu_host branch.

Speakpig, 2024-08-20:
Why not from a specified tag or release version?

Speakpig, 2024-08-20:
> However, there is one further step, 3.2. Disabling Native Wi-Fi Support, which does not work for you because you want to use a ESP32-C3 as both a host and a slave device under ESP-Hosted. I will need some time to work on this so you can use a ESP32-C3 as the host.

When coming to this, why shall I do `step 3.2` by:

> To do this, edit the ESP-IDF
`components/soc/<host soc>/include/soc/Kconfig.soc_caps.in` file and change
all `WIFI` related configs to `n`. 

Can I configure this by modify sdkconfig and rebuild the project or directly modify sdkconfig.h in directory `build\config\sdkconfig.h`?

SohKamYung-Espressif, 2024-08-21:
> Why not from a specified tag or release version?

As mentioned, there is some 'magic' in the background that tells the IDF Component Manager to do a release from the GitHub source.

If you check the IDF Component Page for ESP-Hosted https://components.espressif.com/components/espressif/esp_hosted/ there are various version releases, currently at 0.0.9. Each release is linked to a particular commit in the GitHub repository.

SohKamYung-Espressif, 2024-08-21:
> Can I configure this by modify sdkconfig and rebuild the project or directly modify sdkconfig.h in directory build\config\sdkconfig.h?

I have not looked at this option yet due to other work. You are welcome to try, but I don't know what the results will be.

Speakpig, 2024-08-21:
Thx, as now you are sure about ESP32-C3 is not available for host, I will waiting ur further sugguestion meanwhile I wil be trying meself, thanks a lot!

SohKamYung-Espressif, 2024-08-21:
> Thx, as now you are sure about ESP32-C3 is not available for host

For now, until I have some time to look at it.

Do you have another ESP32-x (not a C3) which can be used as the host?

Speakpig, 2024-08-21:
Sry, two C3s are all I got, and I have done prepared host and slave projects and connected the C3s refering to #186 , ready to go, but waiting for more guidance, am I doing right for now?
[image]

HP4LEVI, 2024-08-21:
@Speakpig 您好，请问您的host主机工程编译通过了吗？

Speakpig, 2024-08-21:
@HP4LEVI Yes, it has passed the build process.

SohKamYung-Espressif, 2024-08-21:
> Yes, it has passed the build process.

Did you build the code for host after you did this:

> Can I configure this by modify sdkconfig and rebuild the project or directly modify sdkconfig.h in directory build\config\sdkconfig.h?

Speakpig, 2024-08-21:
I am using the `VSCode extension for IDF`, in the configure page(PIC 1), no configs like SOC_WIFI_xxx were found, so I haven't done that yet, but I am about to try modify the sdkconfig.h directly, there were 10 related(PIC 4) configs found in two files(one under directory `build\bootloader` PIC 2,one under directory `build`PIC 2 3), I'm not sure which to modify and if it's ok doing so.
[image]
[image]
[image]
[image]

Speakpig, 2024-08-21:
> 你好，能说明一下您的编译操作吗？我的编译显示配置文件有问题，但我的配置文件打不开。

Have u formed ur host project? Then I did this using `VSCode extension for IDF` as below.
[image]

HP4LEVI, 2024-08-21:
[image]
我直接使用vscode打开了这个文件，然后进行了组件下载，这样对吗？
[image]

Speakpig, 2024-08-21:
@HP4LEVI We were doing it quite the same, I just opened the host prj, and then just built it, all things were done automatically, it should be ok, I don't know what u are running into, what's the problem？

HP4LEVI, 2024-08-21:
[image]
我甚至都不能打开配置文件

Speakpig, 2024-08-21:
@Speakpig Add my wechat: `stay1daybabe` , let's discuss this.

Speakpig, 2024-08-21:
> but I am about to try modify the sdkconfig.h directly, there were 10 related(PIC 4) configs found in each two files(one under directory build\bootloader PIC 2,one under directory buildPIC 3), I'm not sure which to modify and if it's ok doing so.

@SohKamYung-Espressif If I want to do this, what is ur suggestion？Which file or both shall be changed out of the best consideration? Thanks a lot.

SohKamYung-Espressif, 2024-08-21:
@Speakpig  Sorry, I can't make any suggestions now, as I haven't tried it yet.

I have to work on another task first.

Speakpig, 2024-08-21:
Thanks, I will try myself first, but if u got any progress in C3 as host, let me know, much thanks.

mantriyogesh, 2024-08-22:
Although it is not a fix but it could be treated a workaround to just get you going:
#### 1. Slave C3 flashing
- Build the slave and flash the slave c3 first, **without** ESP-IDF changes
- If any IDF repo changes, revert them, to get clean ESP-IDF GitHub repo state
- Slave project sdkconfig generated after 'set-target' would contain these lines (not in sequence)
- Flash and monitor slave 
```
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_ESP_WIFI_ENABLED=y
```

#### 2. Host C3 flashing
- Apply `3.2 Disabling Native Wi-Fi Support` for Host C3
- Now host c3 project sdkconfig generated after 'set-target' would contain these lines (not in sequence)
```
CONFIG_SOC_WIFI_SUPPORTED=n
CONFIG_ESP_WIFI_ENABLED=n
```
- Flash and monitor host C3 

> [!Note]
> These steps might only be required if you have same chipset type as host and slave ESP
> In other normal cases, these steps not required

Speakpig, 2024-08-22:
@mantriyogesh I entirly got u, I should edit the local ESP-IDF repo applying  `3.2 Disabling Native Wi-Fi Support` after I built the slave, which may causing incorrect slave project, but I have to do this for host, right? My confusion is can I do this as below when configuring host prj, which is more convenient for the same chipset type as host and slave ESP as I doubt before?

> but I am about to try modify the sdkconfig.h directly, there were 10 related(PIC 4) configs found in each two files(one under directory build\bootloader PIC 2,one under directory buildPIC 3), I'm not sure which to modify and if it's ok doing so.

Speakpig, 2024-08-22:
And as it is mentioned by SohKamYung, specificly what is the reason C3 could not be a host? Is it like that when using another slave chip, C3 could be a host? So it is just a matter of configuration, right?

> which does not work for you because you want to use a ESP32-C3 as both a host and a slave device under ESP-Hosted

mantriyogesh, 2024-08-22:
sdkconfig.h cannot be changed, it is generated file and some of the configuration would still be loaded assuming wifi is enabled, which is not really correct. Some of the components might still think that the wifi is available.

Instead, I would really keep different directories for host and slave. And use above comment to evaluate.

As IDF changes would be done, 
```
1. rm -rf sdkconfig build
2. Change IDF source as your needs
3. idf.py -p <usb_port> set-target esp32c3 # sdkconfig.h generated here
4. idf.py -p <usb_port> flash monitor 
```

This would make sure the sdkconfig.h is correctly co-related with all configurations.

mantriyogesh, 2024-08-22:
C3 or for that matter, any ESP chipset can be used as host. Problem right now you face is both conditions satisfied at the same time:
1. slave and host is same type of chipset
2. Host chipset already supports wifi natively

Speakpig, 2024-08-22:
@mantriyogesh OK, I will try and let u know, thx.

SohKamYung-Espressif, 2024-08-28:
@Speakpig I have taken a look and the solution proposed by @mantriyogesh in https://github.com/espressif/esp-hosted/issues/462#issuecomment-2303465083 is the solution that works for your case.

My attempts at resolving it by modifying the `sdkconfig.default` file for the host did not work. And as Yogesh as mentioned, modifying the generated `sdkconfig.h` file may not work as expected.