espressif/esp-hosted #463: Advertisement signal strenght

luisCactuasys, 2024-08-21:
Hello,

I've been using the ESP-hosted solution in my Linux project and was wondering if it's possible to change the advertisement signal strength. Here are the details of my setup:

-ESP32-C3 spi_only
-Kernel: 5.15
-Bluez 5.55
I searched for documentation on this topic but couldn't find much information. I also attempted a few commands to see if this was possible, but from what I observed, it doesn't seem to be. However, I would appreciate your confirmation on this.

```
root@beaglebone-yocto:~# hciconfig hci0 inqtpl
Can't read inquiry transmit power level on hci0: Input/output error (5)
root@beaglebone-yocto:~# hciconfig hci0 inqtpl 4
Can't set inquiry transmit power level on hci0: Input/output error (5)
root@beaglebone-yocto:~# bccmd psget 0x0017
Unsupported manufacturer
```
As far as I know, ```hcitool cmd``` can also be used for this purpose. In the past, I've worked with hardware where the following command was used:

```hcitool cmd 0x3F 0x026 0xFE 0x00 <value>```

Here, ```value``` was a multiple of 4 within the range [0xF0, 0x0C]. However, this didn't work in my current setup, likely because it’s a manufacturer-specific command and the **OCF, OGF and OPCODE** will differ.

To notice, I am more interested in Advertisment Power Level that in the Transmission Power Level.

Thank you!

mantriyogesh, 2024-09-02:
Check https://github.com/nkolban/esp32-snippets/issues/674:

[esp_ble_tx_power_set()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv420esp_ble_tx_power_set20esp_ble_power_type_t17esp_power_level_t) has type in first argument.

Although, you need to change this locally at slave.

Host doesn't have a specific rpc command for this. yet. if you verify and confirm, we can add this command for you (you also can add on yourself, as code is available for you).

luisCactuasys, 2024-09-12:
Hi @mantriyogesh 

Apologies for the delayed response!

I’ve tested the [esp_ble_tx_power_set()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv420esp_ble_tx_power_set20esp_ble_power_type_t17esp_power_level_t) function, and it works exactly as I hoped. For this initial test, I used it directly in the ```recv_task``` .
 
Now, I’m trying to find the best way to implement this feature so I can adjust the power level using Bluez tools.

The challenge I’m facing is identifying where the BLE commands are processed. I managed to trace the code to this section in esp-idf, but beyond this, it’s unclear to me:

```
void esp_vhci_host_send_packet(uint8_t *data, uint16_t len)
{
    if (btdm_controller_status != ESP_BT_CONTROLLER_STATUS_ENABLED) {
        return;
    }
    async_wakeup_request(BTDM_ASYNC_WAKEUP_SRC_VHCI);

    API_vhci_host_send_packet(data, len);

    async_wakeup_request_end(BTDM_ASYNC_WAKEUP_SRC_VHCI);
}
```

I’m considering simply adding a few lines to network_adapter and reserving a specific command sequence to handle my request. Something like:

`hcitool cmd 0x3F 0x026 0xFE 0x00 <value>
`
I can see the command in the ESP debug output:

> I (41825) H->S BT: 0x3fcba786   26 fc 03 fe 00 08                                 |&.....|

t doesn’t seem too complicated, but I’m wondering if this approach is appropriate? Would love to get your thoughts.

Best regards
Luís