espressif/esp-hosted #469: why networkmanager doesn't show espsta0 network

wangzipai, 2024-08-28:
I can see the ESPsta0 network card using ifconfig, as follows:
# ifconfig
espsta0   Link encap:Ethernet  HWaddr D2:CE:93:AC:13:04
          inet addr:192.168.110.13  Bcast:192.168.110.255  Mask:255.255.255.0
          inet6 addr: fe80::ba7f:65bd:a854:a5ef/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:277 errors:0 dropped:0 overruns:0 frame:0
          TX packets:105 errors:0 dropped:34 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:47048 (45.9 KiB)  TX bytes:10804 (10.5 KiB)

eth0      Link encap:Ethernet  HWaddr 2A:23:C8:2A:FD:61
          inet addr:192.168.100.199  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:999 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2527 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:218464 (213.3 KiB)  TX bytes:269170 (262.8 KiB)
          Interrupt:52 Base address:0xc000

eth1      Link encap:Ethernet  HWaddr AE:39:9E:A9:4D:CB
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:54 Base address:0x8000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:63885 errors:0 dropped:0 overruns:0 frame:0
          TX packets:63885 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4283043 (4.0 MiB)  TX bytes:4283043 (4.0 MiB)

But it seems that it is not displayed in Network Manager.as follows:
NAME                UUID                                  TYPE      DEVICE
Wired connection 1  a6a38eae-8ad2-3b98-8b54-60caa1a701e2  ethernet  --
Wired connection 2  7ef97620-73be-384b-aa5a-1236983a4652  ethernet  --

mantriyogesh, 2024-08-28:
@kapilkedawat ++

kapilkedawat, 2024-09-02:
Hi @wangzipai, could you please check if it shows other wifi adapters?