espressif/esp-hosted #470: NG mode, esp firmware and hosted driver  can’t be compatible with older versions

lwj850533527, 2024-08-29:
Hi @kapilkedawat.
HI @Shreyas0-7.
When I upgrade separately, only esp firmware or hosted driver, the esp32 unable to communicate host.
How should I maintain my old product

```
[    9.323200] esp32_spi: loading out-of-tree module taints kernel.             
[    9.542096] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [30]                                           
[   12.641884] esp32_spi: process_esp_bootup_event: Received ESP bootup event   
[   12.647475] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3         
[   12.653806] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI                                                                        
[   12.661531] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2         
[   12.667948] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0         
[   12.674251] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1         
[   12.680653] esp32_spi: process_fw_data: Length not matching to firmware data size                                                                            
[   12.688089] esp32_spi: process_event_esp_bootup: failed to process tag=1 in bootup event   
```

kapilkedawat, 2024-08-29:
@lwj850533527 suggest to rebase your code.. also may we ask what kind of changes do you have in host/firmware?

lwj850533527, 2024-08-29:
Hi @kapilkedawat.
I upgrade separately only esp firmware, but hosted driver didn't upgrade.
I think the issue because of the commit https://github.com/espressif/esp-hosted/commit/2e6603687a2dab44e1ef4a6fdf615b61b2b4bd3a

[image]

kapilkedawat, 2024-08-29:
Hi @lwj850533527, Please don't only upgrade firmware.. host and firmware always supposed to be upgraded in sync..

lwj850533527, 2024-08-29:
Hi @kapilkedawat.
At present, there is no upgrade channel for firmware.

How should I maintain my old product

Esp firmware and hosted driver can’t be compatible with older versions.

kapilkedawat, 2024-08-29:
@lwj850533527 will share OTA patch with you.

@Shreyas0-7 please help..

Shreyas0-7, 2024-08-29:
Hey @lwj850533527 please find the below patch
[hosted_ota.zip](https://github.com/user-attachments/files/16799597/hosted_ota.zip)
This patch has changes for fw too, please flash these changes for firmware once
Later on ota can be done by using host using this command 
`./rpi_init.sh ota_file=/path/to/ota_file"`

closed 2024-09-30 by lwj850533527