espressif/esp-hosted #473: NG mode, AP not found to connect.

lwj850533527, 2024-09-02:
Hi @kapilkedawat.

During the connection process, sometimes authentication may return a failure
The latest code has already set auth_type, Can the scan action be deleted

[image]
[image]
[image]

lwj850533527, 2024-09-10:
Hi @kapilkedawat.
HI @Shreyas0-7.
Is there any attempt to solve this problem?

kapilkedawat, 2024-09-10:
Hi @lwj850533527, apologies for the delay. Its most likely again a porting issue on your side, if its not able to scan, it definitely won't be able to connect. Could you please check if there are some local changes which may be causing this.

lwj850533527, 2024-09-11:
Hi @kapilkedawat.
The code of this file has not been modified in porting.
At the same time, I tried to write the auth_type as WPA-PSK when “AP not found to connect” happen. It can connect AP well.

[image]
[image]

lwj850533527, 2024-09-11:
Hi @kapilkedawat.
I have set the scan type to WIFI_SCAN_TYPE_ACTIVE (=0), which seems to reduce the probability of this issue

[image]

lwj850533527, 2024-09-11:
Is there any other solution to directly pass auth_type from host. 
Thoroughly solve the issue

Shreyas0-7, 2024-09-11:
Hey @lwj850533527 it might be because we while we are doing passive scan we are not able to find the AP. Will check if we can et it to active scanning.
Till that time can you please inform us on what channel AP is on. Also if possible can you please share wpa_supplicant logs so that we can check if we are receving probe response from the AP or not.
`wpa_supplicant -c "conf_file" -i "interface" -dd `

lwj850533527, 2024-09-11:
This is the log of wpa_supplicant and ESP32 host.
There have been three times “AP not found to connect” here

[image]

```

Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'REMOVE_NETWORK 0'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: REMOVE_NETWORK id=0
Sep 11 14:47:52 wpa_supplicant[1534]: RSN: Flush PMKSA cache entry for d4:3a:2e:d2:4a:d8
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: PMKSA-CACHE-REMOVED d4:3a:2e:d2:4a:d8 0
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendmsg: sock=15 sndbuf=180224 outq=0 send_len=39
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE monitor sent successfully to /tmp/wpa_ctrl_1537-1\x00
Sep 11 14:47:52 wpa_supplicant[1534]: nl80211: Delete PMKID for d4:3a:2e:d2:4a:d8
Sep 11 14:47:52 wpa_supplicant[1534]: nl80211: NL80211_CMD_DEL_PMKSA failed: -95 (Operation not supported)
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: RSN: PMKSA cache entry free_cb: d4:3a:2e:d2:4a:d8 reason=0
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendmsg: sock=15 sndbuf=180224 outq=704 send_len=28
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE monitor sent successfully to /tmp/wpa_ctrl_1537-1\x00
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=1408 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'ADD_NETWORK'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: ADD_NETWORK
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendmsg: sock=15 sndbuf=180224 outq=0 send_len=26
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE monitor sent successfully to /tmp/wpa_ctrl_1537-1\x00
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=704 send_len=2
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'SET_NETWORK [REMOVED]'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: SET_NETWORK id=0 name='ssid'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: value - hexdump(len=11): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: ssid - hexdump(len=9): 65 73 73 65 6e 74 69 65 6c
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=0 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'SET_NETWORK [REMOVED]'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: SET_NETWORK id=0 name='scan_ssid'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: value - hexdump(len=1): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: scan_ssid=1 (0x1)
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=0 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'SET_NETWORK [REMOVED]'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: SET_NETWORK id=0 name='psk'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: value - hexdump(len=10): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: PSK (ASCII passphrase) - hexdump(len=8): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: PSK (from passphrase) - hexdump(len=32): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=0 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'SET_NETWORK [REMOVED]'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: SET_NETWORK id=0 name='key_mgmt'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: value - hexdump(len=11): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: key_mgmt: 0x402
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=0 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'SET_NETWORK [REMOVED]'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: SET_NETWORK id=0 name='ieee80211w'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: value - hexdump(len=1): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: ieee80211w=1 (0x1)
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=0 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'ENABLE_NETWORK 0'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: ENABLE_NETWORK id=0
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=0 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'SAVE_CONFIG'
Sep 11 14:47:52 wpa_supplicant[1534]: Writing configuration file '/etc/wpa_supplicant.conf.tmp'
Sep 11 14:47:52 wpa_supplicant[1534]: Configuration file '/etc/wpa_supplicant.conf' written successfully
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: SAVE_CONFIG - Configuration updated
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL-DEBUG: ctrl_sock-sendto: sock=15 sndbuf=180224 outq=0 send_len=3
Sep 11 14:47:52 wpa_supplicant[1534]: wlan0: Control interface command 'SELECT_NETWORK 0'
Sep 11 14:47:52 wpa_supplicant[1534]: CTRL_IFACE: SELECT_NETWORK id=0
Sep 11 14:47:52 wpa_supplicant[1534]: SAE: Derive PT - group 19
Sep 11 14:47:52 wpa_supplicant[1534]: SAE: SSID - hexdump(len=9): 65 73 73 65 6e 74 69 65 6c
Sep 11 14:47:52 wpa_supplicant[1534]: SAE: password - hexdump(len=8): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: SAE: pwd-seed - hexdump(len=32): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: SAE: pwd-value (u1 P1) - hexdump(len=48): [REMOVED]
Sep 11 14:47:52 wpa_supplicant[1534]: SAE: u1 - hexdump(len=32): [REMOVED]
[... cut ...]
Sep 11 14:48:03 udhcpc[1549]: started, v1.34.1
Sep 11 14:48:03 udhcpc[1549]: broadcasting discover
Sep 11 14:48:03 udhcpc[1549]: broadcasting select for 192.168.2.134, server 192.168.2.100
Sep 11 14:48:03 udhcpc[1549]: lease of 192.168.2.134 obtained from 192.168.2.100, lease time 43200
Sep 11 14:48:03 kernel: [16772.692441] esp32_spi: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.002.134
Sep 11 14:48:04 wpa_supplicant[1534]: EAPOL: startWhen --> 0
Sep 11 14:48:04 wpa_supplicant[1534]: EAPOL: disable timer tick

```

Shreyas0-7, 2024-09-12:
Hey @lwj850533527 we have merged a change where we are trying to do active scan, can you please check if you are still facing the issue after that change.

lwj850533527, 2024-09-13:
HI @Shreyas0-7.
After that change, I haven't faced this issue yet.
This scanning action is to get the auth_type, can it be transmitted from the host

Shreyas0-7, 2024-09-16:
Hey @lwj850533527 the auth_type which we receive from host has different context than what we are setting for our esp wifi config for esp_wifi_connect.

closed 2024-09-18 by lwj850533527