espressif/esp-hosted #474: NG mode, disconnect event cannot report

lwj850533527, 2024-09-03:
Hi @kapilkedawat.

I use the latest code for testing. But the disconnect event cannot report
It can only report "WPA_EVENT_REGDOM_CHANGE" event. 
I deleted the change in this commit https://github.com/espressif/esp-hosted/commit/20c88e58ca59d1aaf0b807b77c907e2567b9fea6
It can report disconnect event normally

[image]

kapilkedawat, 2024-09-03:
Hi @lwj850533527, could you please share supplicant logs? https://github.com/espressif/esp-hosted/commit/20c88e58ca59d1aaf0b807b77c907e2567b9fea6 makes sure that supplicant doesn't get dummy deauth frame when disconnect was issued from userspace(this messes up supplicant's state machine sometimes). Also how are you disconnecting the station from AP?

lwj850533527, 2024-09-03:
> Also how are you disconnecting the station from AP?

I turned off the power of the router

kapilkedawat, 2024-09-03:
Please share supplicant logs and dmesg.

lwj850533527, 2024-09-03:
I tried to reproduce it, but it doesn't seem to certainly happen

kapilkedawat, 2024-09-03:
Please do share the logs when you are able to reproduce it. As I mentioned above, the patch above fixes a specific issue and decrease our total connection time in that case.

kapilkedawat, 2024-09-24:
@lwj850533527 can we close this if this issue no longer comes in your setup?

lwj850533527, 2024-09-25:
Hi @kapilkedawat.
OK.
We can reopen this issue if the next time same problem happen

closed 2024-09-25 by lwj850533527