espressif/esp-hosted #476: [NG Mode, iMX8QXP Host, SPI mode] It looks like was duplicated RSSI get command

nguyenlkdn, 2024-09-04:
Host
- iMX8QXP
- SPI Protocol
- Kernel: 5.10.9

Slave
- CHIP: ESP32S3
- Branch: master (70de43ac9cc30ef55960d8025274df7a0fe1dfc8)

> [   12.667950] esp32_spi: process_esp_bootup_event: Received ESP bootup event
> [   12.667980] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
> [   12.667988] esp32_spi: esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
> [   12.668146] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
> [   12.668152] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
> [   12.668157] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
> [   12.668162] esp32_spi: process_fw_data: ESP chipset's last reset cause:
> [   12.668167] esp32_spi: print_reset_reason: POWERON_RESET
> [   12.668176] esp32_spi: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
> [   12.675380] esp32_spi: esp_reg_notifier: Driver init is ongoing
> [   12.675471] prepare_command_request code: 1
> [   12.728909] prepare_command_request code: 3
> [   12.753634] prepare_command_request code: 15
> [   12.776917] esp32_spi: print_capabilities: Capabilities: 0xa0. Features supported are:
> [   12.776935] esp32_spi: print_capabilities:    * WLAN on SPI
> [   12.825941] prepare_command_request code: 15
> [   12.912454] prepare_command_request code: 15
> [   14.054514] [VPU Decoder] ctx[0] amount inconsistent between input(65) and output(63)
> [   17.481149] fda903d_trigger
> [   17.481185] fda903d_mute_stream: 1, 0
> [   17.482043] fda903d_shutdown
> [   21.932345] prepare_command_request code: 15
> [   21.957144] prepare_command_request code: 15
> [   21.984470] prepare_command_request code: 15
> [   22.001025] prepare_command_request code: 15
> [   22.023814] prepare_command_request code: 15
> [   22.174918] prepare_command_request code: 4
> [   24.048672] prepare_command_request code: 15
> [   24.620885] prepare_command_request code: 11
> [   24.620918] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
> [   25.066167] prepare_command_request code: 12
> [   25.066203] esp32_spi: cmd_assoc_request: Association request: 30:4f:75:8c:e0:08 1 35
> [   25.101023] esp32_spi: process_assoc_event: Connection status: 0
> [   25.101149] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for VN, current=\x00\x00
> [   25.101160] prepare_command_request code: 18
> [   25.118079] prepare_command_request code: 15
> [   25.193420] prepare_command_request code: 8
> [   25.221284] prepare_command_request code: 10
> [   25.245341] prepare_command_request code: 8
> [   25.293345] prepare_command_request code: 15
> [   25.576712] esp32_spi: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.100.014
> [   25.576728] prepare_command_request code: 13
> [   25.604678] prepare_command_request code: 14
> [   25.871987] prepare_command_request code: 27
> [   25.907116] prepare_command_request code: 27
> [   27.939059] prepare_command_request code: 27
> [   27.982755] prepare_command_request code: 27
> [   30.032484] prepare_command_request code: 27
> [   30.071107] prepare_command_request code: 27
> [   31.716048] SD1_SPWR: disabling
> [   32.114549] prepare_command_request code: 27
> [   32.140228] prepare_command_request code: 27
> [   34.174598] prepare_command_request code: 27
> [   34.206343] prepare_command_request code: 27
> [   36.238206] prepare_command_request code: 27
> [   36.270285] prepare_command_request code: 27
> [   38.305392] prepare_command_request code: 27
> [   38.349365] prepare_command_request code: 27
> [   40.390330] prepare_command_request code: 27
> [   40.424648] prepare_command_request code: 27
> [   42.474284] prepare_command_request code: 27
> [   42.513591] prepare_command_request code: 27
> [   44.550825] prepare_command_request code: 27
> [   44.598823] prepare_command_request code: 27
> [   46.634935] prepare_command_request code: 27
> [   46.673684] prepare_command_request code: 27
> [   48.709569] prepare_command_request code: 27
> [   48.745909] prepare_command_request code: 27

above get_rssi commands were executed very fast ~0.03ms, is it needed?

> [   25.871987] prepare_command_request code: 27
> [   25.907116] prepare_command_request code: 27

> [   40.390330] prepare_command_request code: 27
> [   40.424648] prepare_command_request code: 27

mantriyogesh, 2024-09-04:
@kapilkedawat ++

kapilkedawat, 2024-09-04:
@nguyenlkdn are you running any application besides wpa_supplicant which fetches RSSI regularly?

@Shreyas0-7 please help to check.

nguyenlkdn, 2024-09-04:
@kapilkedawat yes we are using command "iw dev wlan0 link" to query signal strength  with period is 2 seconds.

kapilkedawat, 2024-09-09:
@nguyenlkdn this is probably due to your own application query, meanwhile supplicant may also query the link status periodically due to that you may see this. Suggest you to get the link status from supplicant instead.

nguyenlkdn, 2024-09-09:
Thank for your infor, i will try to do not querry iw command to see

nguyenlkdn, 2024-09-13:
@kapilkedawat it seems like wpa hasn't trigger scan request
```
root@gridania-pvt1-soc:~# wpa_supplicant -u -Dnl80211 -iwlan0 -c/etc/wpa_supplicant.conf
Successfully initialized wpa_supplicant
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
rfkill: Cannot open RFKILL control device
dbus: fill_dict_with_properties dbus_interface=fi.w1.wpa_supplicant1.Interface.P2PDevice dbus_property=P2PDeviceConfig getter failed
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: SME: Trying to authenticate with d8:b3:70:b4:e4:ed (SSID='Ion Mobility VN' freq=2437 MHz)
wlan0: Trying to associate with d8:b3:70:b4:e4:ed (SSID='Ion Mobility VN' freq=2437 MHz)
wlan0: Associated with d8:b3:70:b4:e4:ed
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan0: WPA: Key negotiation completed with d8:b3:70:b4:e4:ed [PTK=CCMP GTK=CCMP]
wlan0: CTRL-EVENT-CONNECTED - Connection to d8:b3:70:b4:e4:ed completed [id=1 id_str=]
bgscan simple: Failed to enable signal strength monitoring

```

I am checking other sides

kapilkedawat, 2024-09-13:
Please use `wpa_cli signal_poll` to get connected AP's signal strength.