espressif/esp-hosted #477: esp32_sdio.ko

tgt008, 2024-09-05:
Hi  esp
I loaded the driver.But there's no device in sight.What's going on?

root@maaxboard8ulp:~/test#
root@maaxboard8ulp:~/test#
root@maaxboard8ulp:~/test#
root@maaxboard8ulp:~/test# insmod esp32_sdio.ko
root@maaxboard8ulp:~/test# iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

root@maaxboard8ulp:~/test#

mantriyogesh, 2024-09-05:
Please provide the related log. The issue is not clear enough. Please refer to [Porting Guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) for getting the correct bus setup. unless required things are set correctly, the bus cannot perform. As a result, unfortunately, any higher applications as you tried will just fail.

1. FG or NG
2. Linux details and kernel version
3. host log: dmesg from start of boot
4. esp log: minicom or idf.py monitor
5. git commit used at esp and host

mantriyogesh, 2024-09-08:
I fail to see any sdio hardware configured at host.

Please review your system's device tree config so that the sdio hardware is configured and detected.

Please review the porting guide steps earlier shared.

adityat-1812, 2024-09-10:
Hello @mantriyogesh,
I'm new at ESP32, and in my custom board I want to use ESP32 for wifi and bluetooth over the SDIO interface on yocto Linux.
Can you help me with integrating the device tree for my board and also the drivers.
I cant understand how to integrate it for my AM625 device.
Please help me with this.
Best Regards,
Aditya T

mantriyogesh, 2024-09-10:
Device tree customisation and configurations are out of scope for ESP-Hosted. They are actually requirements.

We genuinely wish to help, but every Linux system and device trees vary per Linux SoCs. And it needs some patience, if you are doing it for first time.

I think, it would be easier if you start with a commonly known or generally easier Linux SoC. As you would get device tree dumps in public domain. Obviously copying would not work, but at least you would get some idea.

In general it is easier to set up spi, as it also allows to use jumper cables of around 10cm.

sreejitharikath, 2024-09-10:
why iam not abled to generate virtual serial interface to set up wifi and set internet using esp hosted FG even after flashing corresponding repositories of FG in both raspberry pi 4 b and esp32 s3 wroom 1 devices 
dmesg | tail -n 50
[   12.268969] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video20
[   12.269712] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   12.269753] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   12.270757] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video21
[   12.282325] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video22
[   12.282900] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   12.282939] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   12.283479] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video23
[   12.283554] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[   12.283574] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[   12.283591] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[   12.283657] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[   12.283966] rpivid feb10000.codec: Device registered as /dev/video19
[   12.295543] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[   12.306115] alsactl[408]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   12.311881] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   12.311931] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   12.320233] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[   12.320276] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[   12.336739] bcm2835-codec bcm2835-codec: Device registered as /dev/video31
[   12.336787] bcm2835-codec bcm2835-codec: Loaded V4L2 encode_image
[   13.762696] Adding 204796k swap on /var/swap.  Priority:-2 extents:6 across:3465216k SS
[   15.709808] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   15.711050] bcmgenet fd580000.ethernet eth0: Link is Down
[   19.835969] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   82.233704] bcmgenet fd580000.ethernet eth0: Link is Down
[  204.596250] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[  204.627263] esp32_spi: loading out-of-tree module taints kernel.
[  204.627761] esp32_spi: esp_reset: Resetpin of Host is 518
[  204.627979] esp32_spi: esp_reset: Triggering ESP reset.
[  204.628110] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[534] DataReady[539]
[  204.628116] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[518] Handshake[534] Dataready[539]
[  204.628120] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
[  204.628128] esp32_spi: esp_spi_thread: esp spi thread created
[  485.863572] esp32_spi: esp_serial_cleanup: 
[  486.333378] esp32_spi: esp_spi_thread: esp spi thread cleared
[  499.956462] esp32_spi: esp_reset: Resetpin of Host is 518
[  499.956684] esp32_spi: esp_reset: Triggering ESP reset.
[  499.956817] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[534] DataReady[539]
[  499.956823] esp32_spi: esp_spi_thread: esp spi thread created
[  499.956823] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[518] Handshake[534] Dataready[539]
[  499.956827] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
[  634.947893] esp32_spi: esp_serial_cleanup: 
[  635.413359] esp32_spi: esp_spi_thread: esp spi thread cleared
[  649.013527] esp32_spi: esp_reset: Resetpin of Host is 518
[  649.013748] esp32_spi: esp_reset: Triggering ESP reset.
[  649.013898] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[534] DataReady[539]
[  649.013904] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[518] Handshake[534] Dataready[539]
[  649.013908] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
[  649.014189] esp32_spi: esp_spi_thread: esp spi thread created

currently i have commented the dts file, device_tree_dependency_spi()  and device overlay in boot/firmware/config.text and additional spi device overlay if spi in rpi.insit.sh  just make spi enable for getting this log.iam not able to make virtual serial interface espso  in fg set up
itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control $ bash -x ./rpi_init.sh wifi=spi
+ resetpin=518
+ clockspeed=10
+ spi_bus=0
+ spi_cs=0
+ spi_mode=2
+ spi_handshake=534
+ spi_dataready=539
+ cpu_perf=on
+ XTRA_MODULE_PARAMS=
+ BT_CONFIG=
+ WIFI_CONFIG=wifi_on
+ CROSS_COMPILE=/bin/arm-linux-gnueabihf-
++ uname -r
+ KERNEL_BUILD_DIR=/lib/modules/6.6.47+rpt-rpi-v7l/build
+ ARCH=
[... 259 lines cut ...]
+ log 'Entering insert_module'
++ date '+%Y-%m-%d %H:%M:%S'
+ echo '[2024-09-10 16:46:10] Entering insert_module'
[2024-09-10 16:46:10] Entering insert_module
+ '[' '' '!=' '' ']'
+ sudo insmod esp32_spi.ko resetpin=518 clockspeed=10 spi_bus=0 spi_cs=0 spi_mode=2 spi_handshake=534 spi_dataready=539
+ '[' 0 -ne 0 ']'
++ lsmod
++ grep esp32
++ wc -l
+ '[' 1 '!=' 0 ']'
+ log 'esp32 module inserted using script'
++ date '+%Y-%m-%d %H:%M:%S'
+ echo '[2024-09-10 16:46:10] esp32 module inserted using script'
[2024-09-10 16:46:10] esp32 module inserted using script
+ log_exit
+ log 'Exiting insert_module'
++ date '+%Y-%m-%d %H:%M:%S'
+ echo '[2024-09-10 16:46:10] Exiting insert_module'
[2024-09-10 16:46:10] Exiting insert_module
+ log '--- finished ---'
++ date '+%Y-%m-%d %H:%M:%S'
+ echo '[2024-09-10 16:46:10] --- finished ---'
[2024-09-10 16:46:10] --- finished ---
please help me in fixing this why after the module is inserted it is exiting why iam getting warning og memory lekage , why iam getting message that kernel is getting tainted

mantriyogesh, 2024-09-10:
@sreejitharikath , it is better to create new issue, so as not.to spam original issue. If you think they are related can mark related, but different issue would to support in better way.

adityat-1812, 2024-09-11:
> Device tree customisation and configurations are out of scope for ESP-Hosted. They are actually requirements.
>
> We genuinely wish to help, but every Linux system and device trees vary per Linux SoCs. [...]

But I've connected it using SDIO interface and now I don't have any other option than configuring the device using SDIO only

mantriyogesh, 2024-09-11:
Send a picture of the setup

adityat-1812, 2024-09-11:
> Can you please send a picture of the setup

The hardware setup isn't ready yet, but I want to enable it from software side before hardware is ready. 
Once hardware is ready I'll share the image of my setup

mantriyogesh, 2024-09-11:
You would be connecting the ESP with Linux host in some way?

adityat-1812, 2024-09-11:
> You would be connecting the ESP with Linux host in some way?

Yes, I'm going to connect it using SDIO interface which my AM625 has

mantriyogesh, 2024-09-11:
So is it jumpers? Is it tf card or is it PCB?

mantriyogesh, 2024-09-11:
Unless connected, how would sdio or ESP-Hosted work?

mantriyogesh, 2024-09-11:
>  I can use the ESP hosted NG on my linux host system

Definitely. You can use FG or NG. Please use same git commit at host and slave, preferably latest master.

So, ESP-Hosted NG requirements:
1. Linux host
2. Kernel version - 5.x + is preferred
3. SPI or SDIO bus at least on headers (For SDIO, can also use TF card slot, if you embed ESP on TF card)
4. For SDIO, external pull-up registers are mandatory. Values are present in porting guide.
5. If SDIO slave used on ESP32, additional requirement of eFuse burning might be required (simple command described in porting guide)

For more and accurate details, please refer [Porting guide for NG](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)

Detailed documentation is available at https://github.com/espressif/esp-hosted

mantriyogesh, 2024-09-11:
1. micro sd card slot

5. If you are asking the device tree changes for your Linux SoC, it is **out of scope** for esp-hosted. Check with your Linux SoC support channel.
Requirement for ESP-Hosted: SDIO platform is up with your Linux SDIO hardware configured using the configuration method provided by your Linux SoC.

Check [proting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) for hardware considerations and extra GPIO required to be configured in your Linux.

adityat-1812, 2024-09-11:
> 1 micro sd card slot
> 
> 5 If you are asking the device tree changes for your Linux SoC, it is **out of scope** for esp-hosted. Check with your Linux SoC support channel. Requirement for ESP-Hosted: SDIO platform is up with your Linux SDIO hardware configured using the configuration method provided by your Linux SoC.
> 
> Check [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) for hardware considerations and extra GPIO required to be configured in your Linux.

I've the micro SD card slot and I'm trying to remove the SD card connection and connect the ESP32 with it.

The community of Ti can't help me with because they doesn't support the ESP 32, so I need to ask you for helping me out.

Also how to Integrate the drivers into my kernel, like where to copy the .c files and what to add in whole Linux SDK?

mantriyogesh, 2024-09-11:
> Also how to Integrate the drivers into my kernel, like where to copy the .c files

Instead of in kernel module, use itt as external loadable module first, once tested fine, can embed as part of kernel modules.

> This is my device tree, will this work?

No harm in trying, I cannot surely say, it would work or not. check your SoC data sheet and use hardware config accordingly.

> I've the micro SD card slot and I'm trying to remove the SD card connection and connect the ESP32 with it.

How to connect , through GPIO header, or card slot or something else is upto you. SDIO has signal integrity issues if connected with jumpers. using proper PCB is only way o get SDIO correctly working. SPI could be used with wires for evaluation. SDIO 1 bit mode could be tried with jumper wires.

Length of jumper (<=6cm), quality, equal trace length for all signals and external pull-ups are mandatorily checked from **porting guide.**
If I were you, I wouldn't do any changes unless I refer the porting guide and hardware limitations understood clearly and fulfilled.

SPI on the other hand, should be available easily on headers and allows jumper cables with <=10cm length.
Obviously, it also needs some hardware considerations, but less strict than sdio. **Check porting guide for details**.

adityat-1812, 2024-09-11:
> Instead of in kernel module, use itt as external loadable module first, once tested fine, can embed as part of kernel modules.
>
>
> No harm in trying, I cannot surely say, it would work or not. check [...]

I'll use it first as loadable kernel module, but how to build it and keep the kernel module and where to keep it?
I'm directly soldering the ESP32's SDIO interface to my board's SD card interface also I've removed the SD card connector

mantriyogesh, 2024-09-11:
> I'll use it first as loadable kernel module, but how to build it and keep the kernel module and where to keep it?

upto you? Once cross compile done, copy the module binary and insert with correct module params

> I'm directly soldering the ESP32's SDIO interface to my board's SD card interface also I've removed the SD card connector

Again upto you. As long as the connections are done correctly as per porting guide, no issues.

adityat-1812, 2024-09-11:
> upto you? Once cross compile done, copy the module binary and insert with correct module params
>
>
> Again upto you. As long as the connections are done correctly as per porting guide, no issues.

Where to copy the module binary i.e. .ko file generated and what module parameters are you talking about?

mantriyogesh, 2024-09-11:
porting guide step:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#22-rpi_initsh

module param, 'resetpin'
https://github.com/espressif/esp-hosted/blob/20d939491fb20841ae0b56221f249be02ad0ac69/esp_hosted_ng/host/rpi_init.sh#L64-L69

where?
/root/ or any path which you are comfortable? Finally, it is a loadable module.

adityat-1812, 2024-09-11:
> porting guide step: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#22-rpi_initsh
>
> module param, 'resetpin' [...]

Well on my setup I'm going to use GPIO1_10 for reset pin, then in this case what value is needed to be given in variable RESETPIN?

mantriyogesh, 2024-09-11:
From your Linux data sheet, find out what exact gpio number 'GPIO1_10' is translated to. use that gpio number.

mantriyogesh, 2024-09-11:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#241-gpio-numbering-in-raspberry-pi-os

adityat-1812, 2024-09-13:
Hello,
Now I'm getting this error
`[  769.669173] mmc2: Timeout waiting for hardware cmd interrupt.
[  769.675009] mmc2: sdhci: ============ SDHCI REGISTER DUMP ===========
[  769.681467] mmc2: sdhci: Sys addr:  0x00000000 | Version:  0x00001004
[  769.687933] mmc2: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[  769.694394] mmc2: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[  769.700853] mmc2: sdhci: Present:   0x01f00001 | Host ctl: 0x00000001
[  769.707314] mmc2: sdhci: Power:     0x0000000a | Blk gap:  0x00000080
[  769.713774] mmc2: sdhci: Wake-up:   0x00000000 | Clock:    0x00004e43
[  769.720233] mmc2: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[  769.726692] mmc2: sdhci: Int enab:  0x00ff0003 | Sig enab: 0x00ff0003
[  769.733152] mmc2: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[  769.739612] mmc2: sdhci: Caps:      0x3de8c801 | Caps_1:   0x18002407
[  769.746073] mmc2: sdhci: Cmd:       0x0000371a | Max curr: 0x00000000
[  769.752534] mmc2: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[  769.758994] mmc2: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[  769.765452] mmc2: sdhci: Host ctl2: 0x00000000
[  769.769913] mmc2: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x0000000000000000
[  769.777069] mmc2: sdhci: ============================================
`
This line is getting printed continously

mantriyogesh, 2024-09-14:
Sorry, but your sdio is not setup yet.

Open data path or start data path is not printed.
This means that set up is yet not correct.

Share photo of setup

adityat-1812, 2024-09-16:
[image]
This is the photo of my setup, please let me know, what am I missing?

mantriyogesh, 2024-09-16:
1. Where is ESP chipset?

2. Wired setup is very unstable in 4-bit sdio mode. Use 1 -bit sdio instead. Configure your sdio in host (using device tree) accordingly.

3. Ext pull up registers mandatory

4. eFuse might also needed as you are using ESP32.

Unless hardware configured correctly it cannot work. Check porting guide for more details.

adityat-1812, 2024-09-16:
> 1. Where is ESP chipset?
>
>     2. Wired setup is very unstable in 4-bit sdio mode. Use 1 -bit sdio instead. Configure your sdio in host (using device tree) accordingly.
>
>     3. Ext pull up registers [...]

1. ESP chipset is on top side of ESP
2. How to use 1-bit SDIO? Also, why is my setup unstable.
    Configured the SDIO in host as
    `
[image]
`
3. Yes, I've added 10K Ohm registers for pull up
4. What is eFuse and how to use it?
Best Regards,
Aditya T

mantriyogesh, 2024-09-16:
I first doubt if your host is correctly configured with sdio correct.

Second, the sdio 4 bit mode face signal integrity issues over wires. Search this, it is applicable for any sdio 4 -bit (not ESP specific)

I couldn't spot ESP and pull up registers in you're setup.

eFuse burning details and pull up requirements: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

Depending upon ESP chipset used, Compatibility (problems) is something there with board type. Solutions for each section detailed in lower section.

mantriyogesh, 2024-09-16:
Why I say sdio not correctly configured:

Sdio should be detected by hardware first.

Probing hardware is different thing happens after detecting it. Probing would trigger ESP driver.

You are currently not detecting hardware itself.

Possible issues:
1. Hardware in configured in device tree
2. Connections not correct

Detecting sdio is not blocked on eFuse, pull up etc.

Once detection done probing would be triggered, but probing would fail if eFuse or pull up not correct.

Sometimes probing also goes fine but next transactions fail if eFuse and pull up not as expected.

You need to focus on sdio being detected.

Just to reconfirm, attach full dmesg from system bootup as a file.

adityat-1812, 2024-09-16:
> I first doubt if your host is correctly configured with sdio correct.
>
> Second, the sdio 4 bit mode face signal integrity issues over wires. Search this, it is applicable for any sdio 4 -bit (not ESP [...]

While building the dtb I didn't got any errors or warnings, How to edit it so the SDIO form host side get correctly configured.
Below attaching the code for configuring SDIO port for SD card which is working properly, then how to do it for wifi.
`&sdhci1 {
	/* SD/MMC */
	vmmc-supply = <&vdd_mmc1>;
	vqmmc-supply = <&vdd_sd_dv>;
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&main_mmc1_pins_default>;
	disable-wp;
	no-1-8-v;
	sdhci-caps-mask = <0x00000007 0x00200000>;
};`
Attached the image of ESP32 and pull ups over it
[image]
I'll check the eFuse details.

Logs of Host [LINUX]: 
[Linux-logs.txt](https://github.com/user-attachments/files/17009147/Linux-logs.txt)
Logs of Slave [ESP32]:
[ESP-logs.txt](https://github.com/user-attachments/files/17009150/ESP-logs.txt)

mantriyogesh, 2024-09-16:
I am not entirely sure, why would you re-use sdmmc1 , as i can see from traces, i somehow sense that you are using mmc1 already for booting up linux.
[image]

From you device tree you seem to use sdhci1.
While I cannot verify, I am just asking you to ensure that you are doing it correctly.
I will leave to you to get the correct sdio hardware 'detectable'.

How would one confirm?

You would see similar trace:
[image]

linux would detect and print something like this,

mmcxX: new high speed SDIO card at address 0001

or

mmcX: new SDIO card at address 0001

or something that sorts.

if this first step is failed, no point going ahead.

I would suggest to read and understand which mmcX is mapped where from your Linux SoC and then try to load that one in your device tree. If you already know what you are doing and sure of it, please disregard this comment in that case.

mantriyogesh, 2024-09-16:
The question remains, then why sdhci1 ? :

[image]

adityat-1812, 2024-09-16:
> The question remains, then why sdhci1 ? :
> 
> ![Screenshot 2024-09-16 at 3 13 44 PM](https://private-user-images.githubusercontent.com/67477199/367672165-eb75f88a-2c39-44a7-95db-207658e7ac6d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY0NzExODEsIm5iZiI6MTcyNjQ3MDg4MSwicGF0aCI6Ii82NzQ3NzE5OS8zNjc2NzIxNjUtZWI3NWY4OGEtMmMzOS00NGE3LTk1ZGItMjA3NjU4ZTdhYzZkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE2VDA3MTQ0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUyNTk2YzFlNDUzY2U0M2IyMDczNWVkY2I4OGJkZmZjZWM3OTYzYjlkN2I1NGE1ZGU0MDdjZGIzYjgwMDQ2N2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.foFyxiea0FblrsKWAyeiDmJSR5JL5V2I0lUZ6WzZiqo)

Sorry, I'm using sdhci2, below is the code attached for sdhci2
`&sdhci2 {
	status = "okay";
	vmmc-supply = <&vcc_3v3_sys>;
	pinctrl-names = "default";
	pinctrl-0 = <&main_mmc2_pins_default>;
	bus-width = <4>;
	no-1-8-v;
	broken-cd;
	assigned-clocks = <&k3_clks 157 158>;
	assigned-clock-parents = <&k3_clks 157 160>;

	#address-cells = <1>;
	#size-cells = <0>;
	esp32: esp32@1 {
		compatible = "esp32_sdio";
		reg = <1>;
		
		pinctrl-names = "default";
		pinctrl-0 = <&main_wlirq_pins_default>;
	};
};
`

mantriyogesh, 2024-09-16:
It looks correct in general, however, repeating myself,

Device tree correction falls outside our area of focus. Please get your platform correctly configured with reference to your SoC hardware manual and/or datasheet.

We cannot assess your device tree as it is very specific to your hardware as such.

mantriyogesh, 2024-09-16:
I also don't know if you have correctly configured reset pin in your device tree. if not, i would suggest you to do so.
Reseting slave on module loading is important aspect for synchronisation (between host and slave) at sdio level, else you would fall in some unexpected issues.

adityat-1812, 2024-09-16:
Well, for reset pin, I've used 1 gpio for this purpose, below is how I set GPIO for reset pin
`root@am62xx-evm:~# lsmod | grep esp32
root@am62xx-evm:~# echo 1914 > /sys/class/gpio/export 
root@am62xx-evm:~# echo out > /sys/class/gpio/gpio1914/direction 
root@am62xx-evm:~# cat /sys/kernel/debug/gpio 
gpiochip1: GPIOs 1904-1955, parent: platform/601000.gpio, 601000.gpio:
 gpio-1914 (                    |sysfs               ) out lo 
 gpio-1953 (                    |am62-sk:green:heartb) out lo 

gpiochip0: GPIOs 1956-2047, parent: platform/600000.gpio, 600000.gpio:
 gpio-1987 (                    |tlv71033            ) out lo ACTIVE LOW
root@am62xx-evm:~# modprobe esp32_sdio resetpin=1914
root@am62xx-evm:~# lsmod | grep esp32
esp32_sdio             77824  0
bluetooth             507904  3 esp32_sdio
cfg80211              389120  1 esp32_sdio
root@am62xx-evm:~# ifconfig -a
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 1c:63:49:1c:04:8c  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.126  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::685a:b4ff:fe58:5aba  prefixlen 64  scopeid 0x20<link>
        ether 6a:5a:b4:58:5a:ba  txqueuelen 1000  (Ethernet)
        RX packets 1571  bytes 153414 (149.8 KiB)
        RX errors 0  dropped 125  overruns 0  frame 0
        TX packets 221  bytes 24806 (24.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 92  bytes 7654 (7.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 92  bytes 7654 (7.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@am62xx-evm:~#`

mantriyogesh, 2024-09-16:
for temporary, okay. I meanupto you how you wish to configure.

```
root@am62xx-evm:~# echo 1914 > /sys/class/gpio/export 
root@am62xx-evm:~# echo out > /sys/class/gpio/gpio1914/direction 
root@am62xx-evm:~# cat /sys/kernel/debug/gpio 
```

is only for current session and not persistent in my opinion.

How you expose hardware, is up to you. Just ensure, the hardware is correctly configured before loading esp32 kernel module **always**, to avoid any unexpected behaviour.

adityat-1812, 2024-09-16:
> for temporary, okay. I meanupto you how you wish to configure.
> 
> ```
> root@am62xx-evm:~# echo 1914 > /sys/class/gpio/export 
> root@am62xx-evm:~# echo out > /sys/class/gpio/gpio1914/direction 
> root@am62xx-evm:~# cat /sys/kernel/debug/gpio 
> ```
> 
> is only for current session and not persistent across reboots in my opinion.
> 
> How you expose hardware, is up to you. Just ensure, the hardware is correctly configured before loading esp32 kernel module **always**, to avoid any unexpected behaviour.

Yes, I need to set this GPIO for every power cycle of my Host.

> How you expose hardware, is up to you. Just ensure, the hardware is correctly configured before loading esp32 kernel module always, to avoid any unexpected behaviour.
This may not be an issue because I'm configuring this GPIO before loading the esp32_sdio.ko file