espressif/esp-hosted #478: [Question][NG Mode, iMX8QXP Host, SPI mode] NetworkManager Or WPA Suppliciant to managment wifi connection

nguyenlkdn, 2024-09-05:
Hi all!
I am comparing performance between NetworkManager and WPA Supplicant services to manage WiFi network interface for ESP NG

**Host**
- iMX8QXP
- SPI Protocol
- Kernel: 5.10.9

**Slave**
- CHIP: ESP32S3
- Branch: master (https://github.com/espressif/esp-hosted/commit/70de43ac9cc30ef55960d8025274df7a0fe1dfc8)

**Software**
- NetworkManager 1.22.14

I found an issues with NetworkManager to wifi connect to AP broascast on channel 1
It always connect failed with reason 201
[image]

```
nmcli -t dev wifi connect VIETTEL_AP_8CE000 password 1234567890a
Sep  5 11:44:18 gridania-pvt1-soc kernel: [ 2126.216462] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  5 11:44:20 gridania-pvt1-soc kernel: [ 2128.343267] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  5 11:44:20 gridania-pvt1-soc kernel: [ 2128.444054] prepare_command_request code: 4
Sep  5 11:44:25 gridania-pvt1-soc kernel: [ 2133.345838] prepare_command_request code: 6
Sep  5 11:44:25 gridania-pvt1-soc kernel: [ 2133.871614] prepare_command_request code: 4
Sep  5 11:44:27 gridania-pvt1-soc kernel: [ 2135.576906] prepare_command_request code: 11
Sep  5 11:44:27 gridania-pvt1-soc kernel: [ 2135.576937] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  5 11:44:29 gridania-pvt1-soc kernel: [ 2137.699479] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
```
After i changed AP to broadcase on other channel ex. 10
[image]

It was able to connect successfully
```
root@gridania-pvt1-soc:~# nmcli dev wifi connect VIETTEL_AP_8CE000 password 1234567890a
Device 'wlan0' successfully activated with 'c4f09538-d512-4005-aee1-bd2afac2f0e8'.
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.175164] prepare_command_request code: 6
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.224946] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:3]
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.225077] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=VN
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.225087] prepare_command_request code: 18
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.239659] prepare_command_request code: 15
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.256375] prepare_command_request code: 13
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.269801] esp32_spi: esp_inetaddr_event: Interface Down: 0
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.300516] prepare_command_request code: 15
Sep  5 11:55:31 gridania-pvt1-soc kernel: [ 2799.390447] prepare_command_request code: 4
Sep  5 11:55:33 gridania-pvt1-soc kernel: [ 2801.745847] prepare_command_request code: 4
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.454739] prepare_command_request code: 11
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.454771] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 10 0 0 0
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.906366] prepare_command_request code: 12
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.906400] esp32_spi: cmd_assoc_request: Association request: 30:4f:75:8c:e0:08 10 35
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.948415] esp32_spi: process_assoc_event: Connection status: 0
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.948536] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for VN, current=00
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.948549] prepare_command_request code: 18
Sep  5 11:55:35 gridania-pvt1-soc kernel: [ 2803.969629] prepare_command_request code: 15
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.043909] prepare_command_request code: 8
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.059203] prepare_command_request code: 10
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.077356] prepare_command_request code: 8
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.117638] prepare_command_request code: 27
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.141148] prepare_command_request code: 15
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.167818] prepare_command_request code: 15
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.391703] esp32_spi: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.100.012
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.391721] prepare_command_request code: 13
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.427995] prepare_command_request code: 14
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.526154] prepare_command_request code: 27
Sep  5 11:55:36 gridania-pvt1-soc kernel: [ 2804.549524] prepare_command_request code: 27
Sep  5 11:55:37 gridania-pvt1-soc kernel: [ 2805.206766] prepare_command_request code: 27
Sep  5 11:55:37 gridania-pvt1-soc kernel: [ 2805.236272] prepare_command_request code: 27
Sep  5 11:55:37 gridania-pvt1-soc kernel: [ 2806.006467] prepare_command_request code: 27
Sep  5 11:55:38 gridania-pvt1-soc kernel: [ 2806.030453] prepare_command_request code: 27
root@gridania-pvt1-soc:~# ifconfig wlan0
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.100.12  netmask 255.255.255.0  broadcast 192.168.100.255
        ether dc:da:0c:30:c1:0c  txqueuelen 1000  (Ethernet)
        RX packets 17828  bytes 11074452 (10.5 MiB)
        RX errors 0  dropped 593  overruns 0  frame 0
        TX packets 17897  bytes 12611828 (12.0 MiB)
        TX errors 0  dropped 29 overruns 0  carrier 0  collisions 0

```
For the other functions it looks work very well on both NetworkManager and WPA Supplicant
I refer to use NetworkManager because it support a lot of management tasks such as easy to connect, store AP configs easier...

So which one is the best choose for manage our wifi interface and ESPNG full compatible with both services?

mantriyogesh, 2024-09-05:
@kapilkedawat ++

kapilkedawat, 2024-09-05:
could you please share wpa_supplicant logs as well?

nguyenlkdn, 2024-09-06:
@kapilkedawat 

### Wifi Ap broadcast on channel 1 Using WPA Supplicant

Kernel Log
```
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908244] esp32_spi: process_esp_bootup_event: Received ESP bootup event
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908271] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908282] esp32_spi: esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908289] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908295] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908302] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908308] esp32_spi: process_fw_data: ESP chipset's last reset cause:
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908314] esp32_spi: print_reset_reason: POWERON_RESET
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908326] esp32_spi: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908800] esp32_spi: esp_reg_notifier: Driver init is ongoing
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.908887] prepare_command_request code: 1
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   35.985482] prepare_command_request code: 3
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   36.010517] prepare_command_request code: 15
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   36.029197] esp32_spi: print_capabilities: Capabilities: 0xa0. Features supported are:
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   36.029211] esp32_spi: print_capabilities: 	 * WLAN on SPI
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   36.063171] prepare_command_request code: 15
Sep  6 12:02:05 gridania-pvt1-soc kernel: [   36.122216] prepare_command_request code: 15
Sep  6 12:02:14 gridania-pvt1-soc kernel: [   45.232264] prepare_command_request code: 15
Sep  6 12:02:14 gridania-pvt1-soc kernel: [   45.261501] prepare_command_request code: 15
Sep  6 12:02:15 gridania-pvt1-soc kernel: [   45.281979] prepare_command_request code: 15
Sep  6 12:02:15 gridania-pvt1-soc kernel: [   45.297511] prepare_command_request code: 15
Sep  6 12:02:15 gridania-pvt1-soc kernel: [   45.322644] prepare_command_request code: 15
Sep  6 12:02:15 gridania-pvt1-soc kernel: [   45.466850] prepare_command_request code: 4
Sep  6 12:02:22 gridania-pvt1-soc kernel: [   52.904245] prepare_command_request code: 4
Sep  6 12:02:30 gridania-pvt1-soc kernel: [   60.344369] prepare_command_request code: 4
Sep  6 12:02:32 gridania-pvt1-soc kernel: [   62.796312] prepare_command_request code: 11
Sep  6 12:02:32 gridania-pvt1-soc kernel: [   62.796349] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:02:32 gridania-pvt1-soc kernel: [   63.218866] esp32_spi: decode_common_resp: [0xb] Command failed
Sep  6 12:02:32 gridania-pvt1-soc kernel: [   63.218891] esp32_spi: cmd_auth_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
Sep  6 12:02:33 gridania-pvt1-soc kernel: [   63.319557] prepare_command_request code: 4
Sep  6 12:02:35 gridania-pvt1-soc kernel: [   65.749956] prepare_command_request code: 11
Sep  6 12:02:35 gridania-pvt1-soc kernel: [   65.749990] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.288614] prepare_command_request code: 12
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.288646] esp32_spi: cmd_assoc_request: Association request: 30:4f:75:8c:e0:08 1 35
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.406214] esp32_spi: process_assoc_event: Connection status: 0
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.406349] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for VN, current=
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.429500] prepare_command_request code: 15
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.519003] prepare_command_request code: 27
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.553438] prepare_command_request code: 27
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.704052] prepare_command_request code: 8
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.731595] prepare_command_request code: 10
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.753339] prepare_command_request code: 8
Sep  6 12:02:36 gridania-pvt1-soc kernel: [   66.805646] prepare_command_request code: 15
Sep  6 12:02:38 gridania-pvt1-soc kernel: [   68.312630] esp32_spi: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.100.012
Sep  6 12:02:38 gridania-pvt1-soc kernel: [   68.312653] prepare_command_request code: 13
Sep  6 12:02:38 gridania-pvt1-soc kernel: [   68.340021] prepare_command_request code: 14
Sep  6 12:02:38 gridania-pvt1-soc kernel: [   68.356928] prepare_command_request code: 14
Sep  6 12:02:38 gridania-pvt1-soc kernel: [   68.607079] prepare_command_request code: 27
Sep  6 12:02:38 gridania-pvt1-soc kernel: [   68.634927] prepare_command_request code: 27
Sep  6 12:02:40 gridania-pvt1-soc kernel: [   70.672183] prepare_command_request code: 27
Sep  6 12:02:40 gridania-pvt1-soc kernel: [   70.725911] prepare_command_request code: 27
Sep  6 12:02:42 gridania-pvt1-soc kernel: [   72.763522] prepare_command_request code: 27
Sep  6 12:02:42 gridania-pvt1-soc kernel: [   72.802984] prepare_command_request code: 27
[... cut ...]
wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=VN
wlan0: WPA: Key negotiation completed with 30:4f:75:8c:e0:08 [PTK=CCMP GTK=CCMP]
wlan0: CTRL-EVENT-CONNECTED - Connection to 30:4f:75:8c:e0:08 completed [id=5 id_str=]
```

Ah i see that if AP broadcast on channel 1, WPA Supplicant also got some unexpected error on the first time authenticate , after retry 2nd time it successes. 

I tried several times no any problem. It always success on the first time authentication for AP Channel 10, and always need 2nd time to success authen on channel 1.

kapilkedawat, 2024-09-06:
Hi @nguyenlkdn, from the logs, it looks like AP has sent deauth with reason code 2(WLAN_REASON_PREV_AUTH_NOT_VALID), this could be due to the fact that station was previously connected to AP and then your reboot it. What happens at third attempt?

nguyenlkdn, 2024-09-06:
> Hi @nguyenlkdn, from the logs, it looks like AP has sent deauth with reason code 2(WLAN_REASON_PREV_AUTH_NOT_VALID), this could be due to the fact that station was previously connected to AP and then your reboot it. What happens at third attempt?

I always reboot the board to perform the connection, so any previous connections cannot be established.

Which procedure do you want me to attempt for 3rd?

I performed third time by restart wpa service and It always authencator failed by 201 code. 
I had seen that ESP disconnect immediately at time i stopped wpa supplicant.
```
Sep  6 12:52:25 gridania-pvt1-soc kernel: [ 2380.872955] prepare_command_request code: 27
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.727060] prepare_command_request code: 6
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.772890] prepare_command_request code: 13
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.778907] prepare_command_request code: 14
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.785519] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:3]
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.791237] esp32_spi: esp_inetaddr_event: Interface Down: 0
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.796771] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=VN
```

And after restart wpa suppliciant it alway authen failed. 

```
Sep  6 12:52:25 gridania-pvt1-soc kernel: [ 2380.872955] prepare_command_request code: 27
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.727060] prepare_command_request code: 6
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.772890] prepare_command_request code: 13
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.778907] prepare_command_request code: 14
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.785519] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:3]
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.791237] esp32_spi: esp_inetaddr_event: Interface Down: 0
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.796771] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=VN
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.796791] prepare_command_request code: 18
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.832457] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.498444] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.525344] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.548750] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.566129] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.591436] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.725350] prepare_command_request code: 4
Sep  6 12:52:58 gridania-pvt1-soc kernel: [ 2413.440158] prepare_command_request code: 4
Sep  6 12:53:04 gridania-pvt1-soc kernel: [ 2420.154969] prepare_command_request code: 4
Sep  6 12:53:11 gridania-pvt1-soc kernel: [ 2426.867551] prepare_command_request code: 4
Sep  6 12:53:13 gridania-pvt1-soc kernel: [ 2428.582602] prepare_command_request code: 11
Sep  6 12:53:13 gridania-pvt1-soc kernel: [ 2428.582634] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:53:15 gridania-pvt1-soc kernel: [ 2430.694993] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  6 12:53:15 gridania-pvt1-soc kernel: [ 2430.795878] prepare_command_request code: 4
Sep  6 12:53:20 gridania-pvt1-soc kernel: [ 2435.699312] prepare_command_request code: 6
Sep  6 12:53:20 gridania-pvt1-soc kernel: [ 2436.217940] prepare_command_request code: 4
Sep  6 12:53:22 gridania-pvt1-soc kernel: [ 2437.935325] prepare_command_request code: 11
Sep  6 12:53:22 gridania-pvt1-soc kernel: [ 2437.935356] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:53:24 gridania-pvt1-soc kernel: [ 2440.056292] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  6 12:53:24 gridania-pvt1-soc kernel: [ 2440.056456] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=00
Sep  6 12:53:24 gridania-pvt1-soc kernel: [ 2440.056652] prepare_command_request code: 15
Sep  6 12:53:28 gridania-pvt1-soc kernel: [ 2443.368174] prepare_command_request code: 6
Sep  6 12:53:29 gridania-pvt1-soc kernel: [ 2444.386667] prepare_command_request code: 4
Sep  6 12:53:30 gridania-pvt1-soc kernel: [ 2446.103008] prepare_command_request code: 11
Sep  6 12:53:30 gridania-pvt1-soc kernel: [ 2446.103041] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:53:32 gridania-pvt1-soc kernel: [ 2448.216566] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  6 12:53:32 gridania-pvt1-soc kernel: [ 2448.216738] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=00
Sep  6 12:53:32 gridania-pvt1-soc kernel: [ 2448.216925] prepare_command_request code: 15
Sep  6 12:53:36 gridania-pvt1-soc kernel: [ 2451.529031] prepare_command_request code: 6
Sep  6 12:53:41 gridania-pvt1-soc kernel: [ 2456.550007] prepare_command_request code: 4
Sep  6 12:53:48 gridania-pvt1-soc kernel: [ 2463.276133] prepare_command_request code: 4
[... cut ...]
wlan0: SME: Authentication request to the driver failed
wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=5 ssid="VIETTEL_AP_8CE000" auth_failures=3 duration=46 reason=CONN_FAILED
wlan0: CTRL-EVENT-SSID-REENABLED id=5 ssid="VIETTEL_AP_8CE000"
wlan0: SME: Trying to authenticate with 30:4f:75:8c:e0:08 (SSID='VIETTEL_AP_8CE000' freq=2412 MHz)
wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=5 ssid="VIETTEL_AP_8CE000" auth_failures=4 duration=77 reason=CONN_FAILED
```

nguyenlkdn, 2024-09-06:
++Updated for 3rd attempt results

It able to connect AP after 9 times retry authen

Kernel Log
```
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.785519] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:3]
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.791237] esp32_spi: esp_inetaddr_event: Interface Down: 0
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.796771] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=VN
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.796791] prepare_command_request code: 18
Sep  6 12:52:26 gridania-pvt1-soc kernel: [ 2381.832457] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.498444] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.525344] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.548750] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.566129] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.591436] prepare_command_request code: 15
Sep  6 12:52:51 gridania-pvt1-soc kernel: [ 2406.725350] prepare_command_request code: 4
Sep  6 12:52:58 gridania-pvt1-soc kernel: [ 2413.440158] prepare_command_request code: 4
Sep  6 12:53:04 gridania-pvt1-soc kernel: [ 2420.154969] prepare_command_request code: 4
Sep  6 12:53:11 gridania-pvt1-soc kernel: [ 2426.867551] prepare_command_request code: 4
Sep  6 12:53:13 gridania-pvt1-soc kernel: [ 2428.582602] prepare_command_request code: 11
Sep  6 12:53:13 gridania-pvt1-soc kernel: [ 2428.582634] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:53:15 gridania-pvt1-soc kernel: [ 2430.694993] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  6 12:53:15 gridania-pvt1-soc kernel: [ 2430.795878] prepare_command_request code: 4
Sep  6 12:53:20 gridania-pvt1-soc kernel: [ 2435.699312] prepare_command_request code: 6
Sep  6 12:53:20 gridania-pvt1-soc kernel: [ 2436.217940] prepare_command_request code: 4
Sep  6 12:53:22 gridania-pvt1-soc kernel: [ 2437.935325] prepare_command_request code: 11
Sep  6 12:53:22 gridania-pvt1-soc kernel: [ 2437.935356] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:53:24 gridania-pvt1-soc kernel: [ 2440.056292] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  6 12:53:24 gridania-pvt1-soc kernel: [ 2440.056456] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=00
Sep  6 12:53:24 gridania-pvt1-soc kernel: [ 2440.056652] prepare_command_request code: 15
Sep  6 12:53:28 gridania-pvt1-soc kernel: [ 2443.368174] prepare_command_request code: 6
Sep  6 12:53:29 gridania-pvt1-soc kernel: [ 2444.386667] prepare_command_request code: 4
Sep  6 12:53:30 gridania-pvt1-soc kernel: [ 2446.103008] prepare_command_request code: 11
Sep  6 12:53:30 gridania-pvt1-soc kernel: [ 2446.103041] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:53:32 gridania-pvt1-soc kernel: [ 2448.216566] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  6 12:53:32 gridania-pvt1-soc kernel: [ 2448.216738] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=00
Sep  6 12:53:32 gridania-pvt1-soc kernel: [ 2448.216925] prepare_command_request code: 15
Sep  6 12:53:36 gridania-pvt1-soc kernel: [ 2451.529031] prepare_command_request code: 6
Sep  6 12:53:41 gridania-pvt1-soc kernel: [ 2456.550007] prepare_command_request code: 4
Sep  6 12:53:48 gridania-pvt1-soc kernel: [ 2463.276133] prepare_command_request code: 4
Sep  6 12:53:49 gridania-pvt1-soc kernel: [ 2465.000123] prepare_command_request code: 11
Sep  6 12:53:49 gridania-pvt1-soc kernel: [ 2465.000157] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:53:51 gridania-pvt1-soc kernel: [ 2467.120939] esp32_spi: process_disconnect_event: Disconnect event for ssid VIETTEL_AP_8CE000 [reason:201]
Sep  6 12:53:51 gridania-pvt1-soc kernel: [ 2467.121101] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=00
Sep  6 12:53:51 gridania-pvt1-soc kernel: [ 2467.121348] prepare_command_request code: 15
Sep  6 12:53:55 gridania-pvt1-soc kernel: [ 2470.432552] prepare_command_request code: 6
Sep  6 12:54:05 gridania-pvt1-soc kernel: [ 2480.464264] prepare_command_request code: 4
Sep  6 12:54:11 gridania-pvt1-soc kernel: [ 2487.184168] prepare_command_request code: 4
Sep  6 12:54:18 gridania-pvt1-soc kernel: [ 2493.905350] prepare_command_request code: 4
Sep  6 12:54:20 gridania-pvt1-soc kernel: [ 2495.623267] prepare_command_request code: 11
Sep  6 12:54:20 gridania-pvt1-soc kernel: [ 2495.623302] esp32_spi: cmd_auth_request: Authentication request: 30:4f:75:8c:e0:08 1 0 0 0
Sep  6 12:54:20 gridania-pvt1-soc kernel: [ 2496.042721] esp32_spi: decode_common_resp: [0xb] Command failed
Sep  6 12:54:20 gridania-pvt1-soc kernel: [ 2496.042743] esp32_spi: cmd_auth_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
Sep  6 12:54:30 gridania-pvt1-soc kernel: [ 2506.052918] prepare_command_request code: 4
Sep  6 12:54:37 gridania-pvt1-soc kernel: [ 2512.776147] prepare_command_request code: 4
Sep  6 12:54:44 gridania-pvt1-soc kernel: [ 2519.502004] prepare_command_request code: 4
[... cut ...]
wlan0: SME: Trying to authenticate with 30:4f:75:8c:e0:08 (SSID='VIETTEL_AP_8CE000' freq=2412 MHz)
wlan0: Trying to associate with 30:4f:75:8c:e0:08 (SSID='VIETTEL_AP_8CE000' freq=2412 MHz)
wlan0: Associated with 30:4f:75:8c:e0:08
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=VN
wlan0: WPA: Key negotiation completed with 30:4f:75:8c:e0:08 [PTK=CCMP GTK=CCMP]
wlan0: CTRL-EVENT-CONNECTED - Connection to 30:4f:75:8c:e0:08 completed [id=5 id_str=]

```

xiaoyaofa, 2024-12-10:
Hello, I also encountered the same problem, when I connected to the hotspot of channel 1, I could not connect successfully.
[image]
[image]
[image]

kapilkedawat, 2024-12-10:
@xiaoyaofa there are multiple timeout failures, could you please verify the transport first?

nguyenlkdn, 2024-12-11:
@xiaoyaofa this issue happened with wpa supplicant old version (2.9) It got some issues with MESH network configuration. it always try to authenticate with parent AP although we are selecting other APs

After updated to wpa 2.11 it already fixed.