espressif/esp-hosted #479: issues with esp32_spi.ko getting exiting after being inserted

sreejitharikath, 2024-09-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

virtual serial interface espso is not created after runing esphosted fg in both rpi4b and esp32s3 even after flashing their respective repositories from github

### Actual behavior (suspected bug)

 when runing the commad  bash -x ./rpi_init.sh wifi=spi
[ 204.596250] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 204.627263] esp32_spi: loading out-of-tree module taints kernel.
please help me in fixing this why after the module is inserted it is exiting why iam getting warning of memory leakage , why iam getting message that kernel is getting tainted

### Error logs or terminal output

```shell
currently i have commented the dts file seprately and also commented  device_tree_dependency_spi() and device overlay in boot/firmware/config.text and additional spi device overlay if spi  all in rpi.insit.sh just make spi enable for getting this log.iam not able to make virtual serial interface espso in fg set up

issues to to kernel module esp32_spio.ko its getting inserted by script but its exiting
[[ 204.596250] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 204.627263] esp32_spi: loading out-of-tree module taints kernel.
[ 485.863572] esp32_spi: esp_serial_cleanup: 
[ 486.333378] esp32_spi: esp_spi_thread: esp spi thread cleared

dmesg | tail -n 50
[ 12.268969] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video20
[ 12.269712] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[ 12.269753] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[ 12.270757] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video21
[ 12.282325] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video22
[ 12.282900] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[ 12.282939] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[ 12.283479] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video23
[ 12.283554] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[ 12.283574] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[ 12.283591] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[ 12.283657] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[ 12.283966] rpivid feb10000.codec: Device registered as /dev/video19
[ 12.295543] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[ 12.306115] alsactl[408]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[ 12.311881] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[ 12.311931] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[ 12.320233] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[ 12.320276] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[ 12.336739] bcm2835-codec bcm2835-codec: Device registered as /dev/video31
[ 12.336787] bcm2835-codec bcm2835-codec: Loaded V4L2 encode_image
[ 13.762696] Adding 204796k swap on /var/swap. Priority:-2 extents:6 across:3465216k SS
[ 15.709808] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[ 15.711050] bcmgenet fd580000.ethernet eth0: Link is Down
[ 19.835969] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 82.233704] bcmgenet fd580000.ethernet eth0: Link is Down
[ 204.596250] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 204.627263] esp32_spi: loading out-of-tree module taints kernel.
[ 204.627761] esp32_spi: esp_reset: Resetpin of Host is 518
[ 204.627979] esp32_spi: esp_reset: Triggering ESP reset.
[ 204.628110] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[534] DataReady[539]
[ 204.628116] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[518] Handshake[534] Dataready[539]
[ 204.628120] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
[ 204.628128] esp32_spi: esp_spi_thread: esp spi thread created
[ 485.863572] esp32_spi: esp_serial_cleanup:
[ 486.333378] esp32_spi: esp_spi_thread: esp spi thread cleared
[ 499.956462] esp32_spi: esp_reset: Resetpin of Host is 518
[... 323 lines cut ...]

ESP-Hosted-FG-release_0.0.5.tgz for rpi4b and ESP-Hosted-FG-release_0.0.5.tgz for esp32 s3 wroom 1

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

 PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)" NAME="Raspbian GNU/Linux" VERSION_ID="12" VERSION="12 (bookworm)" VERSION_CODENAME=bookworm ID=raspbian ID_LIKE=debian

### Shell

Bash

### Additional context

SPI Initialization: The esp32_spi module is initialized successfully with SPI bus 0, Chip Select 0, and SPI mode 2. The GPIOs for handshake and data ready are correctly set.
Resets and Reinitializations: The logs show multiple resets of the ESP32 and reinitializations of the SPI module. This might be triggered by specific actions like resetting the ESP32 or reloading the module.
Potential Overlay Warning: The warning about a potential memory leak when removing the overlay
why esps0 is not created or enabled why module is getting exit

mantriyogesh, 2024-09-10:
Can you please use same got commit, preferably hosted master on ESP and host?

mantriyogesh, 2024-09-10:
1. Also attach the photo of setup,

2. Logs with latest master:
- Host Dmesg from system bootup
- Esp - minicom or IDF.py monitor from start
- Git commit used both sides

sreejitharikath, 2024-09-10:
please find Also attach the photo of setup, Esp - minicom as attachments
[image]
[image]
[image]
[image]

)

sreejitharikath, 2024-09-10:
host dmesg from system boot up
0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.47+rpt-rpi-v7l (serge@raspberrypi.com) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian) 2.40) #1 SMP Raspbian 1:6.6.47-1+rpt1 (2024-09-02)
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef64f20..0x000000003ef6531f (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef65360..0x000000003ef65395 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 18 pages/cpu s41940 r8192 d23596 u73728
[    0.000000] pcpu-alloc: s41940 r8192 d23596 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:E2:BC:AC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=e59f67cd-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=IN
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000007890000-0x000000000b890000] (64MB)
[    0.000000] Memory: 3382000K/4050944K available (12288K kernel code, 1527K rwdata, 3488K rodata, 2048K init, 628K bss, 144656K reserved, 524288K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 38772 entries in 114 pages
[    0.000000] ftrace: allocated 114 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000016] Switching to timer-based delay loop, resolution 18ns
[    0.000357] Console: colour dummy device 80x30
[    0.000374] printk: console [tty1] enabled
[    0.001018] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001054] CPU: Testing write buffer coherency: ok
[    0.001105] CPU0: Spectre BHB: enabling loop workaround for all CPUs
[    0.001127] pid_max: default: 32768 minimum: 301
[    0.001222] LSM: initializing lsm=capability,integrity
[    0.001409] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001445] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002475] cgroup: Disabling memory control group subsystem
[    0.002956] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003913] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004039] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004173] Setting up static identity map for 0x200000 - 0x20003c
[    0.004332] rcu: Hierarchical SRCU implementation.
[    0.004353] rcu: 	Max phase no-delay instances is 1000.
[    0.005044] smp: Bringing up secondary CPUs ...
[    0.005895] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006777] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007618] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007708] smp: Brought up 1 node, 4 CPUs
[... 345 lines cut ...]
[  204.628116] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[518] Handshake[534] Dataready[539]
[... cut ...]
[  649.013908] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
[  649.014189] esp32_spi: esp_spi_thread: esp spi thread created
[  843.513627] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[10904.531833] bcmgenet fd580000.ethernet eth0: Link is Down
[18918.828218] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[19433.099945] v3d fec00000.v3d: MMU error from client CLE (4) at 0x4781000, pte invalid
[26393.349400] bcmgenet fd580000.ethernet eth0: Link is Down
itas@raspberrypi:~/esp-hos

mantriyogesh, 2024-09-11:
wires are too long. You can limit the clock using https://github.com/espressif/esp-hosted/blob/20d939491fb20841ae0b56221f249be02ad0ac69/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L17

Solution is already available for Raspberry Pi, which is your host. Porting is **not** needed in your case.

However, it is important to note the important hardware considerations from [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md).
This includes smaller wire lengths, lower clocks etc, with respect to your transport, spi.

Additionally, I couldn't find the esp log, yet, which is crucial for debugging.
First transaction from esp (bootup event) is not observed in logs. So it is suggested to cross-verify if the GPIOs used in ESP and host, are correctly connected, to the ones used in software.

sreejitharikath, 2024-09-11:
hi we have set the clock as 10mhz only
please find the kernel logs after i have uncommented device tree source file, device_tree_dependency_spi() ,and additional spi device overlay for spi  all in rpi.insit.sh device overlay in boot/firmware/config.text
itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.47+rpt-rpi-v7l (serge@raspberrypi.com) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian) 2.40) #1 SMP Raspbian 1:6.6.47-1+rpt1 (2024-09-02)
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef64f20..0x000000003ef6531f (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef65360..0x000000003ef65395 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 18 pages/cpu s41940 r8192 d23596 u73728
[    0.000000] pcpu-alloc: s41940 r8192 d23596 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:E2:BC:AC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=e59f67cd-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=IN
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000007890000-0x000000000b890000] (64MB)
[    0.000000] Memory: 3382000K/4050944K available (12288K kernel code, 1527K rwdata, 3488K rodata, 2048K init, 628K bss, 144656K reserved, 524288K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 38772 entries in 114 pages
[    0.000000] ftrace: allocated 114 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000017] Switching to timer-based delay loop, resolution 18ns
[    0.000358] Console: colour dummy device 80x30
[    0.000377] printk: console [tty1] enabled
[    0.001019] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001058] CPU: Testing write buffer coherency: ok
[    0.001109] CPU0: Spectre BHB: enabling loop workaround for all CPUs
[    0.001132] pid_max: default: 32768 minimum: 301
[    0.001227] LSM: initializing lsm=capability,integrity
[    0.001417] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001453] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002498] cgroup: Disabling memory control group subsystem
[    0.002988] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003952] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004066] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004210] Setting up static identity map for 0x200000 - 0x20003c
[    0.004368] rcu: Hierarchical SRCU implementation.
[    0.004390] rcu: 	Max phase no-delay instances is 1000.
[    0.005080] smp: Bringing up secondary CPUs ...
[    0.005932] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... cut ...]
[ 1009.427187] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 1009.457237] esp32_spi: esp_reset: Resetpin of Host is 518
[ 1009.457464] esp32_spi: esp_reset: Triggering ESP reset.
[ 1009.457592] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[534] DataReady[539]
[ 1009.457594] esp32_spi: esp_spi_thread: esp spi thread created
[ 1009.457598] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[518] Handshake[534] Dataready[539]
[ 1009.457603] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2

mantriyogesh, 2024-09-11:
You have reattached the dmesg log.

What is flashed on esp chipset? Where are esp chipset logs?

Where is /boot/firmware/config.txt?

Also confirm the raspberry and esp connections. which pins connected to what.

sreejitharikath, 2024-09-11:
hi with respect to esp32s3 we have flashed ESP-Hosted-FG-release_0.0.5.tgz for esp32 s3 wroom 1
we have flashed spi +uart part
And please find the config.txt below 
GNU nano 7.2                                                   config.txt                                                       M     
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
dtoverlay=spi-bcm2835

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1
arm_64bit=0

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtparam=spi=on
dtoverlay=disable-bt
dtoverlay=disable-wifi
enable_uart=1
dtoverlay=spidev_disabler

sreejitharikath, 2024-09-11:
With respect to esp32s3 we have flashed ESP-Hosted-FG-release_0.0.5.tgz for esp32 s3 wroom 1
we have flashed spi+uart 
please find the contents of config.txt below
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
dtoverlay=spi-bcm2835

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1
arm_64bit=0

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtparam=spi=on
dtoverlay=disable-bt
dtoverlay=disable-wifi
enable_uart=1
dtoverlay=spidev_disabler

mantriyogesh, 2024-09-11:
Flash latest master on esp and use same commit on host.

dtoverlay=spidev_disabler
this is alreasy taken care of in rpi_init.sh and not needed here.

mantriyogesh, 2024-09-11:
Once you flash the
- esp from latest master
- build and load host driver from master
get us all textual logs from esp, host.

Also verify the connection. use smaller 10cm, equal and good quality wires.

sreejitharikath, 2024-09-11:
please find the esp logs for the repository which we flashed which is  ESP-Hosted-FG-release_0.0.5.tgz
we will send the latest logs immediately
$ dmesg
[    0.000000] microcode: microcode updated early to revision 0x2f, date = 2019-11-12
[    0.000000] Linux version 5.15.0-119-generic (buildd@lcy02-amd64-068) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #129~20.04.1-Ubuntu SMP Wed Aug 7 13:07:13 UTC 2024 (Ubuntu 5.15.0-119.129~20.04.1-generic 5.15.160)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-119-generic root=UUID=59698b96-c066-467a-b0f5-7e2d0725eccd ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x000000001000afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000001000b000-0x00000000beccffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000becd0000-0x00000000ccdfefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ccdff000-0x00000000ccf7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ccf7f000-0x00000000ccffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000ccfff000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ff000fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000022effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: LENOVO 20DSA0FV00/Intel powered classmate PC, BIOS JDET49WW (1.11 ) 03/02/2015
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 2194.765 MHz processor
[    0.001037] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001042] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001054] last_pfn = 0x22f000 max_arch_pfn = 0x400000000
[    0.001206] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001993] last_pfn = 0xbecd0 max_arch_pfn = 0x400000000
[    0.002119] found SMP MP-table at [mem 0x000f0100-0x000f010f]
[    0.002139] Using GB pages for direct mapping
[    0.002597] RAMDISK: [mem 0x2d927000-0x32c8afff]
[    0.002604] ACPI: Early table checksum verification disabled
[    0.002607] ACPI: RSDP 0x00000000000F0120 000024 (v02 LENOVO)
[    0.002613] ACPI: XSDT 0x00000000CCFFE1C0 0000EC (v01 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002620] ACPI: FACP 0x00000000CCFF8000 00010C (v05 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002628] ACPI: DSDT 0x00000000CCFE3000 010F88 (v01 LENOVO TP-JD    00001110 INTL 20120711)
[    0.002633] ACPI: FACS 0x00000000CCF68000 000040
[    0.002637] ACPI: SLIC 0x00000000CCFFD000 000176 (v01 LENOVO TP-JD    00001110 PTEC 00000001)
[    0.002641] ACPI: ASF! 0x00000000CCFFC000 0000A5 (v32 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002647] ACPI: HPET 0x00000000CCFFB000 000038 (v01 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002651] ACPI: ECDT 0x00000000CCFFA000 000052 (v01 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002656] ACPI: APIC 0x00000000CCFF7000 000098 (v01 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002660] ACPI: MCFG 0x00000000CCFF6000 00003C (v01 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002665] ACPI: SSDT 0x00000000CCFF5000 000033 (v01 LENOVO TP-SSDT1 00000100 INTL 20120711)
[    0.002669] ACPI: SSDT 0x00000000CCFF4000 000486 (v01 LENOVO TP-SSDT2 00000200 INTL 20120711)
[    0.002674] ACPI: SSDT 0x00000000CCFE2000 0009CB (v01 LENOVO SataAhci 00001000 INTL 20120711)
[    0.002678] ACPI: SSDT 0x00000000CCFE1000 00062A (v01 LENOVO Cpu0Ist  00003000 INTL 20120711)
[    0.002683] ACPI: SSDT 0x00000000CCFE0000 000B74 (v02 LENOVO CpuSsdt  00003000 INTL 20120711)
[    0.002687] ACPI: SSDT 0x00000000CCFDF000 000369 (v02 LENOVO CtdpB    00001000 INTL 20120711)
[    0.002692] ACPI: SSDT 0x00000000CCFDD000 001477 (v01 LENOVO SaSsdt   00003000 INTL 20120711)
[    0.002696] ACPI: SSDT 0x00000000CCFDC000 000394 (v02 LENOVO CppcTabl 00001000 INTL 20120711)
[    0.002701] ACPI: PCCT 0x00000000CCFDB000 00006E (v05 LENOVO TP-JD    00001110 PTEC 00000002)
[    0.002705] ACPI: SSDT 0x00000000CCFDA000 000AC4 (v02 LENOVO Cpc_Tabl 00001000 INTL 20120711)
[    0.002709] ACPI: TCPA 0x00000000CCFD9000 000032 (v02 PTL    LENOVO   06040000 LNVO 00000001)
[    0.002714] ACPI: SSDT 0x00000000CCFD8000 0006A5 (v01 Intel_ TpmTable 00001000 INTL 20120711)
[... cut ...]
[37506.144268] ACPI: \: failed to evaluate _DSM (0x1001)
[37506.144271] ACPI: \: failed to evaluate _DSM (0x1001)
[37507.085857] wlp4s0: authenticate with ec:1a:59:fd:82:90
[37507.090733] wlp4s0: send auth to ec:1a:59:fd:82:90 (try 1/3)
[37507.093728] wlp4s0: authenticated
[37507.094053] wlp4s0: associate with ec:1a:59:fd:82:90 (try 1/3)
[37507.097839] wlp4s0: RX AssocResp from ec:1a:59:fd:82:90 (capab=0x411 status=0 aid=3)
[37507.101227] wlp4s0: associated
[37507.176629] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready

mantriyogesh, 2024-09-11:
As mentioned in https://github.com/espressif/esp-hosted/issues/479#issuecomment-2342762897, Please use latest master code from esp-hosted at both places, esp and host.

Complete documentation of how to flash the ESP and host is available under https://github.com/espressif/esp-hosted/blob/master/README.md

sreejitharikath, 2024-09-11:
please find the connection reference which we followed to do the spi connection
[image]

mantriyogesh, 2024-09-11:
Please note, slave flashing and getting log on the slave chipsets is different from that of hosts.

For ESP-Hosted to work:
- Set up communication bus like SDIO or SPI connection, with correct pins
  - ESP software and hardware should match GPIOs used
  - Host software and hardware should match GPIOs used
  - Host and slave connection done correctly to specified function
- Slave firmware flashing from latest master
  - Ensure the `idf.py menuconfig` done correctly to use correct bus before flashing
  - The GPIOs expected are correct to their functions
  - Get the log from start in textual format
- Host module build and inserted using latest master
  - Get dmesg log only when you have all above done.

sreejitharikath, 2024-09-12:
please find the git commit on the esp32 side
C:\esp-hosted>git log -1
commit 70de43ac9cc30ef55960d8025274df7a0fe1dfc8 (HEAD)
Merge: 5f85029 01a9eae
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Fri Aug 23 21:50:05 2024 +0800

    Merge branch 'feat/version_debugfs_hostedng' into 'master'

    esp_hosted_ng: Add version support in debugfs

    See merge request app-frameworks/esp_hosted!488

sreejitharikath, 2024-09-12:
please find the git commit in host rpi side
itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control $ git log -1
commit 70de43ac9cc30ef55960d8025274df7a0fe1dfc8 (HEAD, origin/master, origin/HEAD, master)
Merge: 5f85029 01a9eae
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Fri Aug 23 21:50:05 2024 +0800

    Merge branch 'feat/version_debugfs_hostedng' into 'master'
    
    esp_hosted_ng: Add version support in debugfs
    
    See merge request app-frameworks/esp_hosted!488

mantriyogesh, 2024-09-12:
Sure. Check other things mentioned in https://github.com/espressif/esp-hosted/issues/479#issuecomment-2342999046

mantriyogesh, 2024-09-12:
It is surprising that you are not getting the first event was not through.

Can you please attach the sdkconfig used?
Also, Can you please confirm you have connected through: 
[image]

sreejitharikath, 2024-09-12:
please find the kernal logs from host side
dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.47+rpt-rpi-v7l (serge@raspberrypi.com) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian) 2.40) #1 SMP Raspbian 1:6.6.47-1+rpt1 (2024-09-02)
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef64f20..0x000000003ef6531f (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef65360..0x000000003ef65395 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 18 pages/cpu s41940 r8192 d23596 u73728
[    0.000000] pcpu-alloc: s41940 r8192 d23596 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:E2:BC:AC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=e59f67cd-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=IN
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000007890000-0x000000000b890000] (64MB)
[    0.000000] Memory: 3382000K/4050944K available (12288K kernel code, 1527K rwdata, 3488K rodata, 2048K init, 628K bss, 144656K reserved, 524288K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 38772 entries in 114 pages
[    0.000000] ftrace: allocated 114 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000016] Switching to timer-based delay loop, resolution 18ns
[    0.000354] Console: colour dummy device 80x30
[    0.000370] printk: console [tty1] enabled
[    0.001004] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001041] CPU: Testing write buffer coherency: ok
[    0.001088] CPU0: Spectre BHB: enabling loop workaround for all CPUs
[    0.001109] pid_max: default: 32768 minimum: 301
[    0.001201] LSM: initializing lsm=capability,integrity
[    0.001375] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001411] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002426] cgroup: Disabling memory control group subsystem
[    0.002906] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003851] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.003965] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004105] Setting up static identity map for 0x200000 - 0x20003c
[    0.004256] rcu: Hierarchical SRCU implementation.
[    0.004277] rcu: 	Max phase no-delay instances is 1000.
[    0.004964] smp: Bringing up secondary CPUs ...
[    0.005809] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006686] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007515] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[... 349 lines cut ...]
[17359.131043] esp32_spi: esp_reset: Triggering ESP reset.
[... cut ...]
[18968.894492] esp32_spi: esp_reset: Resetpin of Host is 518
[18968.894720] esp32_spi: esp_reset: Triggering ESP reset.
[18968.894878] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[534] DataReady[539]
[18968.894882] esp32_spi: esp_spi_thread: esp spi thread created
[18968.894887] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[518] Handshake[534] Dataready[539]
[18968.894891] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control $

sreejitharikath, 2024-09-12:
hi sir 
Please find the contents of sdkconfig file
 Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.3.0 Project Configuration
#
CONFIG_SOC_MPU_MIN_REGION_SIZE=0x20000000
CONFIG_SOC_MPU_REGIONS_MAX_NUM=8
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_PHY_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_AHB_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_MPU_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_RNG_SUPPORTED=y
CONFIG_SOC_LIGHT_SLEEP_SUPPORTED=y
CONFIG_SOC_DEEP_SLEEP_SUPPORTED=y
CONFIG_SOC_LP_PERIPH_SHARE_INTERRUPT=y
CONFIG_SOC_PM_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_APPCPU_HAS_CLOCK_GATING_BUG=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_ARBITER_SUPPORTED=y
[... 2199 lines cut ...]
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

mantriyogesh, 2024-09-12:
from sdkconfig, I could see, 

```
CONFIG_ESP_SPI_HOST_INTERFACE=y

#
# SPI Configuration
#
CONFIG_ESP_SPI_GPIO_HANDSHAKE=2
CONFIG_ESP_SPI_GPIO_DATA_READY=4
CONFIG_ESP_SPI_TX_Q_SIZE=20
CONFIG_ESP_SPI_RX_Q_SIZE=20
CONFIG_ESP_SPI_CHECKSUM=y
```

From code:

https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L114-L129

I assume you have not changed the code anywhere.
So, 
1. Question again:

>  -- **https://github.com/espressif/esp-hosted/issues/479#issuecomment-2345943165 is connected to Raspberry Pi correctly?** --

2. Can you try shorter wires or smaller SPI frequency like 1MHz ?
On latest master, you can send change the SPI clock freq at https://github.com/espressif/esp-hosted/blob/20d939491fb20841ae0b56221f249be02ad0ac69/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L17

sreejitharikath, 2024-09-13:
Hi Sir

We have rechecked the connections properly
its fine from our part.
we have changed 1mhz
but still the issue is not yet resolved

mantriyogesh, 2024-09-13:
1. Short the wires, recommended length <=10cm for spi https://github.com/espressif/esp-hosted/issues/479#issuecomment-2341672737
2. Check the connectivity using multi meter
3. Do you have logic analyser? Analyser would be best measure to understand what is happening on gpios
4. Before testing kernel module directly, first enable spidev and check if the spi works with spidev first in user space.

sreejitharikath, 2024-09-14:
Hi sir
we have used wires of length =10cm and continuity also checked
based on the command dmesg | tail -n 20 log is
dmesg | tail -n 20
[  192.848837] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[  192.878462] esp32_spi: loading out-of-tree module taints kernel.
[  192.878965] esp32_spi: esp_reset: Resetpin of Host is 506
[  192.879175] esp32_spi: esp_reset: Triggering ESP reset.
[  192.879317] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[522] DataReady[527]
[  192.879320] esp32_spi: esp_spi_thread: esp spi thread created
[  192.879323] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[506] Handshake[522] Dataready[527]
[  192.879328] esp32_spi: spi_dev_init: Config - SPI: clock[1MHz] bus[0] cs[0] mode[2]
[  380.873806] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 1091.805994] v3d fec00000.v3d: MMU error from client CLE (4) at 0x2341000, pte invalid
[ 1182.991997] esp32_spi: esp_serial_cleanup: 
[ 1183.458018] esp32_spi: esp_spi_thread: esp spi thread cleared
[ 1197.272786] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 1197.302412] esp32_spi: esp_reset: Resetpin of Host is 506
[ 1197.302626] esp32_spi: esp_reset: Triggering ESP reset.
[ 1197.302830] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[522] DataReady[527]
[ 1197.302836] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[506] Handshake[522] Dataready[527]
[ 1197.302841] esp32_spi: spi_dev_init: Config - SPI: clock[1MHz] bus[0] cs[0] mode[2]
[ 1197.303104] esp32_spi: esp_spi_thread: esp spi thread created
[ 3865.501090] esp32_spi: esp_serial_ioctl: IOCTL unsupported 21505

mantriyogesh, 2024-09-15:
Either test in user space as said earlier or logic analyser as last sort.

Problem is your spi pins, miso, clk, cs right now. Other gpios also need testing later. For now very sure of these tree keys not correct.

sreejitharikath, 2024-09-20:
Hi sir 
Is there any specific command other than ./rpi_init.sh wifi=spi to build the kernel module esp32_spio.ko

mantriyogesh, 2024-09-20:
> Hi sir Is there any specific command other than ./rpi_init.sh wifi=spi to build the kernel module esp32_spio.ko

You meant 'esp32_spi.ko' . So yes, this should get you wifi on spi built on kernel module side.

mantriyogesh, 2024-09-20:
it is worth to cross check your gpios are working fine first for spi.

https://github.com/espressif/esp-hosted/issues/479#issuecomment-2351293886

Once SPI gpios are correct, you can also check handshake and data ready gpios are correct or not.

sreejitharikath, 2024-11-11:
Hi sir
Please be informed that last issue related to kernel building was resolved.We were able to build the kernel log after flashing 64bit rasbian 64bit os .

- We have changed the wires  for doing the hardware spi connections between esp32s3 and rpi,used wires of less than10cm  now
- .And we have checked the spi dev test using python script for checking the spi communication between rpi and esp32s3 we are facing issues related to clock synchronisation .the dayta which we are getting in rpi is invalid.

- Initialy we have used  50khz as clk frequency but failed in establishing spi communication and set up wifi.

- After that since 10mhz was the expected frequency changed the clock frquency in the python script to 10mhz from 50khz was ble to invoke handshake interrupts ,start spi transaction but and still failing in the communication and wifi set up.
- - Changed the frequency back to 10 mhz and also used control commands in the test.c in the directory c_support files to get wifi mode didnt got any response got issues like segmentation error have faced same response previous too while using control commands.
- Now we have done spi dev test by maintaing 40mhz clok req in script and also changed macro for clk freq in rpi got dataready interupt also invoked but not recieving any data in rpi and no changes is seen in the logs of esp32s3.
- After got proper logs in both esp32s3 and rpi kernel logs but failing to establish spi communication and set up wifi.
- please find the logs attatched ,photo of the hardware set up kindly guide us what has to done to resolve the following issues.
- What should be the exact clock frequency to set communication both rpi and esp32s3 for proper spi communication.
Please let me know if any changes/modification  is required to be made to this content will add all logs and relevant materials for the supposrts to understand is attatched here

sreejitharikath, 2024-11-11:
log for 
rpi-50khz esp32-40mhz
[  689.199146] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199148] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199151] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199153] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199155] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199158] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199160] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199163] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199165] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199168] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199170] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199173] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199175] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199178] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199180] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199182] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199185] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199188] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199190] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199192] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199195] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199197] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199200] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199202] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199204] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199207] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199210] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199212] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199214] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199217] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199219] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199222] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199224] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199227] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199229] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199231] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199234] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199237] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199239] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199241] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199243] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199246] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199248] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199251] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199253] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199256] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199258] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199260] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199263] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199266] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199268] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199270] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199273] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199275] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199278] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199280] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199282] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199285] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199288] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199290] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199292] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199295] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199297] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199300] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  689.199302] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  689.199305] esp32_spi: esp_spi_thread: entered esp spi loop
[  689.199307] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  689.199309] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[... 904 lines cut ...]
[  689.201737] lr : esp_spi_transaction+0x3e8/0x480 [esp32_spi]
[  689.201748] sp : ffffffc08371bd30
[  689.201751] x29: ffffffc08371bd30 x28: 0000000000000000 x27: 0000000000000000
[  689.201760] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8044839a00
[  689.201768] x23: ffffffeac25d4508 x22: ffffffeac25d0208 x21: ffffffeac25d4cc8
[  689.201776] x20: ffffffeac25d4508 x19: ffffffeac25d0740 x18: ffffffffffffffff
[  689.201784] x17: 73696d736e617274 x16: 20726f6620646576 x15: 6569727465722073
[  689.201792] x14: 692074656b636170 x13: 697073207265766f x12: 206e6f697373696d
[  689.201800] x11: 736e61727420726f x10: ffffffeb398b2788 x9 : ffffffeac25cb9b8
[... cut ...]
[  689.201859]  esp_spi_thread+0xec/0x130 [esp32_spi]
[  689.201870]  kthread+0x11c/0x128
[  689.201886]  ret_from_fork+0x10/0x20
[  689.201897] Code: f9000bf3 d0000033 911d0273 f9401a60 (f9400400) 
[  689.201904] ---[ end trace 0000000000000000 ]---
[  689.203226] SPI Interrupt Handler called, IRQ: 55
[  689.203232] Semaphore acquired: 3
[  689.203974] SPI Interrupt Handler called, IRQ: 55
[  689.203980] Semaphore acquired: 4
itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support $

sreejitharikath, 2024-11-11:
log for Contral command test using 10mhz
[  361.453848] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453851] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453853] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453856] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453858] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453861] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453863] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453866] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453868] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453871] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453873] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453875] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453878] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453881] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453883] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453885] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453887] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453890] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453892] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453895] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453897] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453900] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453902] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453905] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453907] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453910] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453912] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453914] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453917] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453919] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453922] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453924] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453926] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453929] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453931] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453934] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453936] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453939] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453941] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453943] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453946] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453948] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453951] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453953] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453955] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453958] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453960] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453963] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453965] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453968] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453970] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453972] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453975] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453978] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453980] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453982] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453984] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453987] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453989] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.453992] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.453994] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.453997] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.453999] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.454002] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.454004] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[  361.454007] esp32_spi: esp_spi_thread: entered esp spi loop
[  361.454009] esp32_spi: esp_spi_thread: Semaphore acquired: 1
[  361.454011] esp32_spi: esp_spi_thread: SPI adapter ready: 2
[  361.454014] esp32_spi: esp_spi_transaction: entered spi transaction after initilisation
[... 900 lines cut ...]
[  361.456394] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[  361.456398] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  361.456403] pc : esp_tx_resume+0x24/0xe0 [esp32_spi]
[  361.456428] lr : esp_spi_transaction+0x3e8/0x480 [esp32_spi]
[  361.456440] sp : ffffffc0813f3d30
[  361.456443] x29: ffffffc0813f3d30 x28: 0000000000000000 x27: 0000000000000000
[  361.456452] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff806f6c4b00
[  361.456461] x23: ffffffd328cff508 x22: ffffffd328cfb208 x21: ffffffd328cffcc8
[... cut ...]
[  361.456518] x2 : 0000000000000000 x1 : ffffff8041a18000 x0 : 0000000000000000
[  361.456527] Call trace:
[  361.456531]  esp_tx_resume+0x24/0xe0 [esp32_spi]
[  361.456543]  esp_spi_transaction+0x3e8/0x480 [esp32_spi]
[  361.456554]  esp_spi_thread+0xec/0x130 [esp32_spi]
[  361.456564]  kthread+0x11c/0x128
[  361.456579]  ret_from_fork+0x10/0x20
[  361.456590] Code: f9000bf3 d0000033 911d0273 f9401a60 (f9400400) 
[  361.456596] ---[ end trace 0000000000000000 ]---
[  361.458174] SPI Interrupt Handler called, IRQ: 55
[  361.458182] Semaphore acquired: 3

sreejitharikath, 2024-11-11:
latest kernel log with both 40mhz in spidev test  
  136.666170] esp32_spi: loading out-of-tree module taints kernel.
[  136.667289] esp32_spi: esp_reset: Resetpin of Host is 506
[  136.667509] esp32_spi: esp_reset: Triggering ESP reset.
[  136.667715] esp32_spi: spi_init: entered spi initilisation
[  136.667767] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[522] DataReady[527]
[  136.667770] esp32_spi: esp_spi_thread: esp spi thread created
[  136.667773] esp32_spi: spi_dev_init: entered  dpi_dev_init fn after intilisation
[  136.667776] esp32_spi: spi_dev_init: set up the adapter ,spi board by configuring board parameters
[  136.667779] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[506] Handshake[522] Dataready[527]
[  136.667782] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
[  136.667786] esp32_spi: spi_busnum_to_master: entered  spi_dev_init fn after intilisation
[  136.668091] esp32_spi: spi_busnum_to_master: alloacted a device  on plattform alloc
[  136.668109] esp32_spi: spi_busnum_to_master: on successful memory alocation to spi controller
[  136.668122] esp32_spi: spi_busnum_to_master: on successful retrival of spi controller
[  136.668201] esp32_spi: spi_busnum_to_master: on successful release of resources on task completion
[  136.668206] esp32_spi: spi_dev_init: on successful claiming of spi bus by setting the bit
[  136.668370] esp32_spi: spi_dev_init: Successfully added new SPI device
[  136.668378] esp32_spi: spi_dev_init: Successfully set up new SPI device
[  136.668392] esp32_spi: spi_dev_init: Successfully obtain goio for handshake
[  136.668555] esp32_spi: spi_dev_init: Successfully request IRQ for Handshake pin
[  136.668568] esp32_spi: spi_dev_init: Successfully obtain gpio for Dataready
[  136.668571] esp32_spi: spi_dev_init: Successfully set GPIO direction of Data ready pin
[  136.668644] SPI Data Ready Interrupt Handler called, IRQ: 56
[  136.668664] Semaphore acquired: 0
[  136.668697] esp32_spi: spi_dev_init: Successfully request IRQ for Dataready pin
[  136.875841] esp32_spi: esp_serial_init: sucessful allocated character device numbers for the driver
[  136.875918] esp32_spi: esp_serial_init: sucessful created device files for an instance in the class
[  136.876364] esp32_spi: spi_init: Successfully intiliased serial interface
[  326.962542] SPI Interrupt Handler called, IRQ: 55
[  326.964302] SPI Interrupt Handler called, IRQ: 55
[  326.965050] SPI Interrupt Handler called, IRQ: 55
[  326.965054] esp32_spi: esp_spi_transaction: entered slave_ready
[  326.965058] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[  326.965061] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[  326.965067] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[  326.965182] SPI Interrupt Handler called, IRQ: 55
[  326.966392] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete
[  326.966396] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[  326.966398] skb->data: 
[  326.966400] 00 
[  326.966402] 00 
[  326.966404] 00 
[  326.966406] 00 
[  326.966408] 00 
[  326.966410] 00 
[  326.966411] 00 
[  326.966413] 00 
[  326.966415] 00 
[  326.966417] 00 
[  326.966418] 00 
[  326.966420] 00 
[  326.966422] 00 
[  326.966424] 00 
[  326.966426] 00 
[  326.966427] 00 
[  326.966429] 00 
[  326.966431] 00 
[  326.966432] 00 
[  326.966434] 00 
[  326.966436] 00 
[  326.966438] 00 
[  326.966439] 00 
[  326.966441] 00 
[  326.966443] 00 
[  326.966444] 00 
[  326.966446] 00 
[  326.966448] 00 
[  326.966450] 00 
[  326.966451] 00 
[... 64 lines cut ...]
[  347.773002] 00 
[  347.773004] 00 
[  347.773006] 00 
[  347.773007] 00 
[  347.773009] 00 
[  347.773011] 00 
[  347.773013] 00 
[  347.773015] 00 
[  347.773016] 00 
[  347.773018] 00 
[  347.773020] 00 
[  347.773022] 00 
[  347.773023] 00 
[  347.773025] 00 
[  347.773027] 00 
[  347.773029] 00 
[  347.773030] 00 
[  347.773032] 00 
[  347.773034] 00 
[  347.773036] 00 
[  347.773037] 00 
[  347.773039] 00 
[  347.773041] 00 

itas@raspberrypi:~/esp32_project/spi_test $

sreejitharikath, 2024-11-11:
latest esp32 log
om app_main()
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3 2nd stage bootloader
I (27) boot: compile time Nov  8 2024 11:40:40
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (101) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (113) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2a714h (173844) map
I (158) esp_image: segment 1: paddr=0003a73c vaddr=3fc9d900 size=05120h ( 20768) load
I (162) esp_image: segment 2: paddr=0003f864 vaddr=40374000 size=007b4h (  1972) load
I (164) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=9078ch (591756) map
I (278) esp_image: segment 4: paddr=000d07b4 vaddr=403747b4 size=190ach (102572) load
I (311) boot: Loaded app from partition at offset 0x10000
I (311) boot: Disabling RNG early entropy source...
I (323) cpu_start: Multicore app
I (332) cpu_start: Pro cpu start user code
I (332) cpu_start: cpu freq: 160000000 Hz
I (332) app_init: Application information:
I (335) app_init: Project name:     network_adapter
I (341) app_init: App version:      FG-0.0.6.0.0
I (346) app_init: Compile time:     Nov  8 2024 11:40:32
I (352) app_init: ELF file SHA256:  33e061803...
I (357) app_init: ESP-IDF:          v5.3
I (362) efuse_init: Min chip rev:     v0.0
I (367) efuse_init: Max chip rev:     v0.99
I (372) efuse_init: Chip rev:         v0.2
I (377) heap_init: Initializing. RAM available for dynamic allocation:
I (384) heap_init: At 3FCA70C8 len 00042648 (265 KiB): RAM
I (390) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (396) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (402) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (409) spi_flash: detected chip: gd
I (413) spi_flash: flash io: dio
I (417) sleep: Configure to isolate all GPIO pins in sleep state
I (424) sleep: Enable automatic switching of GPIO sleep configuration
I (431) coexist: coex firmware version: dab85ae96
I (459) coexist: coexist rom version e7ae62f
I (459) main_task: Started on CPU0
I (460) main_task: Calling app_main()
I (461) NETWORK_ADAPTER: *********************************************************************
I (470) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: FG-0.0.6.0.0
I (478) NETWORK_ADAPTER:                 Transport used :: SPI only
I (487) NETWORK_ADAPTER: *********************************************************************
I (497) NETWORK_ADAPTER: Supported features are:
I (502) NETWORK_ADAPTER: - WLAN over SPI
I (506) ESP_BT: - BT/BLE
I (510) ESP_BT:    - HCI Over SPI
[... 47 lines cut ...]
I (815) wifi_init: WiFi IRAM OP enabled
I (820) wifi_init: WiFi RX IRAM OP enabled
I (826) wifi:mode : null
I (827) SPI_DRIVER: entered generate_startup_event
I (832) SPI_DRIVER: if buffer available for transmision
E (838) SPI_DRIVER: entered queue new transaction

I (844) NETWORK_ADAPTER: Initial set up done
I (849) NETWORK_ADAPTER: entered send event to host
E (854) protocomm_pserial: entered protocomm_pserial_data_ready after initilisation
I (863) protocomm_pserial: item was successfully received from the req_queue of the pserial_cfg structure
I (873) slave_ctrl: event ESPInit
E (877) protocomm_pserial: Successfuly handled protocomm request
E (884) protocomm_pserial: successfully compose tlv
E (889) NETWORK_ADAPTER:  intilisation loop in serial_write_data

E (896) NETWORK_ADAPTER:  data sent len:22

E (901) NETWORK_ADAPTER: Successfully send an item  to a specific queue[0]

E (909) NETWORK_ADAPTER: sends the item (queue_type) to the front of the meta_to_host_queue
I (909) NETWORK_ADAPTER: Data path enabled in send task
I (918) protocomm_pserial: failed both event and request handling
E (918) protocomm_pserial: Sucessfully indicate data ready
I (937) main_task: Returned from app_main()

mantriyogesh, 2024-11-11:
We might need some steps to reproduce this @sreejitharikath .

It looks like the network queue remained paused. we had observed this issues very long back, but exact steps if you could let us know, we can try at our side.

Also please be noted, this issue may need some time to address (maybe in next week)

sreejitharikath, 2024-11-11:
Hi Sir

Please be informed that we have not done any specfice test than control command test available in the test.c available in the directory https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/host/linux/host_control/c_support
we have executed the commands fw_version,get wifi_mode etc.
 And thd got the response below by setting clk frequencies at 10mhz,40mhz and 50 khz by setting the macro SPI_INITIAL_CLK_MHZ in the esp_spi.c as 50khz,10mhz and 40mhz but in the log the frequency is always printing 10mhz its not changing.

itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support $ sudo ./test.out get_fw_version

Enable heartbeat with duration 20

Control response timed out after 30 sec

Response not received

Control response timed out after 30 sec

Response not received

Segmentation fault

Other than that we have spi dev test with the following combinations
1. Rasberry  pi host-50khz,esp32s3-40mhz(changed clk frequency in the python script of rpi as 50khz and  the macro SPI_INITIAL_CLK_MHZ in the esp_spi.c as 50khz
2.  Rasberry  pi host-10mhz,esp32s3-40mhz(changed clk frequency in the python script of rpi as 10mhz and the macro SPI_INITIAL_CLK_MHZ in the esp_spi.c as 10mhz.
3. Rasberry  pi host-40mhz,esp32s3-40mhz(changed clk frequency in the python script of rpi as 40mhz and  the macro SPI_INITIAL_CLK_MHZ in the esp_spi.c as 10mhz.
Here also in the logs the frequency is always printing 10mhz its not changing.

Other than this we have not done any specfic test.
Please let me know if you need clarification of any more things from our side.

mantriyogesh, 2024-11-11:
Did you check the module param in rpi_init.sh?

The config should be provided through it.
The porting guide  also mentions to port rpi_init.sh for your  platform.

Spi slave frequency cannot be just set to anything. It should be supported by slave.

50mhz is nowhere supported for any ESP chipset for spi as slave.

Did you verify your transport ported is correct first?
You have jumped to last step. Check the [Porting Guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) and test [raw throughput](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md) in both directions as first test with low frequency.

sreejitharikath, 2024-11-13:
Hi Sir 

Based on checking the steps mentioned in the porting guide and implementing the rawtp test to check the spi communication we found the following

•	Again had checked hardware connections with respect to the documentation and also verified continuity.
•	Have checked the module param in rpi_init.sh looks fine wrt to the logs.
•	Have checked the gpio configuration by executing the  Cat /sys/kernel/debug/gpio we have got the  response like this

gpiochip0: GPIOs 512-569, parent: platform/fe200000.gpio, pinctrl-bcm2711:
 gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )	
 gpio-515 (GPIO3               )
 gpio-516 (GPIO4               )
 gpio-517 (GPIO5               )
 gpio-518 (GPIO6               )
 gpio-519 (GPIO7               |spi0 CS1            ) out hi ACTIVE LOW
 gpio-520 (GPIO8               |spi0 CS0            ) out hi ACTIVE LOW
 gpio-521 (GPIO9               )
 gpio-522 (GPIO10              |SPI_HANDSHAKE_PIN   ) in  lo IRQ 
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              )
 gpio-525 (GPIO13              )
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              |SPI_DATA_READY_PIN  ) in  hi IRQ 
 gpio-528 (GPIO16              )
 gpio-529 (GPIO17              )
 gpio-530 (GPIO18              )
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              )
 gpio-535 (GPIO23              )
 gpio-536 (GPIO24              )
 gpio-537 (GPIO25              )
 gpio-538 (GPIO26              )
 gpio-539 (GPIO27              )
 gpio-540 (RGMII_MDIO          )
 gpio-541 (RGMIO_MDC           )
 gpio-542 (CTS0                )
 gpio-543 (RTS0                )
 gpio-544 (TXD0                )
 gpio-545 (RXD0                )
 gpio-546 (SD1_CLK             )
 gpio-547 (SD1_CMD             )
 gpio-548 (SD1_DATA0           )
 gpio-549 (SD1_DATA1           )
 gpio-550 (SD1_DATA2           )
 gpio-551 (SD1_DATA3           )
 gpio-552 (PWM0_MISO           )
 gpio-553 (PWM1_MOSI           )
 gpio-554 (STATUS_LED_G_CLK    |ACT                 ) out lo 
 gpio-555 (SPIFLASH_CE_N       )
 gpio-556 (SDA0                )
 gpio-557 (SCL0                )
 gpio-558 (RGMII_RXCLK         )
 gpio-559 (RGMII_RXCTL         )
 gpio-560 (RGMII_RXD0          )
 gpio-561 (RGMII_RXD1          )
 gpio-562 (RGMII_RXD2          )
 gpio-563 (RGMII_RXD3          )
 gpio-564 (RGMII_TXCLK         )
 gpio-565 (RGMII_TXCTL         )
 gpio-566 (RGMII_TXD0          )
 gpio-567 (RGMII_TXD1          )
 gpio-568 (RGMII_TXD2          )
 gpio-569 (RGMII_TXD3          )

gpiochip1: GPIOs 570-577, parent: platform/soc:firmware:gpio, raspberrypi-exp-gpio, can sleep:
 gpio-570 (BT_ON               |shutdown            ) out hi 
 gpio-571 (WL_ON               )
 gpio-572 (PWR_LED_OFF         |PWR                 ) out lo ACTIVE LOW
 gpio-573 (GLOBAL_RESET        )
 gpio-574 (VDD_SD_IO_SEL       |vdd-sd-io           ) out hi 
 gpio-575 (CAM_GPIO            |cam1_regulator      ) out lo 
 gpio-576 (SD_PWR_ON           |regulator-sd-vcc    ) out hi 
 gpio-577 (SD_OC_N             )

•	Checked the gpio for handshake, dataready and reset found like this
 
•	We have got 506,522 and 527 as reset,handshake and spi datat ready respectively. what is meant by new gpio mentioned in the porting guide.
•	Completed raw tp test  to check spi communication based on that found that we have no data available/received in esp32s3side  from host to esp transmission on checking in the esp side.
•	And we have found almost 679kb transmitted from esp side but not received I host rpi side when esp to host transmission had happened.
•	It was found that reception is the common failure that occurs mainly in both devices have got transmission in rpi and we were able to print transmission data printed earlier in control command test but received invalid data in rpi reception.

sreejitharikath, 2024-11-13:
[image]
[image]
[image]

sreejitharikath, 2024-11-13:
[image]
[image]
[image]
[image]

sreejitharikath, 2024-11-13:
[image]

sreejitharikath, 2024-11-13:
log of rpi side when transmitted data from esp to host using rawtp
[ 5105.276432] esp32_spi: esp_spi_transaction: entered slave_ready
[ 5105.276434] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[ 5105.276437] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[ 5105.276440] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[ 5105.277742] SPI Interrupt Handler called, IRQ: 55
[ 5105.277747] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete
[ 5105.277750] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[ 5105.277752] skb->data: 
[ 5105.277754] 00 
[ 5105.277756] 00 
[ 5105.277758] 00 
[ 5105.277760] 00 
[ 5105.277762] 00 
[ 5105.277764] 00 
[ 5105.277766] 00 
[ 5105.277768] 00 
[ 5105.277770] 00 
[ 5105.277772] 00 
[ 5105.277774] 00 
[ 5105.277776] 00 
[ 5105.277778] 00 
[ 5105.277780] 00 
[ 5105.277781] 00 
[ 5105.277783] 00 
[ 5105.277785] 00 
[ 5105.277787] 00 
[ 5105.277789] 00 
[ 5105.277791] 00 
[ 5105.277792] 00 
[ 5105.277794] 00 
[ 5105.277796] 00 
[ 5105.277798] 00 
[ 5105.277800] 00 
[ 5105.277801] 00 
[ 5105.277803] 00 
[ 5105.277805] 00 
[ 5105.277807] 00 
[ 5105.277809] 00 
[ 5105.277811] 00 
[ 5105.277813] 00 

[ 5105.277818] esp32_spi: esp_spi_transaction: entered slave_ready
[ 5105.277821] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[ 5105.277823] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[ 5105.277827] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[ 5105.279126] SPI Interrupt Handler called, IRQ: 55
[ 5105.279132] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete
[ 5105.279134] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[ 5105.279137] skb->data: 
[ 5105.279139] 00 
[ 5105.279141] 00 
[ 5105.279143] 00 
[ 5105.279145] 00 
[ 5105.279147] 00 
[ 5105.279149] 00 
[ 5105.279150] 00 
[ 5105.279152] 00 
[ 5105.279154] 00 
[ 5105.279156] 00 
[ 5105.279158] 00 
[ 5105.279160] 00 
[ 5105.279162] 00 
[ 5105.279164] 00 
[ 5105.279166] 00 
[ 5105.279167] 00 
[ 5105.279169] 00 
[ 5105.279171] 00 
[ 5105.279173] 00 
[ 5105.279175] 00 
[... 906 lines cut ...]
[ 5557.429889] 00 
[ 5557.429891] 00 
[ 5557.429893] 00 
[ 5557.429895] 00 
[ 5557.429897] 00 
[ 5557.429899] 00 
[ 5557.429900] 00 
[ 5557.429902] 00 
[ 5557.429905] 00 
[ 5557.429906] 00 
[ 5557.429908] 00 
[ 5557.429910] 00 
[ 5557.429912] 00 
[ 5557.429914] 00 
[ 5557.429915] 00 
[ 5557.429917] 00 
[ 5557.429919] 00 
[ 5557.429921] 00 
[ 5557.429923] 00 
[ 5557.429925] 00 
[ 5557.429927] 00 
[ 5557.429928] 00 
[ 5557.429930] 00 

itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control $

mantriyogesh, 2024-11-13:
Please lower log like:

```
[ 5557.429867] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[ 5557.429869] skb->data:
[ 5557.429871] 00
[ 5557.429874] 00
[ 5557.429875] 00
[ 5557.429877] 00
[ 5557.429879] 00
[ 5557.429881] 00
[ 5557.429883] 00
[ 5557.429885] 00
[ 5557.429887] 00
[ 5557.429889] 00
[ 5557.429891] 00
[ 5557.429893] 00
[ 5557.429895] 00
[ 5557.429897] 00
[ 5557.429899] 00
[ 5557.429900] 00
[ 5557.429902] 00
[ 5557.429905] 00
[ 5557.429906] 00
[ 5557.429908] 00
[ 5557.429910] 00
[ 5557.429912] 00
[ 5557.429914] 00
[ 5557.429915] 00
[ 5557.429917] 00
[ 5557.429919] 00
[ 5557.429921] 00
[ 5557.429923] 00
[ 5557.429925] 00
[ 5557.429927] 00
[ 5557.429928] 00
[ 5557.429930] 00
```

You can instead print (start 32 bytes of) frame in single line.

From the logs, it is clearly visible, that spi is not working in expected manner.

please state:
1. host in use (specific of raspberry pi)
2. host kernel version
3. spi mode at slave and host
4. git commit used at both (latest master always preferred)
5. ESP chipset used

sreejitharikath, 2024-11-13:
Hi Sir

 Based on our last communication please find the information furnished below.

- •	We are using Raspberry pi4 model B as our host.
•	The kernel version is 6.6.31+rpt-rpi-v8.
•	The Spi mode at both slave and host is mode 2.
•	**Git commit in esp32s3 side**
C:\esp-hosted\esp_hosted_fg\esp\esp_driver\network_adapter>git log -1
commit 250e29294a740a45dd853115a25b62f1b2cfc51b (HEAD, origin/master, origin/HEAD, master)
Merge: f5522ce 5927119
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Wed Sep 25 17:59:26 2024 +0800
Merge branch 'github_auto_deploy' into 'master'
esp_hosted: deploy github automatically
 See merge request app-frameworks/esp_hosted!489
**Git commit in  master host side**
itas@raspberrypi:~/esp-hosted/esp_hosted_fg/host/linux/host_control $ git log -1
commit 250e29294a740a45dd853115a25b62f1b2cfc51b (HEAD -> master, origin/master, origin/HEAD)
Merge: f5522ce 5927119
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Wed Sep 25 17:59:26 2024 +0800
 Merge branch 'github_auto_deploy' into 'master'
 esp_hosted: deploy github automatically
See merge request app-frameworks/esp_hosted!489
•The Espchipset we have used is esp32s3.
[image]

sreejitharikath, 2024-11-13:
Hi Sir 

- Based on the above screenshot available in the porting guide the current reset pins,spi hand shake and spi data ready is 506,522 and 527. but in the new coumn iam finding  a different setting of pins can you please cross check and explain on that part.

- Regarding the clock i have the following doubts
- SPI clock frequency could be changed from macro SPI_CLK_MHZ in esp/esp_driver/network_adapter/main/spi_slave_api.c

1. The macro for setting the clk frequency in the esp32 side was not found in the specfic source file mentioned in the porting guide and also the network adaptor directory we have use 10 mhz as rpi clk and expecting 40mhz in esp since in the code and sdk file in esp code  it is commeted it is 40mhz so assumed the clk for esp is 40mhz can you please confirm on this part.
2. 
[image]

[image]

[image]

And it is also told in porting guide that more than 30hz as clk freq  rpi performance is not reliable.

so based on the inputs our assumption is rpi clk is 10mhz and esp32s3 clk freq is 40mhz.
What should be the exact clock frequency to set communication both rpi and esp32s3 for proper spi communication.

sreejitharikath, 2024-11-13:
log for raw tp transportation from host to esp in host side
From the log we are getting logging statements executed in rpi side indicating esp32 is transmitting some data
[ 5285.267812] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 5285.300688] esp32_spi: loading out-of-tree module taints kernel.
[ 5285.302127] esp32_spi: esp_reset: Resetpin of Host is 506
[ 5285.302358] esp32_spi: esp_reset: Triggering ESP reset.
[ 5285.302522] esp32_spi: spi_init: entered spi initilisation
[ 5285.302606] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[522] DataReady[527]
[ 5285.302613] esp32_spi: spi_dev_init: entered  dpi_dev_init fn after intilisation
[ 5285.302616] esp32_spi: spi_dev_init: set up the adapter ,spi board by configuring board parameters
[ 5285.302619] esp32_spi: spi_dev_init: Config - GPIOs: resetpin[506] Handshake[522] Dataready[527]
[ 5285.302622] esp32_spi: spi_dev_init: Config - SPI: clock[10MHz] bus[0] cs[0] mode[2]
[ 5285.302627] esp32_spi: spi_busnum_to_master: entered  spi_dev_init fn after intilisation
[ 5285.302743] esp32_spi: esp_spi_thread: esp spi thread created
[ 5285.303111] esp32_spi: spi_busnum_to_master: alloacted a device  on plattform alloc
[ 5285.303132] esp32_spi: spi_busnum_to_master: on successful memory alocation to spi controller
[ 5285.303139] esp32_spi: spi_busnum_to_master: on successful retrival of spi controller
[ 5285.303331] esp32_spi: spi_busnum_to_master: on successful release of resources on task completion
[ 5285.303338] esp32_spi: spi_dev_init: on successful claiming of spi bus by setting the bit
[ 5285.303626] esp32_spi: spi_dev_init: Successfully added new SPI device
[ 5285.303635] esp32_spi: spi_dev_init: Successfully set up new SPI device
[ 5285.303660] esp32_spi: spi_dev_init: Successfully obtain goio for handshake
[ 5285.303831] esp32_spi: spi_dev_init: Successfully request IRQ for Handshake pin
[ 5285.303838] esp32_spi: spi_dev_init: Successfully obtain gpio for Dataready
[ 5285.303842] esp32_spi: spi_dev_init: Successfully set GPIO direction of Data ready pin
[ 5285.303882] esp32_spi: spi_dev_init: Successfully request IRQ for Dataready pin
[ 5285.303882] SPI Data Ready Interrupt Handler called, IRQ: 56
[ 5285.303901] Semaphore acquired: 0
[ 5285.510012] esp32_spi: esp_serial_init: sucessful allocated character device numbers for the driver
[ 5285.510177] esp32_spi: esp_serial_init: sucessful created device files for an instance in the class
[ 5285.510393] esp32_spi: spi_init: Successfully intiliased serial interface
[ 6829.328157] SPI Interrupt Handler called, IRQ: 55
[ 6829.329867] esp32_spi: esp_spi_transaction: entered slave_ready
[ 6829.329868] SPI Interrupt Handler called, IRQ: 55
[ 6829.329873] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[ 6829.329876] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[ 6829.329882] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[ 6829.330614] SPI Interrupt Handler called, IRQ: 55
[ 6829.330745] SPI Interrupt Handler called, IRQ: 55
[ 6829.330748] SPI Interrupt Handler called, IRQ: 55
[ 6829.331212] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete
[ 6829.331215] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[ 6830.126513] esp32_spi: esp_spi_transaction: entered slave_ready
[ 6830.126517] SPI Interrupt Handler called, IRQ: 55
[ 6830.126529] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[ 6830.126531] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[ 6830.126536] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[ 6830.127870] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete
[ 6830.127874] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[ 6830.128273] esp32_spi: esp_spi_transaction: entered slave_ready
[ 6830.128274] SPI Interrupt Handler called, IRQ: 55
[ 6830.128278] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[ 6830.128280] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[ 6830.128283] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[ 6830.129021] SPI Interrupt Handler called, IRQ: 55
[ 6830.129152] SPI Interrupt Handler called, IRQ: 55
[ 6830.129156] SPI Interrupt Handler called, IRQ: 55
[ 6830.129588] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete
[ 6830.129591] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[ 6846.472017] esp32_spi: esp_spi_transaction: entered slave_ready
[ 6846.472019] SPI Interrupt Handler called, IRQ: 55
[ 6846.472033] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[ 6846.472035] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[ 6846.472041] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[ 6846.473378] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete
[ 6846.473384] esp32_spi: process_rx_buf: entered process_rx_buf since invalid data
[ 6846.473863] esp32_spi: esp_spi_transaction: entered slave_ready
[... cut ...]
[ 6846.834647] esp32_spi: esp_spi_transaction:  data is ready to be send from slave 
[ 6846.834650] esp32_spi: esp_spi_transaction:  data transmitted over buffer is not valid 
[ 6846.834653] esp32_spi: esp_spi_transaction: entered reception part after initilisation
[ 6846.835384] SPI Interrupt Handler called, IRQ: 55
[ 6846.835515] SPI Interrupt Handler called, IRQ: 55
[ 6846.835518] SPI Interrupt Handler called, IRQ: 55
[ 6846.835950] esp32_spi: esp_spi_transaction: function sends the data (from the tx_buf) and simultaneously waits for the SPI transfer to complete

mantriyogesh, 2024-11-13:
> our assumption is rpi clk is 10mhz and esp32s3 clk freq is 40mhz.

This is max slave spi clock freq supported for S3.
But you are using wires, you should not use this high freq on wires.

Please use 5MHZ first. Get everything working (first to get TX and RX raw throughput working). Once everything tested fine, increase SPI clock from host in 1MHz step. If using wires, you may not go beyond 30MHz. If using PCB, can go till 40MHz.

sreejitharikath, 2024-11-14:
Hi Sir

- •	SPI clock frequency  for esp32s3 could be changed from macro SPI_CLK_MHZ in esp/esp_driver/network_adapter/main/spi_slave_api.c for esp hosted ng setup for fg setup this macro  SPI_CLK_MHZ is not available.
•	The macro for setting the clk frequency in the esp32s3 side was not found in the specfic source file mentioned in the porting guide and also the network adaptor directory for our fg set up.
•	Can you please explain on how to change the spi_clk frequency to lower frequencies and vary for each test.

sreejitharikath, 2024-11-17:
Hi Sir

Given the timing and synchronization challenges with the SPI communication between the ESP32S3 and Raspberry Pi, here’s a step-by-step approach based on which we are currently executing the test.
•	Set Low SPI Frequency: Start with the SPI clock frequency on the Raspberry Pi at 5 MHz. This slower speed helps ensure stable communication by reducing the likelihood of timing mismatches.
•	Incrementally Increase Clock: Once stable communication is verified at 5 MHz we have started the test from 5mhz as initial clk, incrementally raise the SPI clock by 1 MHz and test each step until you reach the desired speed or encounter issues.
•	Test Different SPI Modes: As suggested, try different SPI modes (1, 2, or 3) to determine which is most stable. Ensure that both the Raspberry Pi and ESP32S3 are set to the same SPI mode during each test this done by varying the macro for spi_mode in .esp/esp_driver/network_adapter/main/spi_slave_api.c  in esp32side and for Rpi it is modified by giving the specific  mode corresponding to the spi_mode variable in the rp_init.sh as an argument like spi_mode=1.
•	Error Checking and Debugging: For each configuration, monitor error rates and log any communication issues. If a configuration works at lower speeds but fails at higher ones, the issue may be signal quality or timing. 
•	ESP32S3 SPI Frequency: Since you cannot adjust the SPI frequency on the ESP32S3, focus on optimizing the Raspberry Pi’s settings varying the spi_clk frequency   by modifying the macro SPI_INITIAL_CLK_MHZ as 5mhz in esp_spi.c
•	Based on the following the above mentioned process we have completed 5mhz for mode1,2 and 3 and found that in the communication from host to esp even though transmission is happening from rpi analyzed using logging statements no data was received in esp32s3 side.
•	And Based on the communication from esp to host for the same frequency and modes almost 689kb of data is sent based on checking the logs.
•	An in the rpi side we have found similar issues like which was found in earlier testing scenarios.
•	Logs indicate multiple entries of "data transmitted over buffer is not valid" during esp_spi_transaction.
•	This implies the tx_buf or rx_buf might be corrupted, not fully populated, or accessed by multiple threads before the transaction completes.
•	The Handshake is likely triggering several SPI interrupts (IRQ: 55) in close succession.
•	Based on all the logs shared to supports related to raw tp test from both host to esp and vice versa they have mentioned that timing issues  and issues are there spi communication.
•	No proper data was received in the rpi side and many of the instances we didn’t received dataready interrupt too it was not invoked.
•	Have started tests with 1mhz since 5mhz itself spi communication is failing switched to lower frequencies.
•	For mode1,2 and 3 and found that in the communication from host to esp e using logging statements found that all the issues which awere available in the 5mhz similar issues are recurring and all the communication issues which were occured and observed in 5mhz are similar to new 1mhz reulting same communication failures.
•	Please find the logs for both 5mz and 1mhz as attatched.

sreejitharikath, 2024-11-17:
[rawtp_spi_comm_test (2).zip](https://github.com/user-attachments/files/17789080/rawtp_spi_comm_test.2.zip)
[rawtp_spi_comm_test.zip](https://github.com/user-attachments/files/17789081/rawtp_spi_comm_test.zip)

mantriyogesh, 2024-11-17:
Let me review the logs and comments you have posted tomorrow and get back to you on this.

mantriyogesh, 2024-12-17:
Hello @sreejitharikath ,

Please **disable all the logging**, set the spi mode to 3 at both the places.

Flash the **latest FG master** at **both** the ESP32-S3 and Raspberry Pi (both kernel module, and C app).
If you have verified something, please disable those logs, as the throughput numbers are not visible in your tests, which is actually the main thing to be seen (hidden by all other sorts or logs)

The raw throughput tests documentation is at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md

In the logs, I also notice the logs are printed per byte. Per byte logs are really not desired.

Please back up your existing code somewhere and directly jump to latest master FG code for testing.

Please confirm again individually:
1. Is the Reset pin working fine?
On every reload of driver (using rpi_init.sh), ESP should reset automatically with use of this pin.
2. Is the Handshake interrupt working?
Once confirmed working fine, disable the log.
3. Is the data ready interrupt working?
Once confirmed working fine, disable the log.
4. Ensure both ESP and RPi use exact same SPI mode. for now, use SPI mode 3, **both** the places.
5. Both ESP and RPi use exact same code (latest master most preffered)

Please confirm above, then as next steps we would move to raw throughput.

Also, always provide the logs from start-up.

Keep the SPI freq as 10MHz for now.

sreejitharikath, 2024-12-26:
Hi Sir

[mode 2.zip](https://github.com/user-attachments/files/18249498/mode.2.zip)
[mode 3.zip](https://github.com/user-attachments/files/18249499/mode.3.zip)
Based on the response given by the express if supports had done the following activities.

•	Has Backed up the repositories cloned in the both esp32s and raspberry pi sides .
•	Has cloned latest code repositories on both sides and flashed the latest code on both sides and maintained same git commit on both sides.
•	Have completed the configuration set up on both sides for completing the initial configuration.
•	Have successfully builded the latest code on both sides and flashed the codes on both sides.
•	Rigged up the hardware connections currently without any pull up connections and confirmed continuity in the connections.
•	Set the spi mode to mode 3 and after building using the script rpi_init.sh checked the kernel logs using dmesg found except logs related to reset ,dataready and handshake logs are not appearing even after logs related to pins are added in the respective isr”s.
•	Checked the mode to spi mode 2 which is default found for it all logs are available.
•	So completed raw tp test for mode 3 and mode 2 respectively by setting spi clock as 10mhz using the rpi_init.sh script.
•	Completed the raw tp test for each mode and on both esp to host and host to esp sides  found that similar to earlier stage spi communication is having issues.
•

sreejitharikath, 2024-12-26:
[mode 2.zip](https://github.com/user-attachments/files/18249500/mode.2.zip)
[mode 3.zip](https://github.com/user-attachments/files/18249501/mode.3.zip)

mantriyogesh, 2024-12-26:
There is absolutely no communication from esp32-s3 to raspberry pi.
This is strange.

Let me reproduce and confirm, and provide you success logs when I test with s3 + Raspberry Pi 4 Model B.

mantriyogesh, 2024-12-30:
git commit: 526ae2e2c6efaedad04e3a99f971c5e383eba957

[image]

spi mode: 2 (default)

Raspberry Pi connected with S3: 

[s3_mode2.txt](https://github.com/user-attachments/files/18272406/s3_mode2.txt)
[rpi_spi_mode2_withs3.txt](https://github.com/user-attachments/files/18272407/rpi_spi_mode2_withs3.txt)

mantriyogesh, 2024-12-30:
The first even comes effortlessly. 

I have captured file contents of relavant configurations used at raspberry pi in the host log, including /boot/config.txt, commands- output, dmesg log etc. also esp32-s3 log.

As my rpi is using old kernel, I had changed rpi_init.sh such that, 
resetpin="6"
spi_handshake="22"
spi_dataready="27"

I also have captured (cat_ rpi_init.sh in the logs for your reference.

mantriyogesh, 2025-01-10:
any update @sreejitharikath ?

sreejitharikath, 2025-01-15:
Hi Sir

By comparing the config.txt which you have send with the config .txt which we have used her in rpi we have added three lines of configuration which are
core_freq=250 
core_freq_min=250 
enable_uart=1 
By adding that in our config.txt we have get a logs complete_dmesg_log9 and complete_dmesg_log8.
These logs are attatched below for your reference.
After commenting those lines we are getting complete_dmesg_log10.
I have attatched our config.txt which  is config_new1.txt for your refernce please check through this config.txt and let me know if any modification is required anything has to be added to this config.txt or removed from it to make spi communication functional.
[config_new1.txt](https://github.com/user-attachments/files/18423913/config_new1.txt)

sreejitharikath, 2025-01-15:
[complete_dmesg_log9.txt](https://github.com/user-attachments/files/18423921/complete_dmesg_log9.txt)
[complete_dmesg_log8.txt](https://github.com/user-attachments/files/18423923/complete_dmesg_log8.txt)
[complete_dmesg_log10.txt](https://github.com/user-attachments/files/18423927/complete_dmesg_log10.txt)

[4 more replies]