espressif/esp-hosted #480: Command[1] timed out

ZlTpiger, 2024-09-11:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I use1.0.2 release version,and use Raspberry Pi 4 Model B and esp32,why can not add_network_iface
[image]

ZlTpiger, 2024-09-11:
Both host and esp32 driver are use 1.0.2 release version

mantriyogesh, 2024-09-11:
The spi is not showing correct transaction. 
What is frequency you use right now, what is the spi mode both sides?

mantriyogesh, 2024-09-11:
Also, I am closing this as this counted as feature request. Please open up a bug report, newly.

closed 2024-09-11 by mantriyogesh