espressif/esp-hosted #481: Raspberry Pi won't compile

LingHsvan, 2024-09-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

rpi_init.sh compilation fails and the esp32_sdio.ko cannot be generated, so the driver module cannot be loaded
rpi_init.sh编译失败，esp32_sdio.ko未生成，所以无法加载driver module

### Actual behavior (suspected bug)

[image]

### Error logs or terminal output

```shell
arm-linux-gnueabihf-gcc-12:error: unrecognized argument in option '-mstack-protector-guard=sysreg'
arm-linux-gnueabihf-gcc-12:error: unrecognized command-line option '-mbranch-protection=pac-ret'
arm-linux-gnueabihf-gcc-12:error: unrecognized command-line option '-mstack-protector-guard-reg=sp_el0'; did you mean '-mstack-protector-gaurd-offset='?
insmod: ERROR: could not load module esp32_sdio.ko: No such file or directory
```

### Steps to reproduce the behavior

``` bash
cd esp_hosted_fg/host/linux/host_control/
./rpi_init.sh sdio
```

### Project release version

release/fg-v0.0.5

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

Raspberry Pi OS 64bit

### Shell

Bash

mantriyogesh, 2024-09-12:
Hello @LingHsvan ,

Thank you for pointing this out.

As you are using arm64 machine, and native building,

can you please change accrdingly,
https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L65-L66

for example, change it to:
```
make -j8 target=$IF_TYPE CROSS_COMPILE= KERNEL="/lib/modules/$(uname -r)/build" \ 
 CONFIG_TEST_RAW_TP="$VAL_CONFIG_TEST_RAW_TP" ARCH=arm64

# arch is changed
# cross compile is changed
```

LingHsvan, 2024-09-13:
Thanks, but I'm still having unexpected situations after following this change.

First of all, bt.c return an error, it is easy to fix, if someone else is having the same problem, just add a macro like the latest commit.

[image]
https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c#L255
Change it to
https://github.com/espressif/esp-hosted/blob/20d939491fb20841ae0b56221f249be02ad0ac69/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c#L307-L309

**But then, main.c reported an error, which I couldn't solve for a while**

[image]
https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L445

mantriyogesh, 2024-09-13:
1. https://github.com/espressif/esp-hosted/blob/20d939491fb20841ae0b56221f249be02ad0ac69/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c#L307-L309

2. https://github.com/espressif/esp-hosted/blob/20d939491fb20841ae0b56221f249be02ad0ac69/esp_hosted_fg/host/linux/host_driver/esp32/esp_kernel_port.h#L104-L106

LingHsvan, 2024-09-13:
Ok, thanks, it seems that there is only one bug left now. I searched and found that it seems that the API has changed after Linux kernel 5.17, but I am not familiar with the kernel and need your help as well.
```
ERROR: modpost: "do_exit" [/home/vmx/esp-hosted/esp_hosted_fg/host/linux/host_driver/esp32/esp32_sdio.ko] undefined!
make[3]: *** [/usr/src/linux-headers-6.6.47+rpt-common-rpi/scripts/Makefile.modpost:145: /home/vmx/esp-hosted/esp_hosted_fg/host/linux/host_driver/esp32/Module.symvers] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.47+rpt-common-rpi/Makefile:1898: modpost] Error 2
make[3]: *** [/usr/src/linux-headers-6.6.47+rpt-common-rpi/Makefile:246: _sub-make] Error 2
make[1]: Leaving directory'/usr/src/linux-headers-6.6.47+rpt-rpi-v8'
make: *** [Makefile:58: all] Error 2 
insmod: ERROR: could not load module esp32_sdio.ko: No such file or directory
```

mantriyogesh, 2024-09-13:
Please use latest master. 

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/esp_kernel_port.h

LingHsvan, 2024-09-13:
The problem still exists, and it hasn't changed, even if I overwrite the esp_kernel_port.h of the latest commit directly. So I just inserted 
``` c
#define do_exit(code) kthread_complete_and_exit(NULL, code)
```
into esp_sdio.c and now reports success.

mantriyogesh, 2024-09-13:
Issues you had reported recently are already on latest master, including other big fixes.

You need complete code base to be used, instead of bits and pieces modified. Doing this way is also maintainable in long term.

Use latest master both sides.

LingHsvan, 2024-09-13:
There are new questions. Even though I've upgraded both sides to the latest, the problem persists.
[image]
These attachments are the logs from both sides.
[Full log of Linux host](https://github.com/user-attachments/files/16991184/err.log)
[Log of module](https://github.com/user-attachments/files/16991226/module.log)

mantriyogesh, 2024-09-13:
Have you followed the porting guide?

mantriyogesh, 2024-09-13:
With the length of your wires, I am not sure you can get sdio working.

I am also not sure if, you have :
1. Added ext pull-ups for required pins with required values?
2. Does your esp module need eFuse burning?

We have software ported for Raspberry Pi. but with respect to hardware, there are multiple steps either missing or need confirmation.

mantriyogesh, 2024-09-13:
details of these can be found in [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md)

LingHsvan, 2024-09-13:
I'm using the ESP32-WROVER module, according to the documentation, does not required additional pull-up.

LingHsvan, 2024-09-13:
But my cable length is 10cm, which is really dangerous, and I'll replace it with a shorter Dupont cable when I get back from vacation.

mantriyogesh, 2024-09-13:
i think it is better that you use sdio 1 bit mode, if you must use wires.

with wires, the success rate is less.

mantriyogesh, 2024-09-13:
It is always better For ESP-WROVER the pull-ups external are needed.

This is compatibility issues. the solutions is once you follow 'no-pullup' hyperlink.

[image]

Similarly, check the eFuse burning. if is needed to be done and not done, you would not get any sdio, with jumper or pcb working.

mantriyogesh, 2024-09-13:
Check slowly and step-by-step: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

Also if you use jumpers, do not try 4 bit mode. evaluate using 1-bit mode and then go with pcb and 4 bit mode