espressif/esp-hosted #483: NG mode, scan causes memory leakage

lwj850533527, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The esp32 is running normally
After each memory request, the memory should be released

### Actual behavior (suspected bug)

After many scans, esp32 unable to apply for memory

### Error logs or terminal output

```shell
I (17478579) FW_MAIN: Scan request

I (17485309) FW_MAIN: Scan request

I (17492039) FW_MAIN: Scan request

I (17498769) FW_MAIN: Scan request

I (17505499) FW_MAIN: Scan request

I (17512229) FW_MAIN: Scan request

assert failed: queue_next_transaction spi_slave_api.c:412 (spi_trans->rx_buffer)
Core  0 register dump:
MEPC    : 0x40380700  RA      : 0x4038b47e  SP      : 0x3fcbfc80  GP      : 0x3fc97000  
TP      : 0x3fc9ce48  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000077  S1      : 0x00000001  A0      : 0x3fcbfcbc  A1      : 0x3fc98cd5  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fca8000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcbfde0  
S4      : 0x3fc98cd4  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fcbfc80: 0xa5a5a5a5 0xa5a5a5a5 0x3c0b9698 0x40392966 0x3fc990ac 0x3c0b9698 0x3fc994bc 0x3c0b933c
3fcbfca0: 0x3fc990bc 0x3fcbfcb4 0x3fc990c0 0x3c0b93c0 0x3fc98cd4 0x00323134 0x0000000f 0x65737361
3fcbfcc0: 0x66207472 0x656c6961 0x71203a64 0x65756575 0x78656e5f 0x72745f74 0x61736e61 0x6f697463
3fcbfce0: 0x7073206e 0x6c735f69 0x5f657661 0x2e697061 0x31343a63 0x73282032 0x745f6970 0x736e6172
3fcbfd00: 0x78723e2d 0x6675625f 0x29726566 0x00000000 0x3fcbe040 0x00000000 0x00000001 0x00000001
3fcbfd20: 0x00000003 0x00000004 0x00000000 0x4039052e 0x00000001 0x00000640 0x00000000 0x4039052e
3fcbfd40: 0x00000000 0x00000014 0x3fca84a0 0x40390576 0x00000000 0x00001800 0x00000000 0x4039052e
3fcbfd60: 0x00000001 0x00000640 0x3fcdc710 0x40390576 0x00001800 0x00000008 0x3fca8854 0x40380bc2
3fcbfd80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000008 0x00000640 0x00000000 0x00000000
3fcbfda0: 0x00000000 0x3fcddd38 0x3fcac3f0 0x4200e9b4 0x00000000 0x00000000 0x00000000 0x00000000
3fcbfdc0: 0x00000000 0x00000000 0x00000000 0x4200ea68 0x00000000 0x00000000 0x00000000 0x00000000
3fcbfde0: 0x00000000 0x00000000 0x00000000 0x3fcbff90 0x00000000 0x00000000 0x00000000 0x4038dd3e
3fcbfe00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcbfe20: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fcbfcf0 0x3fca84b4
3fcbfe40: 0x3fc99f3c 0x3fc99f3c 0x3fcbfe38 0x3fc99f34 0x00000003 0x3fcbe06c 0x3fcbe06c 0x3fcbfe38
3fcbfe60: 0x00000000 0x00000016 0x3fcbee34 0x5f697073 0x74736f70 0x6f72705f 0x00736563 0x00000000
3fcbfe80: 0x3fcbfe30 0x00000016 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca8b40 0x3fca8ba8
3fcbfea0: 0x3fca8c10 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42089c6a
3fcbfec0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcbfee0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcbff00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcbff20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcbff40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcbff60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcbff80: 0x00000000 0x00000000 0x3f000000 0x00000014 0x00003200 0x00003200 0x3fcdd0b0 0x3fcdd6f4
3fcbffa0: 0x00000000 0x00000018 0x3c0bff5c 0x3fcc3954 0x3fcc3c04 0x00000000 0x3fcc3bac 0x3fcbffc4
3fcbffc0: 0x0000000c 0x00000000 0x3fcbffd4 0x00000000 0x00000010 0x3c0c0f70 0x3fcbffe8 0x00000000
3fcbffe0: 0x00000000 0x00000014 0x3fcc3d5c 0x00000000 0x5a69eab0 0x6c248236 0x3fcbffe0 0x00000000
3fcc0000: 0x3fca8844 0x00000034 0x00000034 0x0001c6fc 0x3fcc0014 0xe4cd7c22 0x91783cf3 0x3fcc0014
3fcc0020: 0x3fcc0014 0x204411cc 0x00000354 0x00000001 0x3fcc0038 0x3fcc0068 0x00000050 0x00000000
3fcc0040: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc0060: 0x00000000 0x00000000 0x3fcc0014 0x3fcc0014 0x3fcc0014 0x3fcc0014 0x3fcc45c0 0x3fcc0014
```

### Steps to reproduce the behavior

connect to AP no found

### Project release version

1.0.3.0.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

linux

### Shell

ZSH

### Additional context

Does the SSID applied for here need to be released?

[image]

mantriyogesh, 2024-09-13:
@kapilkedawat , @Shreyas0-7 PTAL

Shreyas0-7, 2024-09-16:
Hey @lwj850533527 thanks for pointing out a memory leak, we have raised a internal MR resolving this as well as other potenial fixes. Fix for this will be available soon.

kapilkedawat, 2024-09-24:
closed by https://github.com/espressif/esp-hosted/commit/4df7e5253398e7b7b8edf1bdf58448e6884f127d

closed 2024-09-25 by lwj850533527