espressif/esp-hosted #484: [esp32-p4] hosted bluetooth with nimble only

chegewara, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Hi,
is there any reason that fg hosted for esp32 mcu, esp32 P4 in this case, has enabled bluetooth stack for nimBLE only?
Why we cant use bluedroid?

Thanks

### Use cases

Classic BT, bluedroid BLE with ble mesh.

mantriyogesh, 2024-09-13:
Well it can be used, no limitations as such.
Just short of time, will do it some time later. Choice of nimble, as it is popular and less memory footprint.

We just added nimble as working example. Many people have hooked their favourite bluetooth stack.

In fact, bluedroid can support classic bt and ble.
Although classic bt depends upon slave's bluetooth controller capability.

chegewara, 2024-09-13:
Thanks for answer.
The thing is that nimble probably is not supporting ble mesh 1.1 yet, and espressif is doing very good progress with it.

Also in my case the problem is that my ble mesh library is based on bluedroid, so i cant use it or port it easy.
It is nothing important, just wanted to check the status.

Thanks

mantriyogesh, 2024-09-13:
It is in our tasks, but lower priority.

I think checking the vHCI driver implementation is worth. We also have some easy to understand documentation for porting any stack. Doc : https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/bluetooth_implementation.md

In general, it is not big effort and easily do-able. just might take some time as not in focus

chegewara, 2024-09-14:
BTW are there any plans to add thread, zigbee and other wireless protocols to hosted lib?

mantriyogesh, 2024-09-14:
One by one, yes.

Planning first the thread. No time line commitment yet.