espressif/esp-hosted #485: [ESP32 FG mode, stm32mp131 host, sdio mode] AP, STA, and Bluetooth do not work properly

renliangli, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

AP, STA, and Bluetooth can work normally

### Actual behavior (suspected bug)

After the driver is loaded, the AP, STA, and Bluetooth cannot work properly

### Error logs or terminal output

```shell
root@myd-yf13x:/# insmod esp32_sdio.ko
[  116.717188] esp32_sdio: loading out-of-tree module taints kernel.
[  116.725149] esp_probe: ESP network device detected
root@myd-yf13x:/#
root@myd-yf13x:/# ifconfig -a
eth0      Link encap:Ethernet  HWaddr B6:54:9E:46:DF:6F
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:46 Base address:0x4000

ethap0    Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ethsta0   Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:98 errors:0 dropped:0 overruns:0 frame:0
          TX packets:98 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:8661 (8.4 KiB)  TX bytes:8661 (8.4 KiB)

usb0      Link encap:Ethernet  HWaddr 2E:8D:9C:3B:50:B0
          inet addr:192.168.7.1  Bcast:192.168.7.255  Mask:255.255.255.0
          inet6 addr: fe80::2c8d:9cff:fe3b:50b0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:262 errors:0 dropped:0 overruns:0 frame:0
          TX packets:51 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:21100 (20.6 KiB)  TX bytes:9116 (8.9 KiB)

root@myd-yf13x:/#
root@myd-yf13x:/#
root@myd-yf13x:/# ./test.out
sudo ./test.out
[... 438 lines cut ...]

1. insmod esp32_sdio.ko
2. mknod /dev/esps0 c 221 0
3. chmod 666 /dev/esps0
4. ./test.out softap_start

### Project release version

release/fg-v0.0.5

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Linux version 5.15.67

### Shell

sh

mantriyogesh, 2024-09-13:
1. Please try with the **latest master** at both esp and host.

2. 
```
root@myd-yf13x:/# mknod /dev/esps0 c 221 0
root@myd-yf13x:/# chmod 666 /dev/esps0
```
is not more required or supported.

3. There are module parameters mentioned, unless you use reset pin, it would not work.

4. I suspect that you have not followed the porting guide correctly. Check it step by step and retry.
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md

5. there are no esp side logs attached.

mantriyogesh, 2024-09-13:
Please use latest master code on both esp and host - same git commit.

renliangli, 2024-09-13:
esp side log:

configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7012
load:0x40078000,len:15384
load:0x40080400,len:3816
entry 0x40080648
I (269) boot: ESP-IDF v5.0-beta1-742-g7bd5af7f1e 2nd stage bootloader
I (269) boot: compile time 09:25:50
I (270) boot: chip revision: V301
I (274) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (281) boot.esp32: SPI Speed      : 40MHz
I (286) boot.esp32: SPI Mode       : DIO
I (290) boot.esp32: SPI Flash Size : 4MB
I (295) boot: Enabling RNG early entropy source...
I (300) boot: Partition Table:
I (304) boot: ## Label            Usage          Type ST Offset   Length
I (311) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (319) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (326) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (334) boot:  3 factory          factory app      00 00 00010000 00100000
I (341) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (349) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (357) boot: End of partition table
I (361) boot: Defaulting to factory image
I (366) boot_comm: chip revision: 3, min. application chip revision: 0
I (373) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=28444h (164932) map
I (441) esp_image: segment 1: paddr=0003846c vaddr=3ffbdb60 size=05874h ( 22644) load
I (450) esp_image: segment 2: paddr=0003dce8 vaddr=40080000 size=02330h (  9008) load
I (454) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9c740h (640832) map
I (689) esp_image: segment 4: paddr=000dc768 vaddr=40082330 size=1bbech (113644) load
I (736) esp_image: segment 5: paddr=000f835c vaddr=50000000 size=00010h (    16) load
I (751) boot: Loaded app from partition at offset 0x10000
I (751) boot: Disabling RNG early entropy source...
I (763) cpu_start: Pro cpu up.
I (763) cpu_start: Starting app cpu, entry point is 0x400813dc
I (0) cpu_start: App cpu up.
I (779) cpu_start: Pro cpu start user code
I (779) cpu_start: cpu freq: 240000000 Hz
I (779) cpu_start: Application information:
I (784) cpu_start: Project name:     network_adapter
I (790) cpu_start: App version:      e0bb23b
I (795) cpu_start: Compile time:     Oct 21 2022 09:25:38
I (801) cpu_start: ELF file SHA256:  e190dae174510bac...
I (807) cpu_start: ESP-IDF:          v5.0-beta1-742-g7bd5af7f1e
I (814) heap_init: Initializing. RAM available for dynamic allocation:
I (821) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (827) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (833) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (839) heap_init: At 3FFCCA38 len 000135C8 (77 KiB): DRAM
I (845) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (851) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (858) heap_init: At 4009DF1C len 000020E4 (8 KiB): IRAM
I (865) spi_flash: detected chip: generic
I (869) spi_flash: flash io: dio
W (873) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (887) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (901) NETWORK_ADAPTER: *********************************************************************
I (907) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (915) NETWORK_ADAPTER:                 Transport used :: SDIO only
I (923) NETWORK_ADAPTER: *********************************************************************
I (931) NETWORK_ADAPTER: Supported features are:
I (935) NETWORK_ADAPTER: - WLAN over SDIO
I (939) ESP_BT: - BT/BLE
I (941) ESP_BT:    - HCI Over SDIO
I (943) ESP_BT:    - BT/BLE dual mode
I (947) NETWORK_ADAPTER: capabilities: 0x1d
I (959) BTDM_INIT: BT controller compile version [8020d24]
I (961) system_api: Base MAC address is not set
I (961) system_api: read default base MAC address from EFUSE
I (967) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1293) NETWORK_ADAPTER: ESP Bluetooth MAC addr: e4:65:b8:50:34:76
I (1293) SDIO_SLAVE: Using SDIO interface
I (29791) NETWORK_ADAPTER: Start Data Path

renliangli, 2024-09-14:
> Please use latest master code on both esp and host - same git commit.

The latest master branch code is ok, the problem has been resolved, thank you.

closed 2024-09-14 by renliangli