espressif/esp-hosted #486: build error: esp driver fg :

preveen-stack, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

build success

### Actual behavior (suspected bug)

build error 

### Error logs or terminal output

```shell
esp-hosted/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c:2602:35: error: 'wifi_event_ap_stadisconnected_t' has no member named 'reason'
 2602 |         ntfy_payload->reason = evt->reason;
      |
```

### Steps to reproduce the behavior

follow the default instructions for building 

### Project release version

latest

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

raspberrypi os

### Shell

Bash

### Additional context

target is esp32, spi transport,

mantriyogesh, 2024-09-13:
fixed in https://github.com/espressif/esp-hosted/commit/3b27bef75fc37bd4e7c11a1bbbb3e0aa7f44d901

preveen-stack, 2024-09-13:
> esp-hosted/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c:2602:35: error: 'wifi_event_ap_stadisconnected_t' has no member named 'reason'
 2602 |         ntfy_payload->reason = evt->reason;
      |                                   ^~
ninja: build stopped: subcommand failed.

> ~/esp-hosted/esp_hosted_fg/esp/esp_driver/network_adapter $ git log --oneline
20d9394 (HEAD -> master, origin/master, origin/HEAD) Merge branch 'fix/esp32_iram' into 'master'
3b27bef free tram for esp32, idc version specific fix

I am reporting on the latest master , I have that commit. But still build error is shown

preveen-stack, 2024-09-13:
The fix is similar to that of [3b27bef](https://github.com/espressif/esp-hosted/commit/3b27bef75fc37bd4e7c11a1bbbb3e0aa7f44d901)
PR raised for the same #487

commit 66f7f374 (2024-09-16): fix: build error: slave_control.c

fixes #486
closed 2024-09-16 by espressif-bot