espressif/esp-hosted #489: NG: Unable to communicate over SPI

sergey-suloev, 2024-09-16:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected to have SPI communication work as before.

### Actual behavior (suspected bug)

I started experiencing SPI communication issues since I have updated my codebase from the current master branch.
The problem is that the NG host driver can't get bootup event anymore. I investigated the issue with logic analyzer and what seems strange to me is that ESP does not rise DATA-READY event before sending bootup packet. This is most likely the reason why IRS isn't called and the packet doesn't arrive. I verified all physical connections again and they all look fine.
Could you get an advice what I should try further, please.

### Error logs or terminal output

```shell
root@nanopi-a64-bookworm:~# dmesg | grep esp
[    6.655317] esp32_spi: esp_spi_probe: Probing ESP32 SPI-driver...
[    6.669734] esp32_spi: esp_spi_probe: raw-tp-mode was not specified
[    6.683096] esp32_spi: esp_spi_probe: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [1], SPI mode [0x02]
```

### Steps to reproduce the behavior

NG host driver with ESP32

### Project release version

latest

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Debian Bookworm

### Shell

Bash

### Additional context

Allwinner A64 (sun50i) / NanoPi A64

[Session1.sal.zip](https://github.com/user-attachments/files/17015202/Session1.sal.zip)

sergey-suloev, 2024-09-16:
[image]

mantriyogesh, 2024-09-17:
1. First transaction is host driven always. It is not dependent upon data ready. Data ready comes in picture for second onwards transaction.
2. Is is code commented yet: https://github.com/espressif/esp-hosted/pull/425/files#diff-376c997542ed8caae30ab0a0d30cbff66a894804f1e92e8dcee9b1cc6aa3fcfdL963 ?

sergey-suloev, 2024-09-17:
@mantriyogesh do you mean hardware reset ?

mantriyogesh, 2024-09-17:
I think it would be better to try this with out modifying latest master code.

The sequence of the steps matter a lot

mantriyogesh, 2024-09-18:
Did you get a chance to try the vanilla latest master?

sergey-suloev, 2024-09-18:
> Did you get a chance to try the vanilla latest master?

It is hard to integrate your driver in my system , because it is written old-style without device tree

mantriyogesh, 2024-09-18:
As you correctly pointed, we need to move to newer way to support device tree. however, we also need to ensure that we also support these using module params or hardcoding, which is currently supported in FG driver.

Worth to check what raspberry pi os you use : https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md#42-reset-pin-and-spi-gpio-numbering

For newer Raspberry Pi OS :  https://github.com/espressif/esp-hosted/blob/ce00ca53189ffd52455dd5426d54f9be6fae1871/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L13-L28

For older Raspberry Pi OS: https://github.com/espressif/esp-hosted/blob/ce00ca53189ffd52455dd5426d54f9be6fae1871/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L33-L36

> Are the numbers 234 and 235 those you need for your driver?

No. But then I am not sure which GPIOs do you use to connect. ideally, same GPIO numbers you can use.
As you are already aware, the `Slave Reset`, `Data Ready` and `Handshake` GPIOs could be freely chosen by user. just make sure they are not already used.

sergey-suloev, 2024-09-18:
I am not using rpi but allwinner platform . There is another napping scheme from platform specific numbers to universal gpio numbers .  The numbers I sent you are probably not correct because they look like indexes within a specific gpio controller

mantriyogesh, 2024-09-18:
understood.. 
The numbers are linear. abstraction to the actual controller.

sergey-suloev, 2024-09-18:
There's probably a way to get the correct numbers using API and log them , I'll see if I can do it

mantriyogesh, 2024-09-18:
I also don't wish to alarm you unnecessarily, but in past, in some specific allwinner board (t507) , we had to introduce some extra logic transition.

We observed that the spi driver was not processing every alternate message due to issue in spi linux driver.

[465.patch.txt](https://github.com/user-attachments/files/17041108/465.patch.txt)

> [ ! IMPORTANT ]
>
> This is based on some old commit.
> You cannot directly apply this change on your repo.

This is internal MR that we had, it has huge unnecessary changes for you. You can only focus on one file, network_adapter/main/spi_slave_api.c

check symbols, `//reset_handshake_gpio();`, `wait_cs_deassert_sem` and `gpio_set_intr_type(gpio_num, GPIO_INTR_ANYEDGE);`
ignore `pkt_num` change. if you add pkt_num, please ensure you use at both sides, esp and host. as it is changing header.

Please disregard this, if you do not face any packet drop as such.

sergey-suloev, 2024-09-18:
Some changes made recently causes my issue . It was working well for a long time before merging with latest master

mantriyogesh, 2024-09-18:
can you please only point me to the latest commit that caused this issue? it might be worth to see the specific change, instead of complete driver.

sergey-suloev, 2024-09-18:
@mantriyogesh  could you provide some help on how to compile your driver, please ?

mantriyogesh, 2024-09-18:
Actually, your code much resembles to FG, you have many FG code inherited.

https://github.com/sergey-suloev/esp-hosted/blob/8331c2ee5ae5fa3436beff39bb2bc7bfff957859/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L431

Just change the rpi_init.sh for correct GPIOs and run it without args, it would  print usage.

sergey-suloev, 2024-09-18:
I was able to build and run the driver, here is my log 

[  243.462435] esp32_spi: loading out-of-tree module taints kernel.
[  243.677432] esp32_spi: spi_dev_init: Failed to obtain SPI master handle
[  244.096684] WARNING: Flushing system-wide workqueues will be prohibited in near future.
[  244.104784] CPU: 1 PID: 267 Comm: modprobe Tainted: G           O       6.10.10-sunxi #stable
[  244.113330] Hardware name: FriendlyARM NanoPi A64 (DT)
[  244.118484] Call trace:
[  244.120941]  dump_backtrace+0x94/0x114
[  244.124715]  show_stack+0x18/0x24
[  244.128046]  dump_stack_lvl+0x74/0x8c
[  244.131727]  dump_stack+0x18/0x24
[  244.135056]  __warn_flushing_systemwide_wq+0x1c/0x28
[  244.140041]  spi_exit+0x80/0x250 [esp32_spi]
[  244.144370]  esp_init_interface_layer+0x398/0x648 [esp32_spi]
[  244.150163]  esp_init+0x14c/0x1000 [esp32_spi]
[  244.154655]  do_one_initcall+0x44/0x28c
[  244.158508]  do_init_module+0x60/0x21c
[  244.162279]  load_module+0x1e98/0x2088
[  244.166048]  init_module_from_file+0x88/0xcc
[  244.170338]  __arm64_sys_finit_module+0x244/0x33c
[  244.175064]  invoke_syscall+0x48/0x114
[  244.178835]  el0_svc_common.constprop.0+0x40/0xe0
[  244.183561]  do_el0_svc+0x1c/0x28
[  244.186895]  el0_svc+0x34/0xd8
[  244.189968]  el0t_64_sync_handler+0x120/0x12c
[  244.194342]  el0t_64_sync+0x190/0x194
[  244.199802] esp32_spi: spi_init: Failed Init SPI device

mantriyogesh, 2024-09-19:
Firmware loaded with same variant? On same commit?

sergey-suloev, 2024-09-19:
> Firmware loaded with same variant? On same commit?

Yes, sure

mantriyogesh, 2024-09-19:
Let me know if you face issue on FG also.

sergey-suloev, 2024-09-22:
Btw you claim that NG works with Raspberry Pi, I tried and found that it doesn't work either.

mantriyogesh, 2024-09-22:
@kapilkedawat , can you please check this?

sergey-suloev, 2024-09-22:
root@nanopi-a64-bookworm:~# dmesg | grep esp
[    7.682638] esp32_spi: esp_init: esp_init: begin
[    7.697798] esp32_spi: esp_spi_probe: Probing ESP32 SPI-driver...
[    7.713372] esp32_spi: esp_spi_probe: raw-tp-mode was not specified
[    7.713617] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 0
[    7.719706] esp32_spi: esp_spi_probe: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [1], SPI mode [0x02]
[    7.745100] esp32_spi: esp_spi_work: SPI work: handshake 0, data_ready 1, data_path 0
[    8.160067] esp32_spi: esp_spi_probe: Probe success.
[    8.165447] esp32_spi: esp_init: esp_init: end 0
[    9.967186] esp32_spi: esp_spi_work: SPI work: handshake 0, data_ready 1, data_path 1
[    9.975275] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 1, data_path 1
[    9.999486] esp32_spi: process_rx_buf: SPI RX buf: IF type = 6
[   10.007454] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1

mantriyogesh, 2024-09-23:
1. Are you using jumper cables?
2. What is the ESP chipset you use? Is it ESP32?
3. Can you also share ESP side log?
4. Does spi mode 1 or spi mode 3 (both sides same), also faces this issue?
5. Does this issue arise only on module reload or even fresh load first time after system bootup?

sergey-suloev, 2024-09-23:
On sunxi platform it seems to work well in SPI mode 1 only.

mantriyogesh, 2024-09-23:
Can you please attach success and failed log for sunxi? I meant the logic analyser output

sergey-suloev, 2024-09-23:
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7120
load:0x40078000,len:15616
load:0x40080400,len:4
load:0x40080404,len:3888
entry 0x40080650
I (29) boot: ESP-IDF e7771c75-dirty 2nd stage bootloader
I (29) boot: compile time Sep 23 2024 16:24:04
I (29) boot: Multicore bootloader
I (34) boot: chip revision: v3.1
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 factory          factory app      00 00 00010000 00100000
I (98) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (105) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (113) boot: End of partition table
I (117) boot: Defaulting to factory image
I (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=205a4h (132516) map
I (178) esp_image: segment 1: paddr=000305cc vaddr=3ffbdb60 size=05884h ( 22660) load
I (188) esp_image: segment 2: paddr=00035e58 vaddr=40080000 size=0a1c0h ( 41408) load
I (205) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8ad44h (568644) map
I (411) esp_image: segment 4: paddr=000cad6c vaddr=4008a1c0 size=15a20h ( 88608) load
I (464) boot: Loaded app from partition at offset 0x10000
I (464) boot: Disabling RNG early entropy source...
I (475) cpu_start: Multicore app
I (476) cpu_start: Pro cpu up.
I (476) cpu_start: Starting app cpu, entry point is 0x40081448
I (0) cpu_start: App cpu up.
I (493) cpu_start: Pro cpu start user code
I (493) cpu_start: cpu freq: 240000000 Hz
I (494) cpu_start: Application information:
I (498) cpu_start: Project name:     network_adapter
I (504) cpu_start: App version:      NG-1.0.3.0.0
I (509) cpu_start: Compile time:     Sep 23 2024 16:23:57
I (515) cpu_start: ELF file SHA256:  304f5b83d2a319cc...
I (521) cpu_start: ESP-IDF:          e7771c75-dirty
I (527) cpu_start: Min chip rev:     v0.0
I (532) cpu_start: Max chip rev:     v3.99 
I (536) cpu_start: Chip rev:         v3.1
I (541) heap_init: Initializing. RAM available for dynamic allocation:
I (549) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (554) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (560) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (567) heap_init: At 3FFC7D18 len 000182E8 (96 KiB): DRAM
I (573) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (579) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (586) heap_init: At 4009FBE0 len 00000420 (1 KiB): IRAM
I (593) spi_flash: detected chip: generic
I (596) spi_flash: flash io: dio
I (601) coexist: coex firmware version: 77cd7f8
I (606) app_start: Starting scheduler on CPU0
I (611) app_start: Starting scheduler on CPU1
mI (611) main_task: Started on CPU0
I (620) main_task: Calling app_main()
I (624) stats: *********************************************************************
I (632) stats:                 ESP-Hosted Firmware version :: NG-1.0.3.0.0                        
I (642) stats:                 Transport used :: SPI only                      
I (650) stats: *********************************************************************
I (658) FW_MAIN: Supported features are:
I (663) FW_MAIN: - WLAN over SPI
I (667) FW_BT: - BT/BLE
I (670) FW_BT:    - HCI Over SPI
I (674) FW_BT:    - BT/BLE dual mode
I (678) FW_MAIN: Capabilities: 0xf8
I (693) wifi:wifi driver task: 3ffbd694, prio:23, stack:6656, core=1
I (695) wifi:wifi firmware version: d23b633
I (696) wifi:wifi certification version: v7.0
I (696) wifi:config NVS flash: disabled
I (700) wifi:config nano formating: disabled
I (704) wifi:Init data frame dynamic rx buffer num: 40
I (709) wifi:Init static rx mgmt buffer num: 5
I (713) wifi:Init management short buffer num: 32
I (718) wifi:Init dynamic tx buffer num: 64
I (722) wifi:Init static rx buffer size: 1600
I (726) wifi:Init static rx buffer num: 16
I (729) wifi:Init dynamic rx buffer num: 40
I (734) wifi_init: rx ba win: 32
I (737) wifi_init: tcpip mbox: 32
I (741) wifi_init: udp mbox: 6
I (745) wifi_init: tcp mbox: 6
I (749) wifi_init: tcp tx win: 5760
I (753) wifi_init: tcp rx win: 5760
I (757) wifi_init: tcp mss: 1440
I (761) wifi_init: WiFi IRAM OP enabled
I (765) wifi_init: WiFi RX IRAM OP enabled
I (771) BTDM_INIT: BT controller compile version [a38dc5c]
I (777) BTDM_INIT: Bluetooth MAC: e4:65:b8:53:f9:0a
I (784) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (1164) FW_MAIN: ESP Bluetooth MAC addr: e4-65-b8-53-f9- a

I (1165) FW_SPI: Using SPI interface
I (1166) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1174) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1183) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1192) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2204) FW_MAIN: Initial set up done
I (2204) main_task: Returned from app_main()
I (3376) FW_MAIN: INIT Interface command

I (3380) wifi:mode : sta (e4:65:b8:53:f9:08)
I (3381) wifi:enable tsf
I (3382) FW_CMD: station started and disabled softap mode
I (3615) FW_MAIN: Get MAC command

I (3826) FW_MAIN: Get Tx power command

I (8531) FW_MAIN: REG set command

mantriyogesh, 2024-09-24:
> Btw you claim that NG works with Raspberry Pi, I tried and found that it works a little better - it can receive bootup event, but bluetooth doesn't work either.
> 
> root@rpi3-aplus-arm64-bookworm:~# dmesg | grep esp [ 7.838095] esp32_spi: esp_init: esp_init: begin [ 7.845239] esp32_spi: esp_spi_probe: Probing ESP32 SPI-driver... [ 7.859765] esp32_spi: esp_spi_probe: raw-tp-mode was not specified [ 7.861923] esp32_spi: esp_spi_probe: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [1], SPI mode [0x02] [ 8.300393] esp32_spi: esp_spi_probe: Probe success. [ 10.762062] esp32_spi: esp_init: esp_init: end 0 [ 10.774853] esp32_spi: process_esp_bootup_event: Received ESP bootup event [ 10.780822] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3 [ 10.783628] esp32_spi: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SPI [ 10.783658] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2 [ 10.783664] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0 [ 10.783669] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1 [ 10.783673] esp32_spi: process_fw_data: ESP chipset's last reset cause: [ 10.783678] esp32_spi: print_reset_reason: POWERON_RESET [ 10.783687] esp32_spi: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0 [ 10.784718] esp32_spi: esp_reg_notifier: Driver init is ongoing [ 11.827393] esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop [ 16.862570] esp32_spi: wait_and_decode_cmd_resp: Command[0x3] timed out [ 16.870931] esp32_spi: cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22 [ 16.882774] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1 [ 16.892394] esp32_spi: process_event_esp_bootup: network iterface init failed [ 24.572884] esp32_spi: esp_reg_notifier: Driver init is ongoing
> 
> root@rpi3-aplus-arm64-bookworm:~# dmesg | grep Bluetooth [ 7.787995] Bluetooth: Core ver 2.22 [ 7.803250] Bluetooth: HCI device and connection manager initialized [ 7.813743] Bluetooth: HCI socket layer initialized [ 7.826112] Bluetooth: L2CAP socket layer initialized [ 7.844690] Bluetooth: SCO socket layer initialized [ 12.966545] esp32_spi: init_bt: ESP Bluetooth init [ 14.567976] Bluetooth: BNEP (Ethernet Emulation) ver 1.3 [ 14.576024] Bluetooth: BNEP filters: protocol multicast [ 14.576061] Bluetooth: BNEP socket layer initialized [ 15.010549] Bluetooth: hci0: Opcode 0x0c03 failed: -110
> 
> root@rpi3-aplus-arm64-bookworm:~# hciconfig hci0: Type: Primary Bus: SPI BD Address: 00:00:00:00:00:00 ACL MTU: 0:0 SCO MTU: 0:0 DOWN RX bytes:0 acl:0 sco:0 events:0 errors:0 TX bytes:3 acl:0 sco:0 commands:0 errors:0
> 
> The SPI exchange doesn't look good to me, I am not sure what's wrong.
> 
> [Session2.sal.zip](https://github.com/user-attachments/files/17089483/Session2.sal.zip)

[image]

In this, clearly is not falling under correct timing. I used mode 2 for decode here.

Can you please show the photo how you had connected this setup?

mantriyogesh, 2024-09-24:
From the log, the current chipset in use is classic esp32.

Can you also attach the esp32 side sdkconfig is use? This would help us know the configurations used.

The ESP32 SPI timing considerations could be referred from https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#adjust-spi-timings . Also note [SPI timing considerations](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#speed-and-timing-considerations): 

> If DMA is enabled, a Device's launch edge is half of an SPI clock cycle ahead of the normal time, shifting to the Master's actual latch edge. In this case, if the GPIO matrix is bypassed, the hold time for data sampling is 68.75 ns and no longer a half of an SPI clock cycle. If the GPIO matrix is used, the hold time will increase to 93.75 ns. The Host should sample the data immediately at the latch edge or communicate in SPI modes 1 or 3.

Question is, can you prefer MODE 1 /3 when esp32 is used?
Are the GPIOs used from IO_MUX or used over GPIO_Matrix?

mantriyogesh, 2024-09-24:
[489_sal_2.zip](https://github.com/user-attachments/files/17107374/489_sal_2.zip)

sergey-suloev, 2024-09-25:
Mode 1 is fully non-functional when RaspberryPi is in use.

mantriyogesh, 2024-09-26:
Can you show me the connections photo for raspberry pi?

Is the wires of raspberry pi long and unequal lengths?

sergey-suloev, 2024-09-26:
> Can you show me the connections photo for raspberry pi?
> 
> Is the wires of raspberry pi long and unequal lengths?

I already sent you a picture of my connections. It hasn't changed since that time. I make connections with 10cm wires. They are of equal length.

mantriyogesh, 2024-09-26:
Sorry , I think there is disconnect, I couldn't find the raspberry pi connections from start of the thread..

Anyway, as the timing diagram suggests the mode 2 has timing difference than mode 1 and 3 (only for ESP32). Also please note that the doc says the half clock ahead is only done for higher frequencies. In your case you used 1m, where we don't expect the early clock.

In our setup we observed no issues wrt timing, but we use PCB and don't rely on jumper cables.

If the spi mode 1 works , I think better to stick with it.
Alternatively would it be possible to ship your board?

sergey-suloev, 2024-09-26:
> Sorry , I think there is disconnect, I couldn't find the raspberry pi connections picture (or your board picture) from start of the thread..
> 
> Anyway, as the timing diagram in the ESP32 spi slave document (lils above), suggests the mode 2 has timing difference than mode 1 and 3 (only for ESP32). Also please note that the doc says the half clock ahead is only done for higher frequencies. In your case you used 1m, where we don't expect the early clock.
> 
> In our setup we observed no issues wrt timing, but we use PCB and don't rely on jumper cables.
> 
> If the spi mode 1 works , I think better to stick with it. Alternatively would it be possible to ship your board?

I do not use any custom boards. I have RPI3A+ and NanoPi A64.  I also have a couple of other  sunxi-based boards OrangePi H3-based and BananaPi R16-based. 
All of them have totally the same problems with ESP32.

sergey-suloev, 2024-09-27:
I tried ESP32S3 with sunxi board and it starts well in both mode 1 & 2. But when I issue a  scan in bluetoothctl the timeouts happen.

[   56.534525] bt_tx: 00000000196c5484: 05 20 06 6f 54 8d 84 3b 28                       . .oT..;(
[   56.543596] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   56.551517] tx: 00000000678c858c: 02 00 00 00 09 00 0f 00 7d 02 00 00 00 00 01 05  ........}.......
[   56.560660] tx: 0000000087d13a68: 20 06 6f 54 8d 84 3b 28 74 6e 61 6d 65 3d 3f 20   .oT..;(tname=? 
[   56.576822] esp32_spi: process_rx_buf: SPI RX buf: IF type = 14
[   56.584372] SPI RX buf: 00000000db5e44e3: fe 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   56.594126] SPI RX buf: 00000000d1d708b7: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   56.603891] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   58.559201] Bluetooth: hci0: command 0x2005 tx timeout
[   58.559209] Bluetooth: hci0: Opcode 0x2005 failed: -110
[   58.569751] bt_tx: 00000000926bb0be: 41 20 0d 00 00 05 01 12 00 12 00 01 36 00 36 00  A ..........6.6.
[   58.579123] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   58.586973] tx: 00000000678c858c: 02 00 00 00 10 00 10 00 29 01 00 00 00 00 01 01  ........).......
[   58.596033] tx: 0000000087d13a68: 41 20 0d 00 00 05 01 12 00 12 00 01 36 00 36 00  A ..........6.6.
[   58.607353] esp32_spi: process_rx_buf: SPI RX buf: IF type = 14
[   58.613293] SPI RX buf: 00000000e38a59ea: fe 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   58.623048] SPI RX buf: 000000002f4fe383: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   58.634962] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   60.575197] Bluetooth: hci0: Opcode 0x2041 failed: -110
[   60.607206] Bluetooth: hci0: command 0x2005 tx timeout

mantriyogesh, 2024-09-28:
Did you get a chance to see https://github.com/espressif/esp-hosted/issues/489#issuecomment-2357860456?

This we used over sunxi last time. But we did not on master, for all, as we tested only one allwinner sunxi board.

The change is in spi transport, that time we were testing FG for this board. There are some timing changes in this code. As far as I remember, The default with this code, runs FG  in raw throughput mode.

mantriyogesh, 2024-09-28:
Also is there photo that you sent ? I cannot find reference of.

sergey-suloev, 2024-09-29:
> Did you get a chance to see [#489 (comment)](https://github.com/espressif/esp-hosted/issues/489#issuecomment-2357860456)?
> 
> This we used over sunxi last time. But we did not on master, for all, as we tested only one allwinner sunxi board.
> 
> The change is in spi transport, that time we were testing FG for this board. There are some timing changes in this code. As far as I remember, The default with this code, runs FG in raw throughput mode.

Yes, I tried it. It had no effect

sergey-suloev, 2024-09-29:
RaspberryPi  with esp32s3

root@rpi3-aplus-arm64-bookworm:~# dmesg | grep spi
[    6.599493] spi spi0.0: setting up native-CS0 to use GPIO
[    7.298502] esp32_sdio:esp32_spi: esp_init: esp_init: begin
[    7.311768] esp32_sdio:esp32_spi: esp_spi_probe: Probing ESP SPI-driver...
[    7.332242] esp32_sdio:esp32_spi: esp_spi_probe: raw-tp-mode was not specified
[    7.335646] esp32_sdio:esp32_spi: esp_spi_probe: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [30], SPI mode [0x02]
[    7.351787] esp32_sdio:esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop
[    7.364503] esp32_sdio:esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop
[    7.766542] esp32_sdio:esp32_spi: esp_spi_probe: ESP SPI-driver probe success
[    9.046204] esp32_sdio:esp32_spi: process_esp_bootup_event: Received ESP bootup event
[    9.054218] esp32_sdio:esp32_spi: esp_init: esp_init: end 0
[    9.056233] esp32_sdio:esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[    9.056256] esp32_sdio:esp32_spi: esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
[    9.056263] esp32_sdio:esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[    9.056267] esp32_sdio:esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[    9.056271] esp32_sdio:esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[    9.056276] esp32_sdio:esp32_spi: process_fw_data: ESP chipset's last reset cause:
[    9.056280] esp32_sdio:esp32_spi: print_reset_reason: POWERON_RESET
[    9.056288] esp32_sdio:esp32_spi: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
[   10.273303] esp32_sdio:esp32_spi: esp_reg_notifier: Driver init is ongoing
[   10.287277] esp32_sdio:esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop
[   15.326823] esp32_sdio:esp32_spi: wait_and_decode_cmd_resp: Command[0x1] timed out
[   15.328063] esp32_sdio:esp32_spi: cmd_init_interface: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   15.334010] esp32_sdio:esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   15.399068] esp32_sdio:esp32_spi: process_event_esp_bootup: network interface init failed
[   24.033902] esp32_sdio:esp32_spi: esp_reg_notifier: Driver init is ongoing

mantriyogesh, 2024-09-30:
Wrt,

https://github.com/espressif/esp-hosted/issues/489#issuecomment-2381388013

Did you try the code as is or with changes?

If you use code as is, what are the issues? In this change the low logs are added, to understand any wrong behaviour. Can you send us the complete textual logs both sides (ESP and Linux)?
If you can capture the logic analyser with this change, it would be also useful.

By the time, @kapilkedawat , can we re-verify hosted-ng on raspberry pi setup?

sergey-suloev, 2024-09-30:
> Wrt,
> 
> [#489 (comment)](https://github.com/espressif/esp-hosted/issues/489#issuecomment-2381388013)
> 
> Did you try the code as is or with changes?
> 
> If you use code as is, what are the issues? In this change the low logs are added, to understand any wrong behaviour. Can you send us the complete textual logs both sides (ESP and Linux)? If you can capture the logic analyser with this change, it would be also useful.
> 
> By the time, @kapilkedawat , can we re-verify hosted-ng on raspberry pi setup?

No, not as is  I tried it in NG. I only took parts that you mentioned regarding the semaphore

sergey-suloev, 2024-09-30:
Regarding Sunxi and Esp32 in mode 1. I said it works well , and it is mostly looks fine . But in fact when I send audio it may sometimes be distorted and I can see dropped packets in console .

mantriyogesh, 2024-10-02:
> UPDATE: I was able to make SPI communication a little more stable by changing spi-sun6i kernel driver. I commented out the following line https://elixir.bootlin.com/linux/v6.10.11/source/drivers/spi/spi-sun6i.c#L368 in order for the sunxi hardware to control CS line automatically .

Oh, i assumed, it is taking care on its own in linux spi driver.
Now, does it have any better effect on mode 3?

Shreyas0-7, 2024-10-03:
Hey @sergey-suloev can you please check and inform are you using 32 bit architecture or 64 bit architecture for host?

sergey-suloev, 2024-10-03:
> Hey @sergey-suloev can you please check and inform are you using 32 bit architecture or 64 bit architecture for host?

For Rpi I am using aarch64, and for Sunxi both aarch64 and armhf, depending on a soc installed onboard

Shreyas0-7, 2024-10-03:
Hey @sergey-suloev I think it's because of aarch64 it is failing. We are not able to read the header correctly and failing to process the boot up. We are able to reproduce the issue and will resolve it.

sergey-suloev, 2024-10-03:
> Oh, i assumed, it is taking care on its own in linux spi driver. Now, does it have any better effect on mode 3?

Interesting but it works with the same stability when I switch host to mode 3 but esp is still in mode 1 ))

sergey-suloev, 2024-10-06:
> Manual cs would change the timings. The spi world has these timing interop issues to be tuned. Once you tune you spi timing, you are all set
> 
> I found some sunxi platform reference, https://linux-sunxi.org/SPIdev
> 
> Please check the end of the section for any 'Bugs/Caveats'. I’m not certain if this applies to your situation, but I thought it might be useful to point out in case it’s relevant.

The changes described are already implemented in kernel stable as of v6.12

sergey-suloev, 2024-10-13:
I am trying esp32s3 now with orangepi-pc board (Allwinner H3, armhf). 
With using SPI mode 1, SPI freq 1MHz , Bootup event is parsed well. 
Any attempt to start LE scan gives timeouts.

[   68.763864] bt_tx: 6aa910d2: 05 20 06 41 8b e2 eb 44 03                       . .A...D.
[   68.772168] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   68.780126] tx: 4c59aaa3: 02 00 00 00 09 00 0f 00 fd 03 00 00 6e 69 01 05  ............ni..
[   68.788581] tx: 3779299c: 20 06 41 8b e2 eb 44 03 3d 6c 6f 67 5f 74 61 69   .A...D.=log_tai
[   68.810758] esp32_spi: process_rx_buf: SPI RX buf: IF type = 14
[   68.816736] SPI RX buf: 9f3910ae: fe 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   68.825821] SPI RX buf: f381df79: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   68.834930] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   70.794296] Bluetooth: hci0: command 0x2005 tx timeout
[   70.794295] Bluetooth: hci0: Opcode 0x2005 failed: -110
[   70.804846] bt_tx: bb3a9bfe: 41 20 0d 00 00 05 01 12 00 12 00 01 36 00 36 00  A ..........6.6.
[   70.804864] Bluetooth: hci0: Opcode 0x2041 failed: -110
[   70.818970] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   70.827191] tx: 7b856ea5: 02 00 00 00 10 00 10 00 69 02 00 00 69 65 73 01  ........i...ies.
[   70.835726] tx: 9e75d3bf: 41 20 0d 00 00 05 01 12 00 12 00 01 36 00 36 00  A ..........6.6.
[   70.858947] esp32_spi: process_rx_buf: SPI RX buf: IF type = 14
[   70.866352] SPI RX buf: 33a7b233: fe 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   70.876556] SPI RX buf: fb545a59: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   70.885663] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   72.874237] Bluetooth: hci0: command 0x2005 tx timeout

sergey-suloev, 2024-10-17:
The behavior is really strange . 

[    7.396173] esp32_spi: esp_init: esp_init: begin
[    7.401096] esp32_spi: esp_spi_probe: Probing ESP SPI-driver...
[    7.417739] esp32_spi: esp_spi_probe: raw-tp-mode was not specified
[    7.417739] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 1, data_path 0
[    7.417774] spi_master spi0: Hardware control over CS line
[    7.417783] spi_master spi0: Negate CS line between bursts
[    7.443112] esp32_spi: esp_spi_probe: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [1], SPI mode [0x01]
[    7.455725] dwmac-sun8i 1c30000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[    7.467007] esp32_spi: process_rx_buf: SPI RX buf: IF type = 3
[    7.467538] dwmac-sun8i 1c30000.ethernet eth0: PHY [mdio_mux-0.1:01] driver [Generic PHY] (irq=POLL)
[    7.472892] SPI RX buf: 9ff31aa7: 03 00 00 00 1f 00 0c 00 1c 02 00 00 01 00 18 00  ................
[    7.472931] dwmac-sun8i 1c30000.ethernet eth0: No Safety Features support found
[    7.482078] SPI RX buf: e2a1351e: 17 00 00 00 03 01 09 02 01 1e 00 01 e8 01 0c 4e  ...............N
[    7.482090] esp32_spi: process_rx_buf: 273 datapath closed
[    7.482116] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[    7.482126] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[    7.493446] esp32_spi: esp_spi_work: SPI work: handshake 0, data_ready 1, data_path 0
[    7.535385] dwmac-sun8i 1c30000.ethernet eth0: configuring for phy/mii link mode
[    7.535381] esp32_spi: esp_spi_work: SPI work: handshake 0, data_ready 0, data_path 0
[    7.887014] esp32_spi: esp_spi_probe: ESP SPI-driver probe success
[    7.893508] esp32_spi: esp_init: esp_init: end 0
[    9.161390] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 1, data_path 1
[    9.182330] esp32_spi: process_rx_buf: SPI RX buf: IF type = 3
[    9.188258] SPI RX buf: 03961b9f: 03 00 00 00 1f 00 0c 00 1c 02 00 00 01 00 18 00  ................
[    9.197378] SPI RX buf: cfbf821c: 17 00 00 00 03 01 09 02 01 1e 00 01 e8 01 0c 4e  ...............N
[    9.206501] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[    9.213454] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[    9.221823] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[    9.228242] esp32_spi: esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
[    9.236090] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[    9.242486] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[    9.248879] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[    9.255250] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[    9.261891] esp32_spi: print_reset_reason: POWERON_RESET
[    9.267236] esp32_spi: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
[    9.274716] esp32_spi: esp_reg_notifier: Driver init is ongoing
[    9.281085] esp32_spi: esp_cfg80211_add_iface: Updated priv[0] to c1cd1580
[    9.337185] esp32_spi: esp_cmd_work: Processing Command [0x1]
[    9.343095] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[    9.351125] tx: 13c6233b: 00 00 01 00 08 00 0c 00 16 00 00 00 01 00 00 00  ................
[    9.359612] tx: 8c466c60: 00 00 00 00 ff ff ed fc ff ff ff ff bf ff ff ff  ................
[    9.381098] esp32_spi: process_rx_buf: SPI RX buf: IF type = 15
[    9.388136] SPI RX buf: 2d19a8a5: ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    9.399538] SPI RX buf: d0ecdba2: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    9.409988] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 1, data_path 1
[    9.437924] esp32_spi: process_rx_buf: SPI RX buf: IF type = 15
[    9.445560] SPI RX buf: 081f1667: ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    9.457452] SPI RX buf: 9fb08f64: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    9.468973] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 1, data_path 1
[    9.489851] esp32_spi: process_rx_buf: SPI RX buf: IF type = 0
[    9.495735] SPI RX buf: 19bcd13d: 00 00 02 00 14 00 0c 00 25 00 00 00 01 02 00 00  ........%.......
[    9.504839] SPI RX buf: 8e922db5: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    9.520639] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[    9.520659] esp32_spi: wait_and_decode_cmd_resp: Resp for command [0x1]
[    9.558293] esp32_spi: esp_cmd_work: Processing Command [0x3]
[    9.564126] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[    9.572021] tx: 6c5ae123: 00 00 01 00 08 00 0c 00 18 00 00 00 03 00 00 00  ................
[    9.580415] tx: 43ddbbc6: 00 00 00 00 ff ff ab ff ff ff be ff ff f7 ff ff  ................
[    9.598466] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[    9.601792] esp32_spi: process_rx_buf: SPI RX buf: IF type = 15
[    9.612967] SPI RX buf: 250e8f2c: ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    9.622059] SPI RX buf: f1f3e615: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    9.631162] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 1, data_path 1
[    9.651959] esp32_spi: process_rx_buf: SPI RX buf: IF type = 15
[    9.657006] esp32_spi: esp_cmd_work: Busy in another cmd execution
[... cut ...]
[   15.419487] esp32_spi: process_rx_buf: SPI RX buf: IF type = 2
[   15.425353] SPI RX buf: 4a2e8203: 02 00 00 00 07 00 0c 00 87 00 00 00 04 0e 04 05  ................
[   15.434425] SPI RX buf: 2752365c: 37 20 00 3f b9 05 00 00 24 17 ca 3f 24 17 ca 3f  7 .?....$..?$..?
[   15.443656] esp32_spi: esp_spi_work: SPI work: handshake 1, data_ready 0, data_path 1
[   20.229739] esp32_spi: esp_inetaddr_event: ------- IP event for interface eth0 -------
[   50.281368] esp32_spi: esp_inetaddr_event: ------- IP event for interface lo -------

sergey-suloev, 2024-10-17:
The behavior is really strange. 

[log.txt](https://github.com/user-attachments/files/17407367/log.txt)

mantriyogesh, 2024-10-17:
@kapilkedawat ++

kapilkedawat, 2024-10-17:
@sergey-suloev could you please try with this patch?
```
diff --git a/esp_hosted_ng/host/spi/esp_spi.c b/esp_hosted_ng/host/spi/esp_spi.c
index b6ee52d9..b54eb0f5 100644
--- a/esp_hosted_ng/host/spi/esp_spi.c
+++ b/esp_hosted_ng/host/spi/esp_spi.c
@@ -178,6 +178,8 @@ int esp_validate_chipset(struct esp_adapter *adapter, u8 chipset)
                break;
        }
 
+       hardware_type = adapter->chipset;
+
        return ret;
 }
 
```

sergey-suloev, 2024-10-17:
> @sergey-suloev could you please try with this patch?
>
> ```
> diff --git a/esp_hosted_ng/host/spi/esp_spi.c b/esp_hosted_ng/host/spi/esp_spi.c
> index b6ee52d9..b54eb0f5 100644
> --- [...]

What is the reason for the patch ? The variable is only used here https://github.com/espressif/esp-hosted/blob/30f4082314b6e13d869e9bdff7949fa428713337/esp_hosted_ng/host/spi/esp_spi.c#L343

I am using ESP32-S3 and so this patch won't affect the issue.

kapilkedawat, 2024-10-17:
Ahh, I thought it was ESP32. I did encounter some timing issues with the ESP32 previously, and the fix was for that.

We also tried using the ESP32-S3 with different kernel versions on the Raspberry Pi (both 32-bit and 64-bit), but we didn't encounter this kind of issue. You can see the [logs](https://github.com/user-attachments/files/17411016/test_spi.1.tgz) for the bootup and scan of the ESP32-S3.

I noticed in the logs that the data-ready pin is not set. Could you please enable [BT data path debugging](https://github.com/espressif/esp-hosted/blob/30f4082314b6e13d869e9bdff7949fa428713337/esp_hosted_ng/esp/esp_driver/network_adapter/main/Kconfig.projbuild#L143) in the firmware and then provide the logs along with logic analyzer dump?

sergey-suloev, 2024-10-17:
Here is ESP32s3  firmware log taken with the sunxi board.

https://pastebin.com/fnGZ4C1s

kapilkedawat, 2024-10-18:
Hi @sergey-suloev,

In the RPI logs, the ESP32-S3 is continuously rebooting. We don't have an RPI 3A+ to test on, but since the GPIO layout seems the same on both the 4B+ and 3A+, it should theoretically work without issues—unless there's a hardware problem on one of the boards.

One question: How are you powering the ESP32-S3?

As for the Sunxi board, I don't see any HCI command timeouts in the logs, and the transport looks okay, at least in the recent logs(all bt_tx from host reaches to firmware and vice versa).

For example, in the logs:

Request:

```
[   10.475642] bt_tx: e3d0e6ff: 03 0c 00 
```

Response:
```
[   10.611671] SPI RX buf: 8b5102f0: 02 00 00 00 07 00 0c 00 3f 00 00 00 04 0e 04 05  ........?.......
[   10.622175] SPI RX buf: 45e719ad: 03 0c 00 3f 00 00 00 00 00 00 00 00 88 40 ca 3f  ...?.........@.?
```
Request:
```
[   10.638161] bt_tx: e31daef9: 03 10 00 
```
Response:
```
[   10.772239] SPI RX buf: 821c38c9: 02 00 00 00 0f 00 0c 00 b3 00 00 00 04 0e 0c 05  ................
[   10.781371] SPI RX buf: 1582ad0b: 03 10 00 00 00 00 00 60 00 00 00 00 88 40 ca 3f  .......`.....@.?
```
Request:
```
[   10.790665] bt_tx: 771f455f: 01 10 00 
```

Response:
```
[   10.945084] SPI RX buf: 4bfa62e0: 02 00 00 00 0f 00 0c 00 76 01 00 00 04 0e 0c 05  ........v.......
[   10.954607] SPI RX buf: c523cd1d: 01 10 00 09 16 00 09 e5 02 16 00 00 88 40 ca 3f  .............@.?
```

Request:
```
[   10.972068] bt_tx: 9fe0fb3d: 09 10 00
```

Response:
```
[   11.126479] SPI RX buf: 557aff6a: 02 00 00 00 0d 00 0c 00 1c 03 00 00 04 0e 0a 05  ................
[   11.137887] SPI RX buf: 06a58ef3: 09 10 00 f6 6a 96 18 85 34 16 00 00 88 40 ca 3f  ....j...4....@.?
```

and so on.. 

I also don't see and bt scan commands in the host console? Logs were captured during issue?

sergey-suloev, 2024-10-18:
here is the full console log
[log.txt](https://github.com/user-attachments/files/17433956/log.txt)

sergey-suloev, 2024-10-18:
> One question: How are you powering the ESP32-S3?

The power schematic is the same as for ESP32 :  5V from the board via 3.3V LDO. I verified voltages, everything is ok.

sergey-suloev, 2024-10-18:
> In the RPI logs, the ESP32-S3 is continuously rebooting. We don't have an RPI 3A+ to test on, but since the GPIO layout seems the same on both the 4B+ and 3A+, it should theoretically work without issues—unless there's a hardware problem on one of the boards.

I tested ESP32 with the same board. The communication isn't 100% stable but this at least worked in SPI mode 2. ESP32s3 isn't working at all. Strange.

[4 more replies]

closed 2024-12-22 by sergey-suloev