espressif/esp-hosted #490: [NG] can you advise how to change spi speed for NG

preveen-stack, 2024-09-18:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

add  spispeed=<spispeed> as a parameter to rp_init.sh script

### Use cases

SPI at 10MHz is unstable

mantriyogesh, 2024-09-18:
[module_param(clockspeed, uint, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
](https://github.com/espressif/esp-hosted/blob/ce00ca53189ffd52455dd5426d54f9be6fae1871/esp_hosted_ng/host/main.c#L47)

Simply use:

```
insmod esp32-spi.ko clockspeed=<freq_in_mhz>
```

closed 2024-09-18 by mantriyogesh

preveen-stack, 2024-09-18:
Thanks for the suggestion but on doing that I am encountering the below error

> [ 4713.236606] esp32_spi: prepare_command_request: command queue init is not done yet
[ 4713.236638] esp32_spi: cmd_get_tx_power: Failed to get command node
[ 4775.984352] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [5]
[ 4807.248372] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[ 4807.248460] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[ 4807.248475] esp32_spi: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SPI
[ 4807.248487] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[ 4807.248497] esp32_spi: adjust_spi_clock: ESP Reconfigure SPI CLK to 10 MHz
[ 4807.248507] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[ 4807.248516] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[ 4807.248524] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[ 4807.248533] esp32_spi: print_reset_reason: POWERON_RESET
[ 4807.248545] esp32_spi: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
[ 4807.249055] esp32_spi: esp_reg_notifier: Driver init is ongoing
[ 4812.307007] esp32_spi: wait_and_decode_cmd_resp: Command[0x1] timed out
[ 4812.307032] esp32_spi: cmd_init_interface: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[ 4812.307052] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[ 4812.307056] esp32_spi: process_event_esp_bootup: network iterface init failed

mantriyogesh, 2024-09-18:
Timeouts in your case may be indicating some issues in spi. please review [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) carefully and step by step.